pcengines/apu2-documentation #59: Mainline with pfSense

kjake, 2018-01-26:
Now that pfSense has gone through major updates and is running FreeBSD 11.1 are there any known issues with using the Mainline branch of coreboot?  I don't have a way to recover, so I'm not brave enough to test moving up from the 4.0.x / Legacy branch.

pietrushnic, 2018-01-27:
Hi @kjake,
we observe various problems with pfSense. At least we saw regression when reading from USB and also when installing to HDD connected through mPCI SATA expander. We did tests with SATA connected directly and this seems to work. USB problems seems to be related to XHCI disabling in v4.6.5.

We working on building automated regression testing for pfSense, OPNSense and other BSD systems.

If you don't have way to recover I would not recommend flashing new firmware. If you loose power during flashing you may end up with not bootable device. Please note that recovery SPI is very cheap 3USD - you can find it [here](http://pcengines.ch/spi1a.htm). Also please note that Raspberry Pi would work as recovery device if you have skills and patience. You can read more about using RPi as recovery tools in various blog posts - 3mdeb also have article about that [here](https://3mdeb.com/firmware/flashing-minnowboard-turbot-with-raspberry-pi-zero-w/).

pietrushnic, 2018-01-27:
@miczyg1 please let @kjake know if anything will improve in area of pfSense. I believe we should make test result public sooner or later.

Veldkornet, 2018-02-01:
I installed version 4.6.1 (over 4.0.11) yesterday on my pfSense. 
It’s not that happy...

Appears as though after a couple of hours it locks up and reboots itself. 

Happened twice now since I’ve upgraded.

miczyg1, 2018-02-01:
Hi @Veldkornet,
Thank You for contacting us. We are currently investigating the issues related to pfSense in 4.6.x versions. 

Could You please tell us on what media have You installed the pfSense (SD card, USB stick, SATA HDD etc.)?

Veldkornet, 2018-02-01:
I’m using a Samsung SSD 850 EVO mSATA 250GB

dacmot, 2018-02-28:
I'm also having random lock-ups and reboots with pfSense v2.4.2_1. I had to flash a new BIOS because the wifi adapter wasn't recognized by pfSense. I have the APU3 BIOS 4.6.1 binary downloaded from github. My hardware is:
1 apu3c4 APU.3C4 system board 4GB
1 msata60b SSD M-Sata 60GB MLC Phison
1 wle200nx Compex WLE200NX miniPCI express card

The lock-ups mostly seemed to happen when I was configuring the network interfaces. Now I managed to setup a LAN bridge with one port and the wifi, and it's been up and running for over 24h. I don't know if that could have anything to do with it.

miczyg1, 2018-03-06:
@kjake @Veldkornet we have found a workaround to fix pfSense issues for BIOS version v4.6.x. Please follow [pfSense install guide](https://github.com/pcengines/apu2-documentation/blob/master/docs/pfSense-install-guide.md) as @dacmot says.

kevindd992002, 2018-03-06:
Is the 4.6.x line better than the 4.0.x line?

Veldkornet, 2018-03-06:
For me, only the following was missing from the /boot/loadef.conf:

`hint.ahci.0.msi="0"`

I've added it and upgraded back to 4.6.1. Will let you know how it goes.

kjake, 2018-03-07:
I did the same as @Veldkornet on my existing install and have updated to
4.6.1.  Running for almost 5 hours without issue.

Veldkornet, 2018-03-07:
Appears to be good for me too, no reboots as yet.

Veldkornet, 2018-03-09:
FYI, the LED’s stopped working after updating to this firmware. The apuled.ko module is still loaded, although if I do a: ls /dev/led it only shows the Ethernet ones, not the front panel ones.

pietrushnic, 2018-03-10:
@Veldkornet I belive this is different issues. Isn't LED driver rely on entries in SMBIOS  for detecting platform name ?

Veldkornet, 2018-03-10:
Well, if it’s the below, then the name in the BIOS has changed in this version.. AGAIN! Need to submit a code change request to pfSense as well because the name changed again... What was wrong with the way it was done in 4.0.7???

For my APU2C2 board prior to v4.0.7 BIOS:
/bin/kenv -q smbios.system.product : apu2
/bin/kenv -q smbios.system.maker : PC Engines

For my APU2C2 board with v4.0.7 BIOS:
/bin/kenv -q smbios.system.product : APU2
/bin/kenv -q smbios.system.maker : PC Engines

For my APU2C2 board with v4.6.1 BIOS:
/bin/kenv -q smbios.system.product : PC Engines apu2
/bin/kenv -q smbios.system.maker : PC Engines

pietrushnic, 2018-03-13:
@Veldkornet the problem was mixing SMBIOS table on which various distro rely on with sign-of-life string on which various customer rely on. Correct version expected by coreboot is what you saw <v4.0.7:

```
/bin/kenv -q smbios.system.product : apu2
/bin/kenv -q smbios.system.maker : PC Engines
```

Confusion with naming was result of couple things: way SMBIOS tables are used, requirements from customers and requirements from coreboot. We were not aware of various requirements from different stakeholders and that why you see those problems. Starting with recent release everything should be fixed as it was for legacy firmware <v4.0.7.

We will publish recent release soon.

charliewutwut, 2018-03-19:
Hi,

I ran into this issue yesterday after upgrading my APU2C4 from coreboot build 20160311 to v4.6.1.
The issue is still there with and without my external drive attached.

"Fixed" in loader.conf with:
```
hint.ahci.0.msi="0"
hint.ahci.1.msi="0"
```
Im still on v4.6.1 and I have not seen any other issues for now :)

OS: FreeBSD 11.1-RELEASE-p8 root @amd64-builder.daemonology.net:/usr/obj/usr/src/sys/GENERIC  amd64

Hardware:
OS disk: Transcend TS64GMSA370 64GB
Data disk: 8TB Seagate ST8000DM002 in a ICY BOX IB-366StU3+B

pietrushnic, 2018-03-20:
@charliewutwut thanks for confirming. We described fix [here](https://github.com/pcengines/apu2-documentation/blob/master/docs/pfSense-install-guide.md).

Veldkornet, 2018-03-22:
@pietrushnic is the BIOS with the updated naming already out? Is it 4.6.7? Or will you let us know? 

Beside the LED’s, I noticed some other small things, like it no longer beeps when you login. I guess pfSense enables these features when it sees its an APU2, which it now doesn’t see because of the naming.

pietrushnic, 2018-03-22:
@Veldkornet fixes are in 4.6.7 and 4.0.15. We also found a fix for pfSense in the firmware that not requires any tweaks during installation. It happened that mainline by default set controller in IDE mode we changed that and things work fine. We will release that in 4.6.8.

kevindd992002, 2018-03-22:
So in 4.6.8 you won't need to follow the pfsense install guide at all and we don't need to do ANY modifications/tweaks as stated in the guide?

Other than the LED and reboot issue with the mainline, any other major issues? Is the mainline better than the legacy?

kjake, 2018-03-22:
I'm on 4.6.1 with an APU2C4 and my system is still beeping appropriately when it should (login, shutdown, boot-up).

pietrushnic, 2018-03-22:
@kevindd992002 from maintainer perspective legacy (4.0.x) is bad, but the problem is that I have not enough validation power to tell you that mainline (4.6.x) is good enough - please test it by yourself and help community what is the quality of that firmware. LED should work fine on 4.6.7 and 4.0.15 (assuming driver use vendor: PC Engines and device: apu2).

Our plan is to have 4.6.8 which do not require following pfsense installation guide and install it correctly.

Veldkornet, 2018-05-17:
I've finally managed to get onto 4.6.8.

So far I have no problems, only in the pfSense interface, the serial number is all garbled...
[image]

Also from the command line:
```
/bin/kenv -q smbios.system.serial
▒▒▒▒▒▒▒▒▒▒
```

pietrushnic, 2018-05-17:
@Veldkornet I would appreciate separate issues for new bugs. BTW we released [4.6.9](https://github.com/pcengines/coreboot/blob/coreboot-4.6.x/CHANGELOG.md#v469---2018-05-11) which fix that problem.

closed 2019-02-26 by miczyg1