pcengines/apu2-documentation #62: Mainline firmware SD speed

c-e-p-x-u-o, 2018-02-20:
Both apu3 v4.5.5 and v4.6.1 roms gives 11 Mb/s on SD card
while v4.0.12  gives 20 Mb/s.

pietrushnic, 2018-02-20:
@c-e-p-x-u-o thank you for this report, we will try to narrow down this regression. Can you send information about SD card you using ?

c-e-p-x-u-o, 2018-02-20:
All that I have:
SanDisk Extreme PRO SDHC 32Gb 95 mb/s
mmc0: new high speed SDHC card at address aaaa
mmcblk0: mmc0:aaaa AGGCD 29.7 GiB 

SanDisk Ultra microSDXC 64Gb
mmc0: new high speed SDXC card at address e624
mmcblk0: mmc0:e624 SU64G 59.5 GiB (ro)

SanDisk Industrial microSDHC 8Gb (white color)
mmc0: new high speed SDHC card at address aaaa
mmcblk0: mmc0:aaaa SL08G 7.40 GiB (ro)

Kingston microSDHC 8Gb  SDC10/8GB
mmc0: new high speed SDHC card at address 0007
mmcblk0: mmc0:0007 SD8GB 7.42 GiB (ro)

Transcend  microSDHC16Gb 
mmc0: new high speed SDHC card at address 0002
mmcblk0: mmc0:0002 00000 14.9 GiB (ro)

miczyg1, 2018-03-06:
@c-e-p-x-u-o we have tracked down the issue and fixed it. Thank You again for reporting it to us.

c-e-p-x-u-o, 2018-03-06:
Thank you for fast reaction! I'm ready to test it. Where can I get it?

Realy I have some more question about hardware features. May be it's better to split it to issues, please let me know.
1. It's hard to find SATA version. After some tests I've realized that SATA version is 3 (6Gb/s), but hw is not able to utilize full speed. (intle's SSD that gives 500Mib/s r/w on corei3 gives about 400Mib/s on apu3. And it's good, but could specify at leas SATA version in HW specs.

2. As I understand, CPU supports SD v3.0 104 MB/s but due to software limitations the real SD speed is only 25Mb/s in SD v1.10 mode. Am I right? Is SD v3.0 support in a roadmap?

3. Watchdog
Even if I blacklist i2c_piix4 I got:

sp5100_tco: SP5100/SB800 TCO WatchDog Timer Driver v0.05
sp5100_tco: PCI Vendor ID: 0x1022, Device ID: 0x780b, Revision ID: 0x42
sp5100_tco: failed to find MMIO address, giving up.
instead of
sp5100_tco: I/O address 0x0cd6 already in use

Does watchdog present on board and how to use it?

miczyg1, 2018-03-06:
@c-e-p-x-u-o You can download it [here](https://cloud.3mdeb.com/index.php/s/T5ZJePvgmEtHxFY/download), but this binary is still under tests. Please use it on Your own responsibility.

According to 1. - we will check if the lowered speed is BIOS related.

According to 2. - yes, SD v3.0 mode is available for apu platforms, but it is currently disabled. It caused many problems to customers and we are still working on a fixes. 

According to 3. - watchdog module conflicts with PIIX4 (they appear to have the same MMIO adress hardcoded). Try to `rmmod sp5100_tco i2c_piix4` and then `modprobe sp5100_tco`.  You can also refer to this bug on the [PC Engines forum](http://www.pcengines.info/forums/?page=post&id=F1D33B75-58E7-4A5A-9A80-1E8CA35AC84D&fid=DF5ACB70-99C4-4C61-AFA6-4C0E0DB05B2A). 

The watchdog is present on the board. You should find a hint how to use it after reading the post.

c-e-p-x-u-o, 2018-03-07:
pv /dev/mmcblk0 > /dev/null shows about 21MiB/s with apu3_v4.6.7.rom

OK, hope to get SD v3.0 support in future.

watchdog works, forgot to update-initramfs

Thank you! (Should I close this issue?)

miczyg1, 2018-03-08:
@c-e-p-x-u-o I am glad it worked for You.

Thank You, but we will handle closing issue by ourselves.

closed 2018-03-13 by miczyg1

c-e-p-x-u-o, 2018-04-10:
apu3_v4.6.7.rom still not in public?

miczyg1, 2018-04-10:
@c-e-p-x-u-o You can find all recent releases on our new github page: https://pcengines.github.io/
We also recommend to track the forum: http://pcengines.info/forums/

c-e-p-x-u-o, 2018-04-10:
So https://github.com/pcengines/apu2-documentation/blob/master/README.md is outdated
Please update it to not confuse users.

miczyg1, 2018-04-10:
@c-e-p-x-u-o this page has been just created few days ago. We are still upgrading it, so of course  there still may be some obsolete/outdated information elsewhere. We will gradually clean everything up.