pcengines/apu2-documentation #64: Reboot/Post hang

fhloston, 2018-02-27:
apu2c4 with BIOS 4.6.1, 4.6.5, 4.6.6 hangs occasionally during soft reboot/post. 

Hard reset clears fault.

4.0.x and 4.5.5 do not show this behaviour.

Reproduce with simple shellscript in tinycore:
date >>reboot.log;sleep 60;reboot

dacmot, 2018-03-02:
Did you try [this procedure](https://github.com/pcengines/apu2-documentation/blob/master/docs/pfSense-install-guide.md)?

fhloston, 2018-03-02:
@dacmot I do not see the relevance here?

Perhaps I wasn't precise enough before. 

The issue is that the apu2c4 does hang during POST when it is warm rebooted from a running OS. Does not matter if that OS is tinycore, debian or opnsense. It also does not matter if the boot media is a usbstick or a sata drive.

When the system hangs only a hard reset helps to get the unit in a working condition. sp5100_tco watchdog does also not help - even when J2 5-6 is shorted.

a1bert01, 2018-03-04:
4.0.7, 4.0.12 SD card not recognized after reboot...

miczyg1, 2018-03-06:
@a1bert01 what operating system are You running? Are You rebooting from system installed on SD card?

@fhloston thank You for reporting the issue. We will check that and try to fix it.

a1bert01, 2018-03-06:
Yes, I am using/rebooting from just SD card, 
with 4.6.1. bios:

Grub (slackware) hangs without message

PCEngine's  apu2-tinycore6.4.img.gz:

> reboot: machine restart
PC Engines apu2
coreboot build 08/30/2017
BIOS version v4.6.1
2032 MB DRAM

>SeaBIOS (version rel-1.10.2.1)

>Press F10 key now for boot menu

>Booting from Hard Disk...

>SYSLINUX 6.03 20150820 Copyright (C) 1994-2014 H. Peter Anvin et al                                                                                                                                                                      
Loading vmlinuz... CHS: Error 0c00 reading sector 30852 (2/12/15)                                                                                                                                                                        
EDD: Error 0c00 reading sector 32900                                                                                                                                                                                                     
CHS: Error 0c00 reading sector 35802 (2/90/51)                                                                                                                                                                                           
EDD: Error 0c00 reading sector 37850                                                                                                                                                                                                     
ok                                                                                                                                                                                                                                       
Loading core.gz...CHS: Error 0c00 reading sector 2 (0/32/35)                                                                                                                                                                             
EDD: Error 0c00 reading sector 2050                                                                                                                                                                                                      
CHS: Error 0c00 reading sector 929 (0/47/17)                                                                                                                                                                                             
EDD: Error 0c00 reading sector 2977                                                                                                                                                                                                      
failed: I/O error                        

with 4.0.12 bios:
Grub prints:
>Restarting system.                                                                                                                                                                                                                       
machine restart                                                                                                                                                                                                                          
PC Engines apu2                                                                                                                                                                                                                          
coreboot build 20170831                                                                                                                                                                                                                  
BIOS version v4.0.12                                                                                                                                                                                                                     
2032 MB DRAM                                                                                                                                                                                                                             
                                                                                                                                                                                                                                         
>SeaBIOS (version rel-1.10.2.1)                                                                                                                                                                                                           
                                                                                                                                                                                                                                         
>Press F10 key now for boot menu                                                                                                                                                                                                          
                                                                                                                                                                                                                                         
[... cut ...]
Entering rescue mode...                                                                                                                                                                                                                  
grub rescue> ls

>grub rescue> 

syslinux (tinycore):

>reboot: machine restart
PC Engines apu2
coreboot build 20170831
BIOS version v4.0.12
2032 MB DRAM

>SeaBIOS (version rel-1.10.2.1)

>Press F10 key now for boot menu

>Booting from Hard Disk...

>SYSLINUX 6.03 EDD 20150820 Copyright (C) 1994-2014 H. Peter Anvin et al

miczyg1, 2018-04-09:
@TheEvilCoder42 these two entries in loader.conf You are referring to were necessary to prevent running or installing pfSense hang due to failed write command on HDD. They are affecting SATA controller which has little to do with reboots and power cycles.

Look at the second @fhloston  comment. No matter which OS/quirks/tweaks do You use, the problem will exist. We are aware of the issue and are working on it.

pietrushnic, 2018-05-22:
@fhloston I automated testing this issue and was able to run over 35 cycles of reboot without issue. I runtest on:

* apu2c
* v4.6.9
* Debian with 4.14.y kernel
* booted over iPXE

Overnight want to run 150 iterations if I will not hit that bug we have to narrow down difference between our configurations. If you can check 4.6.9 on your side it would be great.

fhloston, 2018-05-24:
@pietrushnic cool, will test!

Can you create the v4.6.9 tag in the release manifest repo?

miczyg1, 2018-05-24:
@fhloston we are not using manifests anymore for newest releases. If You want to compile firmware by yourself, please use the new [pce-fw-builder](https://github.com/pcengines/pce-fw-builder). Refer to README for usage and build binary using pce-fw-builder v1.0.0 tag.

Binaries are also published on PC Engines github page: https://pcengines.github.io/

fhloston, 2018-05-24:
@miczyg1 thanks for the headsup, need to create new jenkins job then.

fhloston, 2018-05-24:
@pietrushnic it hangs after 5 reboots here :/

`root@box:/media/TINYCORE# 5 /media/TINYCORE/reboot.log

Syncing all filesystems.
Killing  all processes.
Terminating  all processes.
Unmounting all filesystems. 
Shutdown in progress.

The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
Requesting system reboot
sd 4:0:0:0: [sdc] Synchronizing SCSI cache
sd 1:0:0:0: [sdb] Synchronizing SCSI cache
sd 0:0:0:0: [sda] Synchronizing SCSI cache
reboot: Restarting system
reboot: machine restart
PC Engines apu2
coreboot build 20180511
BIOS version v4.6.9
`

pietrushnic, 2018-05-24:
@fhloston last time we survived 48 reboots and I hit some issue with test framework (exactly Robot Framewrk and terminal emulator called pyte). So it was not failure of system. I think we have to try to recreate your configuration. What storage are you using to boot? Also can you link us to TinyCore you use?

fhloston, 2018-05-24:
@pietrushnic it's TinyCore 6.4 downloaded from the pcengines website especially for apu2 some time ago to reproduce this bug.

I have only added the reboot script.

pietrushnic, 2018-05-29:
@fhloston finally I managed to automate Core 6.4 booting over iPXE. I started 50x reboots and will get back with results. @miczyg1 has some updates, that this may be related more to storage and booting over iPXE maybe not correct reproduction procedure.

fhloston, 2018-05-29:
If you managed to get 50 reboots without hang via iPXE boot i would ollow your assumption, that iPXE does not trigger the problem.

My testcase is an external USB stick with tinycore 6.4. 

Initially i reproduced also via FBSD/Debian on SATA. 

I have never attempted to reproduce with iPXE boot. 

I have started to try with tinycore and SD-Card but stopped due to the different device naming and the then not working autostart.sh.

TL;DR: USB/SATA and Debian/FreeBSD/Tinycore to reproduce

miczyg1, 2018-05-29:
@pietrushnic unfortunately booting a Debian over iPXE and doing reboot hanged platform after 5 repeats. So I would assume it is not storage problem, but overall platform issue.

fhloston, 2018-05-29:
If you boot via iPXE do you disconnect local storage first - if there is any in the first place?

miczyg1, 2018-05-29:
@fhloston I do not disconnect the storage. Actually during the test I had two USB stick plugged to teh front connector. I will try a run without any media.

pietrushnic, 2018-05-29:
@miczyg1 yeah, maybe that is the case that we have some media attached I think I was on setup without any storage connected.

miczyg1, 2018-05-29:
@pietrushnic @fhloston platform survived 50 reboots from Debian booted via iPXE. Storage media might be a good shot in this issue.

NanoCaiordo, 2018-08-11:
Hi, I have the same issue from mSata with v4.8.0.2 on reboots from pfSense.
I have used the following tweaks but none worked:

hint.ahci.0.msi="0"
hint.ahci.1.msi="0"
hw.acpi.handle_reboot="0"
hw.acpi.disable_on_reboot="1"

but it still hangs after the bios version message.

Cold boot are ok, repeated reboot within few minutes are ok, but rebooting after a while the system was on still hangs.

pietrushnic, 2018-08-14:
@miczyg1 is this the same like this `Fch OEM config in INIT RESET Done`. @krystian-hebel FYI

@miczyg1 anyone can validate if this was the case on legacy?

Veldkornet, 2018-08-19:
FYI, I also have this problem on the newer firmwares. Currnetly on 4.8.0.2. 
Never experienced this problem on the older firmwares (< 4.0.11)

pietrushnic, 2018-08-19:
@Veldkornet thanks. I really don't like that legacy firmware didn't have that issue. @miczyg1 @krystian-hebel we should take a look at it as soon as we get free cycles.

Slarky, 2018-09-02:
Hi,

I have the same issue, APU2c4, v4.8 ROM. As soon as I type reboot (running Alpine Linux) this is where it hangs every time:

```
The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
Requesting system reboot
[44404.430227] reboot: Restarting system
PC Engines apu2
coreboot build 20180708
BIOS version v4.8.0.3
PC Engines apu2
coreboot build 20180708
BIOS version v4.8.0.3
4080 MB ECC DRAM
```

I have to cut power to get it up and running again.

krystian-hebel, 2018-09-02:
@Slarky thanks for Your report, but I think this is actually a different issue. Your platform hangs after reporting RAM size, and a lot happens in between. I also see that You're using v4.8.0.3, in which we addressed issue with reset in early platform initialization that resulted in BIOS version reported twice after some kind of unexpected transitions to S5, but in Your log it is printed nonetheless, so I believe it resets somewhere further into boot process (which is valid according to [AGESA documentation](https://support.amd.com/TechDocs/44065_Arch2008.pdf) - page 39, point E).

Could You tell us more about Your configuration? Do You use any USB drives or other devices?

pietrushnic, 2018-09-02:
@krystian-hebel does it make sense to provide debug build? If yes, then please expose it over next cloud.

Slarky, 2018-09-04:
@krystian-hebel nothing fancy, a APU2c4; KINGSTON SMS200S330G mSATA SSD and last but not least a wle600vx mPCI WLAN card; no USB drives whatsoever

lathama, 2018-09-06:
I just noticed this on three identical APU2C4 all at v4.8.0.3 running Debian Stretch. After changing unattended updates to reboot if needed I found them all stuck and a manual power plug pull was required.

What info is needed for troubleshooting this further?

miczyg1, 2018-09-06:
@lathama we have confirmed the reboot problems on all our platforms we are testing during firmware releases. We can reproduce the issue so we will be able to fix it. We do our best.

@lathama by the updates You did mean Debian updates or firmware update?

lathama, 2018-09-06:
Thanks

Updates == Debian Package updates with an automatic reboot if the kernel or libc was updated.

bconway, 2018-09-07:
Too add to the list, I have the same warm reboot hang between BIOS version being printed and memory being printed with OpenBSD 6.3-stable via USB flash drive on apu2c4. If I reboot the device over and over, it will usually work, but if I leave the system running for 1-2 days, it will almost always hang on warm reboot before printing memory.

Currently on 4.8.0.2. I think I first noticed this somewhere between 4.0 and 4.6.

gretel, 2018-09-27:
finally managed to upgrade one `apu2` to `4.8.0.4` and i could reboot fine with a `USB` device attached.
btw, `OpenBSD-6.4` brings support for:
```
ccp0 at pci0 dev 8 function 0 "AMD Cryptographic Co-processor v3" rev 0x00
```

gretel, 2018-09-27:
duh, hung on second reboot. why is this happening.

pietrushnic, 2018-09-27:
@miczyg1 @krystian-hebel can we do anything with that? I assume we have reproduction, but it is not so frequnet?

@gretel I think problem may be in AGESA. We will try to solve that, but if this is inside AGESA it may not be easy. Talking about CCP, did you tested it? We know it doesn't work on Linux, but if this works for OpenBSD that means Linux driver not firmware issue.

gretel, 2018-09-27:
@pietrushnic this issue haunts me from the day i got the boards until today. 
in regards to `ccp`:
```
     The ccp driver supports feeding output of the random number generator
     into the kernel entropy pool.
```
as this seems to be all it does i did not figure any way to check if it's actually working.

pietrushnic, 2018-09-27:
@gretel I thought you can use `ccp` as backend for random number genrator e.g. like described [here](https://cryptotronix.com/2014/08/28/tpm-rng/). ccp related stuff is on roadmap here: https://github.com/pcengines/apu2-documentation/issues/112

pietrushnic, 2018-09-27:
@gretel BTW did you tried legacy, have you got the same issue with it?

gretel, 2018-09-27:
> @gretel I thought you can use `ccp` as backend for random number genrator e.g. like described [here](https://cryptotronix.com/2014/08/28/tpm-rng/). ccp related stuff is on roadmap here: #112

this is what i quoted in regards to openbsd, didn't i? 😏

gretel, 2018-09-27:
> @gretel BTW did you tried legacy, have you got the same issue with it?

no, and i don't feel like flashing a production box just for the try. nothing has worked in all the time.
the boards have been sold containing this bug, it hasn't been fixed as promised, and i don't feel like playing the alpha-tester any longer.

if this can't be fixed i will have to change platforms.
boards that hang on reboot are **not acceptable**.

additionally, the `legacy` firmware lacks features at least now in the `mainline`.

bconway, 2018-09-27:
I have a spare apu2c4, as time allows I will try it with 4.8.0.4 with a non-USB boot device, along with legacy and USB flash boot. I can very reliably reproduce the hang after 1-2 days of uptime and warm reboot.

Veldkornet, 2018-09-27:
I have 1x APU2 on 4.8.0.4 and a family member has 1x APU2 on 4.0.6??? Both using a SATA SSD, both used for pfSense (now 2.4.4) if that would be useful to compare things. 

As already mentioned, legacy didn’t have the issue; 4.0.x and prior when they just had the date as the version. 

I’m happy to help, but honestly I have no idea what I could do except flash to the next version when it’s available :P If anyone needs something from me just let me know :)

pietrushnic, 2018-09-27:
@Veldkornet key problem from our side is reproducibility. It looks like we have different conditions in which platform work, you can see that this issue thread mix many problems which may but don't have to have the same root cause. 

To prove that working hard on this problem please read [this analysis](https://github.com/pcengines/apu2-documentation/pull/108/files). IMO stuff can be related with things going on in AGESA, which is binary blob provided by AMD - not much we can do with it, if we will know issue we may try workaround it before calling to AGESA binary by modification of parameters that we pass to it or values of registers based on which decision is performed. We will experiment with a problem to address issues from this thread.

kevindd992002, 2018-10-25:
Do we have any updates to this issue? I'm holding upgrading to the latest firmware because of this.

miczyg1, 2018-10-25:
@kevindd992002 not yet. You may track the state of reboots on our test results spreadsheet available on the release site: https://pcengines.github.io/ (Regression Test Results).

Also mentioned by many users here, the issue is said to reproduce after 1-2 days of uptime. In this conditions it is hard to even reproduce the issue to debug it properly. Given that reproducing procedure alone takes much time, so does have to the solution. Especially for not trivial issue like that.

gretel, 2018-10-25:
@miczyg1 thanks for the feedback. i won't reproduce spending money on this platform much 😏

blocknonip, 2018-10-25:
I spent some time trying to get a consensus on hangs/versions with my spare boards, but it's proving a lot more difficult to get the hangs with systems that aren't running production loads. I can verify that there are no issues with running 4.0.19 on my production systems (4.0.20 wasn't released at the time).

NanoCaiordo, 2018-10-25:
Yes, there no issues on 4.0.x and on my side I never had such issue on this branch, however it persists on 4.8.x and I'd like to add that I only had to wait roughly 5 minutes to be able to reproduce it.

tudorbarascu, 2018-11-05:
I also exhibited this problem on Centos 7 and Debian Buster, apu2c4, 4.8.0.5 version.
I cannot replicate it anymore on Centos by doing a ``yum remove microcode_ctl`` (inspired from https://www.centos.org/forums/viewtopic.php?t=46524).
I'm on a Kingston SSD.

blocknonip, 2018-11-05:
This issue is about a hang that happens mid-BIOS boot. It is not about the OS hanging during the shutdown process.

tudorbarascu, 2018-11-05:
@blocknonip I know. The issue that I had was exactly the one in this issue: - an OS reboot would freeze the API after this part:
```
PC Engines apu2
coreboot build 20180708
BIOS version v4.8.0.3
```

But cutting the power worked.
Now after doing "yum remove microcode_ctl" on Centos it doesn't hang anymore.
I could have replicated this 90% of the time on multiple apu2c4. Now that it doesn't happen anymore things are surely tied (if I reinstall microcode_ctl I can replicate again).
It's my bad that I posted the link to the forum where I got the idea but the solution works for CentOS (don't know the whys.. but it surely works).
Thanks!

pietrushnic, 2018-11-05:
@tudorbarascu thanks for the feedback. `microcode_ctl` seem to update microcode I'm not sure about its support for AMD. Also, I wonder about *BSD since `microcode_ctl` seem to be Linux thing (there are probably other tools like that). @miczyg1 maybe we should compare microcode versions on those platforms? What would be a reliable way to get that information?

If new microcode is uploaded behind it may cause different behavior on different platforms, we should compare platforms with the same microcode. Also maybe we use kernel without microcode dirver?

miczyg1, 2018-11-05:
AFAIK, the linux-firmware repository contains the same microcode patch level as exists in the firmware. So there is no update in fact from OS side, because AGESA does the patching in early booting process.

The only known way to me is to read MSRs on each core to get current patch level.

gretel, 2018-11-05:
> I also exhibited this problem on Centos 7 and Debian Buster, apu2c4, 4.8.0.5 version.
> I cannot replicate it anymore on Centos by doing a `yum remove microcode_ctl` (inspired from https://www.centos.org/forums/viewtopic.php?t=46524).
> I'm on a Kingston SSD.
i'm on intel `msata` ssd, btw.

krystian-hebel, 2018-11-05:
@miczyg1 and others: `cat /proc/cpuinfo` displays microcode patch number, as far as I know it does read it from MSRs.

NanoCaiordo, 2018-11-06:
On FreeBSD microcode could be probably checked with: `dmidecode -t 4` or `dmesg | grep microcode`

For the meltdown patch: `sysctl vm.pmap.pti`

b-bittner, 2018-11-07:
We have 8 apu2d4 board up and running with OPNsense and BIOS 4.6.1. We bought them in batches with a few month between each. I couldn't replicate the problem with a lab-board (and 2 productions boards) out of the first batch, but some of the boards bought later (which all are running in production) often/always have the problem not coming up after OS-Reboot.

I now swapped my lab-board with one in production (which is affected with the problem) and now I can replicate it in lab (with the already mentioned 1-2 days uptime).

As far as I can see, all the boards are identical and have the some BIOS und OS-Version, but obviously something is different. Are there different hardware-revisions or CPU-Revisions?

In short:
- 8 apu2d4 boards bought in 3 batches
- OS and BIOS are all the same
- first 3 incl. lab device never had the problem
- most of the others have the problem (till now occurrence is not documented well, so MAYBE all of them)

pietrushnic, 2018-11-08:
@michaelsteinmann can You comment on that? @b-bittner can you get serial numbers of those boards? That can be read from motherboard or through `dmidecode`.

b-bittner, 2018-11-09:
I forgot to mention that all of the boards have a 16GB mSATA SSD, ordered together with the boards as a bundle - not sure if all the SSDs are the same.

@pietrushnic 
I have 2 serial numbers for you, if you need more I have to go thru all devices manually
SN: 1217557 - First batch, running fine
SN: 1229561 - ordered later, having problems

@michaelsteinmann 
as posted, at first I couldn't replicate the problem with my lab-device, Then I swapped it with one of the problematic production-devices (device was completely swapped, config was exported and then imported again). Since then I can replicate it with my lab-device - with the some power supply I had the whole time - Lab-PS is still lab-PS, production-PS is still production-PS.
I can't say something detailed about the temperature, but I would say, all boards in production do have higher temperature than my lab-device.

gretel, 2018-11-09:
@pietrushnic 
`hw.serialno=1056887`

@michaelsteinmann
 * running on quality lab supply
 * happens less often when cold

[99 more replies]

closed 2019-06-11 by krystian-hebel