pcengines/apu2-documentation #74: APU2 - 4.6.7 No EEPROM/flash device found

Veldkornet, 2018-04-09:
I'm trying to update from 4.6.1 to 4.6.7, but I keep getting the error below. Any way round this?

```
[2.4.3-RELEASE][admin@pfsense]/root: flashrom -w /tmp/apu2_v4.6.7.rom -p internal
flashrom v1.0 on FreeBSD 11.1-RELEASE-p7 (amd64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 4, resolution: 2ns).
coreboot table found at 0xcffd6000.
Found chipset "AMD FCH".
Enabling flash write... OK.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```

Veldkornet, 2018-04-09:
Think I found the issue... https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=227252

miczyg1, 2018-04-09:
@Veldkornet thank You for reporting the issue. This may be related to the libpci as mentioned in the bug on bugzilla. We are using Debian based system for development and the same goes with flashing the firmware. On Debian system using internal programmer works well on apu boards. However it is interesting that it fails on FreeBSD based systems. We will check that and try to confirm whether it is libpci or something else.

miczyg1, 2018-04-10:
@Veldkornet the problem lies in libpci on FreeBSD. There have been some changes to the libpci code for FreeBSD: https://github.com/pciutils/pciutils/compare/v3.5.5...v3.5.6
As a result of these changes, flashrom fails to read SPI base address. Without SPI base address flashrom cannot perform atomic sequencing (== cannot use FCH SPI). This is why no flash device is recognized.

Veldkornet, 2018-05-17:
Seems to be fixed in libpci-3.5.6_1. Not sure when it will be included in pfSense, but I installed it manually:

```
wget https://pkg.freebsd.org/freebsd:11:x86:64/latest/All/libpci-3.5.6_1.txz
pkg add -f libpci-3.5.6_1.txz
```

After that, it of course complained again because the name is different:
```
flashrom v1.0 on FreeBSD 11.1-RELEASE-p10 (amd64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 4, resolution: 2ns).
coreboot table found at 0xcffd6000.
Found chipset "AMD FCH".
Enabling flash write... OK.
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI) mapped at physical address 0x00000000ff800000.
This coreboot image (PC Engines:apu2) does not appear to
be correct for the detected mainboard (PC Engines:PC Engines apu2).
Aborting. You can override this with -p internal:boardmismatch=force.
```

So I've forced it as noted above:
`flashrom -w /tmp/apu2_v4.6.8.rom -p internal:boardmismatch=force`

All seems to be working now.

closed 2018-05-17 by Veldkornet