pcengines/apu2-documentation #75: Microcode update from AMD

fhloston, 2018-04-17:


fhloston, 2018-04-17:
There is updated microcode from AMD and it can be found in the wild:

[https://github.com/platomav/CPUMicrocodes/blob/master/AMD/cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin](url)

I have successfully loaded that microcode in my debian installation.

Can you include that into the coreboot builds, please?

fhloston, 2018-04-19:
@pietrushnic I tried loading it during initrd and in a live system myself and it does contain some microcode enhancements which weren't there before:

  * Indirect Branch Prediction Barrier (IBPB)
    * PRED_CMD MSR is available:  YES
    * CPU indicates IBPB capability:  YES  (IBPB_SUPPORT feature bit)

Can't you just request the microcode from AMD?

fhloston, 2018-04-19:
I should add, the raw microcode needs to be formatted properly into microcode_amd_fam16h.bin

[https://github.com/groeck/amd-ucodegen](url)

gretel, 2018-04-20:
@fhloston could you possibly post the commands .. script you are using to load the firmware at runtime? regards

fhloston, 2018-04-20:
@gretel easy:

amd-ucodegen cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin -o microcode_amd_fam16h.bin

backup old file in /lib/firmware/amd-ucode (linux) or /usr/local/share/cpucontrol (freebsd)
put new file there
echo 1> /sys/devices/system/cpu/microcode/reload (linux)
cpucontrol -v -u /dev/cpuctl[0-3] (freebsd)

you might have to install devcpu-data on freebsd first, and activate the respective service

gretel, 2018-04-24:
@fhloston thanks! for freebsd this is what i did:

```
$ git clone --depth 1 https://github.com/groeck/amd-ucodegen
$ cd amd-ucodegen
$ make
$ curl -O https://github.com/platomav/CPUMicrocodes/blob/master/AMD/cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin
$ ./amd-ucodegen cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin -o microcode_amd_fam16h.bin
```

pietrushnic, 2018-04-30:
@fhloston decision was made to not include unofficial microcode binaries in official PC Engines firmware releases. I believe we can prepare unofficial firmware release based on `v4.6.9` and/or `v4.0.17` if there is interest in that.

pietrushnic, 2018-04-30:
@miczyg1 let's prepare unofficial releases version with new microcode. @fhloston @gretel I think, because of long weekend in Poland, it would be ready before end of next week. Sorry for delay.

gretel, 2018-04-30:
@pietrushnic cool.

miczyg1, 2018-05-08:
@fhloston we have been able to include microcode updates in coreboot, yet we still need some information about new microcode features You listed:

Indirect Branch Prediction Barrier (IBPB)
- PRED_CMD MSR is available: YES 
- CPU indicates IBPB capability: YES (IBPB_SUPPORT feature bit)

Where did You get these information from? Is there any possibility to verify these features?

fhloston, 2018-05-08:
@miczyg1 I loaded the updated microcode on a Debian Stretch Installation and used spectre-meltdown-checker to verify changes. It is also packaged in stretch-backports.

[https://github.com/speed47/spectre-meltdown-checker](https://github.com/speed47/spectre-meltdown-checker)

pietrushnic, 2018-05-10:
@miczyg1 please publish here links to custom binaries tomorrow EOD.

miczyg1, 2018-05-13:
@fhloston @gretel sorry for the delay. Binaries are available [here](https://cloud.3mdeb.com/index.php/s/yKV5SL0Rcs2L1aH/download). Keep in mind that the microcode comes from wild source. We will not take any responsibility for any damage these binaries will cause to Your boards. We are in contact with AMD and will try to obtain official microcode patch.

gretel, 2018-05-15:
@miczyg1 thanks for getting back to us! going to be back from travels on friday and report back.

nicklowe, 2018-06-11:
The latest AMD microcode/ucode is available from a reputable source:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amd-ucode

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/amd-ucode

nicklowe, 2018-06-15:
Could this be built in to 4.6.10?

miczyg1, 2018-06-15:
@nicklowe unfortunately linux demands microcode in special container designed for kernel. We could use only raw binaries delivered by AMD. Second thing is we finished development cycle for v4.6.10 last Friday, the binaries should be available soon.

gretel, 2018-06-15:
still didnt get a chance to uprade mine :(

nicklowe, 2018-06-15:
Couldn't you use the files from linux-firmware.git and use https://github.com/platomav/MCExtractor to strip the container?

fhloston, 2018-06-15:
guys, what is the issue with the container? 

@gretel posted on April 24th how that container is built from raw microcode.

nicklowe, 2018-06-16:
I flashed the APU2 4.6.9 ucode image supplied by @miczyg1 and observed the following:

root@OpenWrt:~# dmesg | grep microcode
[    1.541701] microcode: CPU0: patch_level=0x07030105
[    1.558224] microcode: CPU1: patch_level=0x07030105
[    1.570174] microcode: CPU2: patch_level=0x07030105
[    1.575078] microcode: CPU3: patch_level=0x07030105
[    1.580070] microcode: Microcode Update Driver: v2.2.

I don't think this is working currently therefore.

miczyg1, 2018-06-17:
@nicklowe it may be that kernel is overriding the microcode patchlevel. I am 100% sure that MSR register responsible for storing current patch level reported 0x07030106 in coreboot. I would have to investigate it.

nicklowe, 2018-06-17:
Hmm, well, OpenWRT's amd64-microcode package is not installed so there would be nothing to downgrade it with.

paravoid, 2018-08-17:
That git log from linux-firmware above shows commits back in May 2018 with microcode updates, but if you look closer these were just for families 15h/17h. The log for 16h (which the APU2 uses) is:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/amd-ucode/microcode_amd_fam16h.bin
…and that was last updated in November 2014.

@miczyg1 if you're in contact with AMD, it might be worth to ask them to also update linux-firmware like they've done with the other families and/or get the person who made those commits there (who seems to have an @amd.com email address :) in the loop. Thanks for your responsiveness here and your efforts in general regardless!

paulmenzel, 2018-08-20:
Judging from [this discussion][1], the linux-firmware files are also outdated for Family 15h (or the metadata wasn’t updated).

> Sadly AMD microcodes at linux-firmware repository are seriously outdated:
>
> *   microcode_amd_fam15h.bin <-- CPU ID 0x610F01 microcode - version 0x6001119 [2012-07-13]
> *   microcode_amd_fam16h.bin <-- CPU ID 0x700F01 microcode - version 0x700010F [2014-02-19]

[1]: https://review.coreboot.org/#/c/coreboot/+/28204/

pietrushnic, 2018-08-20:
@paulmenzel this is because AMD doesn't give permission to publicly release binaries. I'm pretty sure they have ucode with fixes for internal/debugging use cases. I also think that have no issue with providing code for big IBV (like AMI, Insyde, Phoenix etc.).

kolargol, 2018-09-09:
Any update on this issue?

pietrushnic, 2018-09-10:
Cześć @kolargol ,
unfortunately, no. AMD do not reply to our queries. I can say binaries definitely exist, but w have no rights to distribute. Vendor wonders about placing an official statement about AMD support so stay tuned.

What I can advise is getting binaries from not the official source (if you trust those) and building firmware yourself, there is `make menuconfig` option to include external binary.

@miczyg1 do we have some guide about that? Maybe publishing guide and/or blog post would help people watching this thread.

The last resort we probably can produce community firmware with unofficial binaries with a note that it can be potentially dangerous and we are not responsible for any damage.

If you have other suggestions please let us know we would be glad to help, but lack of silicon vendor support just ties our hands. Blame AMD not PC Engines.

kolargol, 2018-10-10:
Hi, I see 4.8.0.5 is tagged, any news on microcode for APU2?

miczyg1, 2018-10-10:
@kolargol unfortunately not. I plan to send upstream the code that will enable loading microcode patches in official coreboot for apu platforms and prepare a guide how to build PC Engines firmware image with microcode this month. But that is all we can do.

Since we cannot publish any microcode blobs (not even mentioning about obtaining any), we can not release firmware containing any microcode updates.

paravoid, 2018-10-11:
If you're in contact with them… my suggestion would be to ask AMD to push those in upstream linux-firmware, like they've done multiple times in the past for other CPU families (commits by AMD's Sherry Hurwitz, who you should probably Cc!).

These would get them licensed under linux-firmware's [LICENSE.amd-ucode](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/LICENSE.amd-ucode) which is a very permissive license and would in turn give you permission to distribute as well :) Probably easier to go with something that's already set up, rather than trying to convince them to do a vendor-specific arrangement…

pietrushnic, 2018-10-11:
@paravoid I will ping them again according to Your suggestion, but please note that Linux community was not able to force AMD to release correct firmware binaries for this family, so I doubt my complaints will have any results. More to that according to my experience this platform is treated like legacy hardware and they don't want to spend too much budget to support it.

gretel, 2018-12-11:
update: https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md

pietrushnic, 2018-12-11:
@gretel in next iteration blog post coming how to patch and validate based on this guide.

kolargol, 2018-12-12:
I confirm all works great and on OpenBSD i see mitigation works as expected. Thanks for making this possible! It is improtant addition.

gretel, 2018-12-12:
> @gretel in next iteration blog post coming how to patch and validate based on this guide.

just build as described on very recent `osx`. worked fine. just had patch out the use of `sudo` in `build.sh`. going to flash later..

kolargol, 2018-12-12:
> > @gretel in next iteration blog post coming how to patch and validate based on this guide.
> 
> @pietrushnic just built as described on very recent `osx`. worked fine. had to patch out the use of `sudo` in `build.sh`. going to flash later..

actually you have to do some more changes to work on macOS: md5sha->md5 and remove realpath reliance
[build.txt](https://github.com/pcengines/apu2-documentation/files/2672466/build.txt)

gretel, 2018-12-12:
@kolargol good point, have [coreutils](https://github.com/Homebrew/homebrew-core/blob/master/Formula/coreutils.rb) installed, guess that made it compile.

kolargol, 2018-12-12:
yes, but on clean macOS you need that changes. If someone want to try binary build for APU2: https://blog.onefellow.com/post/181047767608/apu2-spectre-mitigation

pietrushnic, 2018-12-12:
@kolargol @gretel Many thanks for testing effort. Feel free to spread the word about 3mdeb and PC Engines :)

pietrushnic, 2018-12-12:
@miczyg1 I think we can close that issue or we planning something more here?

miczyg1, 2018-12-12:
@pietrushnic I do not think we have more to do here.

kolargol, 2019-02-12:
Hi, guys. Microcode load vanished from v4.9.0.2 release - this is intended or bug ?

Firefishy, 2019-02-12:
@kolargol I was able to load in the microcode with v4.9.0.2, with an extra step. After step 4 of this: https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md do the following, edit file `release/coreboot/src/cpu/Kconfig` and the setting under "config SUPPORT_CPU_UCODE_IN_CBFS" change "default n" to "default y", then continue the guide from step 5.

miczyg1, 2019-02-12:
@kolargol the microcode update has been changed since it did not patch all cores previously. Current microcode update procedure works well for all cores. The guide is available in https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md

Please do not follow @Firefishy guide since it is wrong way to do that. There is a special Kconfig option under microcode inclusion menu:
`Add microcode patch for AMD fam16h (EXPERIMENTAL)`. It will correctly include the microcode binary into coreboot. Any other method will patch only 1 core and lead to platform instability.

Firefishy, 2019-02-12:
@miczyg1 Try follow https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md with v4.9.0.2. The `Add microcode patch for AMD fam16h (EXPERIMENTAL)` option does not display because `Include CPU microcode in CBFS` is hidden because `SUPPORT_CPU_UCODE_IN_CBFS` is set to false.

kolargol, 2019-02-13:
@miczyg1 the menu option you mention vanished from v4.9.0.2 and @Firefishy seems right. I am building firmware right now and will let you know if all cores are patched.

miczyg1, 2019-02-13:
Something may have changed in upstream repository and probably we did not notice. I will check that.

kolargol, 2019-02-13:
I confirm correct microcode loaded after @Firefishy suggested fix:

FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CFLUSH,MMX,FXSR,SSE,SSE2,HTT,SSE3,PCLMUL,MWAIT,SSSE3,CX16,SSE4.1,SSE4.2,MOVBE,POPCNT,AES,XSAVE,AVX,F16C,NXE,MMXX,FFXSR,PAGE1GB,RDTSCP,LONG,LAHF,CMPLEG,SVM,EAPICSP,AMCR8,ABM,SSE4A,MASSE,3DNOWP,OSVW,IBS,SKINIT,TOPEXT,DBKP,PERFTSC,PCTRL3,ITSC,BMI1,**IBPB**,XSAVEOPT

on all 4 cores

I have pushed bin if someone want to test: https://github.com/kolargol/apu2_firmware

miczyg1, 2019-02-13:
Microcode inclusion Kconfig settings changed upstream, which led to unavailability to patch the microcode for apu. Fix will be included in v4.9.0.3: https://github.com/pcengines/coreboot/pull/271

doktornotor, 2019-03-21:
> I have pushed bin if someone want to test: https://github.com/kolargol/apu2_firmware

It works... On FreeBSD, I get this additional line in CPU features:

`AMD Extended Feature Extensions ID EBX=0x1000`

and

`x86info -a | grep "Microcode patch level"
Microcode patch level: 0x7030106
`

miczyg1, 2019-03-21:
If anybody wants to build fresh v4.9.0.3 with microcode update, it is now possible. Fix for menuconfig has been applied.

https://pcengines.github.io

kolargol, 2019-03-21:
yes i confirm it works - i have uploaded bin's for testing. You can close this issue.

closed 2019-03-21 by miczyg1

soder10, 2019-07-11:
What is the drawback or consequence if the microcode is not updated via the BIOS (via coreboot binary), but rather updated via the OS / kernel during boot or after boot?

miczyg1, 2019-07-11:
@soder10 it depends on Your threat model. The earlier the microcode is updated the better. If You are afraid of attacks during boot time, then the consequences may be high. Let's say somebody has installed malicious service on Your system which executes before the microcode updater does its job. Then it could use the vulnerability to read personal data and leak it via network etc. 

You may also find my blog post interesting:
https://blog.3mdeb.com/2019/2019-05-29-spectre-and-meltdown-on-apu2/
Since the mitigation highly depends on the kernel mitigations implemented, the microcode seems to not influence much on the security. What is more, microcode enables only 1 security feature, while AMD stated that there are 3 features that can be enabled via microcode update (Mitigation V2-4: https://developer.amd.com/wp-content/resources/Managing-Speculation-on-AMD-Processors.pdf). But there is also the AMD retpoline which is told to mitigate the variant 2 (which microcode update is responsible for). Mitigation based only on the hardware/firmware level won't prevent the attacks.

soder10, 2019-07-11:
@miczyg1: well, I am not having any exotic threat model. I am more worried about the difference between early microcode update (before the kernel boots) vs  late microcode update (after the kernel has loaded). Others have posted earlier, that the results are not exactly the same in the 2 methods. Because the kernel does not dynamically re-detect for example the CPU flags after the boot, the output of dmesg differs. And god knows what else will behave differently as well.

miczyg1, 2019-07-12:
@soder10 the kernel is much more complicated than firmware and this is the problem here. 

I also would like to say that we have sent a request to AMD in order to encourage them to publish the newest microcode. Although it has been over a month and processing from AMD side is slow, we constantly push them to make up a decision. When the microcode will be public, we will include it natively in our builds.

soder10, 2019-07-12:
@miczyg1: Thanks very much to keep pushing AMD in this topic. Even if its very unlikely they change their minds.
What I dont really understand, why their stubborn secrecy and restriction? There are many public instances of their binary blob downloadable on the internet (e.g. freebsd port "devcpu-data" contains a combined intel+amd microcode binary blob).

miczyg1, 2019-07-12:
@soder10 on the other side linux-firmware has not the latest microcode present for family 16h. AFAIK FreeBSD grabs the same package as Linux:
https://reviews.freebsd.org/D15523
And the latest microcode there is: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/amd-ucode/microcode_amd_fam16h.bin?id=8ac569dd3ca3ca685bd47ee86c1eeb6050864db3
from year 2014... This is not the same binary as https://github.com/platomav/CPUMicrocodes/blob/master/AMD/cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin from 2018. As long as this binary is not officially pushed to Linux/FreeBSD by AMD or to coreboot, we have to wait for the decision.

gretel, 2019-07-12:
> @miczyg1: well, I am not having any exotic threat model. I am more worried about the difference between early microcode update (before the kernel boots) vs late microcode update (after the kernel [...]

personally, i would not like to let the firmware update application code. so i dislike letting application code update firmware, regardless of any threat model.

both approaches should render the same result but actually they do not. from my experience i feel something dangerous lurking there.

@soder10 having current microcode being part of the actual firmware would get closer to my requirements.

[11 more replies]