pcengines/apu2-documentation #79: Enabling 3rd and 4th serial ports via BIOS

Strykar, 2018-04-30:
Does the latest BIOS have the option to enable the other two serial ports yet? The legacy version had it.

pietrushnic, 2018-05-02:
@Strykar sure please familiarize yourself with [sortbootorder](https://github.com/pcengines/sortbootorder#theory-of-operation) which is payload added by default to all PC Engines firmware releases. Please report and problems you will face.

closed 2018-05-02 by pietrushnic

Strykar, 2018-05-02:
I apologize for the bother but I was unable to install (to test) until confirmation, the APU is remote to me. I will update and post any issues. Thank you.

pietrushnic, 2018-05-02:
@Strykar please be careful when you upgrade the firmware of the remote device. There is no warranty for that. We do our best, but there is not enough testing to cover all possible scenarios. You should have recovery method and physical access to a device or try that on spare hardware before proceeding.

Also if you are doing this on production then you should double check what you trying to do and recovery chip which cost couple EUR should be always ready to help you recover.

Strykar, 2018-05-02:
Absolutely! 👍 It's not in production, just a long way off from me, and I need the extra serial ports or it may as well be bricked. I'm aware about the BIOS recovery module, cheers.