pcengines/apu2-documentation #86: APU2 Serial number is all garbled on 4.6.8

Veldkornet, 2018-05-17:
I've finally managed to get onto 4.6.8 with pfSense 2.4.3-RELEASE-p1 on my APU2.

So far I have no problems, LED's are also working again.

Only in the pfSense interface, the serial number is all garbled...
[image]

Also from the command line:
```
/bin/kenv -q smbios.system.serial : ▒▒▒▒▒▒▒▒▒▒
```

I understand that this may be fixed in [4.6.9](https://github.com/pcengines/coreboot/blob/coreboot-4.6.x/CHANGELOG.md#v469---2018-05-11).

So I'll test that one and confirm.

pietrushnic, 2018-05-17:
@Veldkornet thank you for the separate issue

miczyg1, 2018-05-22:
@Veldkornet this has already been fixed in v4.6.9. The binaries will be available by the end of the day or tommorrow

Veldkornet, 2018-05-23:
Indeed it looks okay now.
Or rather, the command gives an actual output now:
/bin/kenv -q smbios.system.serial

It seems to not be shown in the pfSense interface if it's working.

closed 2018-05-23 by Veldkornet