pcengines/apu2-documentation #96: apu4_v4.6.9.rom gives motherboard mismatch on apu4.

michaelprice, 2018-06-05:
http://pcengines.ch/file/apu4_v4.6.9.rom.tar.gz gives a motherboard mismatch warning when attempting to flash to an apu4 currently running 4.6.4.

Is this a known issue and a force is ok or am I missing something important?

DennisEdlund, 2018-06-05:
I think there is a config miss in a string of the build of the apu4 that you already have in the board that gives a reply of **_apu2_** to the flashrom.
For me this worked:
**_flashrom -w apu4_v4.6.9.rom -p internal:boardmismatch=force_**

michaelprice, 2018-06-06:
Found some courage and went ahead and forced it. Went well. Used the sysutils/flashrom port in OpenBSD rather than booting tinycore linux.

DennisEdlund, 2018-06-07:
@pietrushnic Sorry to cut in but if there could be a info/doc page for APU2, APU3 & APU4 individually regarding specific info about that board and just a link on each page that pointed to a general page about common issues and solutions?

pietrushnic, 2018-06-07:
@DennisEdlund we think about couple things:

1. Feature set website with test results, known issues and tracking ids [here](https://github.com/pcengines/apu2-documentation/pull/90)
2. apu2/3/4/5 are not that specific to have separate pages I think, btw we have [pcengines.github.io](https://pcengines.github.io/) which we can extend

Any community input is welcome, so "cut in" is fine :)

michaelprice, 2018-06-07:
I would have noticed a comment on this issue were it on the wiki or the github.io page where the newer firmwares are linked.

Another thing I think would be helpful is a reference to the md5 checksums of the firmwares on the github.io page especially since the download links aren’t https.

DennisEdlund, 2018-06-07:
I was just thinking about this apu2-document subpage.
There would be ground for the setup of a apu3 & apu4 page, where you could get specific help for the boards or if there would be a **named** page for all boards.
Then you now **where** to address issues and find info for **your** board.

miczyg1, 2018-06-14:
@michaelprice the problem with board mismatch is related to wrong SMBIOS table entries. In v4.6.7 we have switched to short names i.e. `PC Engines apu4` -> `apu4` for partnumber entry. This is correct form which should be used. The information about it was published here: https://pcengines.github.io/
These wrong entries caused certain modules specific for apu not being loaded on LEDE or OpenWrt etc.
`flashrom` detects mainboard by reading the partnumber, so if it is different in the system and coreboot image, it yields warning.

miczyg1, 2019-06-11:
@piotr-kleins could You search through CHANGELOGs on coreboot repository and locate the version where the SMBIOS tables were fixed? Please create some small annotation to https://github.com/pcengines/apu2-documentation/blob/master/docs/firmware_flashing.md
which versions have to be forced, or more precisely which version transitions will require forcing.

piotr-kleins, 2019-06-11:
@miczyg1 Added a section `motherboard mismatch warning` to related document.

miczyg1, 2019-06-11:
Updated firmware flashing guide for the corner cases with different platform names in DMI tables:
https://github.com/pcengines/apu2-documentation/blob/master/docs/firmware_flashing.md

I assume we can close the issue now.

closed 2019-06-11 by miczyg1