pcengines/apu2-documentation #111: Legacy console redirection cannot be disabled?

pietrushnic, 2018-08-08:
Just found it [here](https://forum.ipfire.org/viewtopic.php?t=16032#p117861):

> I had serious problems with newer bios versions. The lastest working was legacy 4.0.14 because in 4.0.15-17 the option “Legacy Console Redirect” cannot be disabled.

pietrushnic, 2018-08-08:
I believe we need clear information in documentation how this option (and others) work.

miczyg1, 2018-09-19:
@pietrushnic according to CHANGELOG on legacy branch legacy console redirection could be broken. However it was tested since v4.0.16 and later releases and worked well. Fine, I will add JIRA item to describe apu features.

miczyg1, 2018-10-23:
https://github.com/pcengines/pcengines.github.io/pull/16