pcengines/apu2-documentation #112: AMD CCP and TPM as enropy sources

pietrushnic, 2018-08-09:
We should provide some documentation and tests to show how entropy sources can be used. We should take a look at entropy sources in light of firewall OSes, which may require entropy to provide various services.

This issue was motivated by discussion from [IPFire forum]()

silentcreek, 2020-07-19:
This would be interesting. As far as I understand, the CCP is not usable on the APU2. But if I'm not mistaken, a TPM module might be. So maybe the first thing to document would be which TPM module(s) work with the APU2 and where they can be sourced. I checked on PC Engines' website and they don't seem to offer a module. I found two places that offer a module for the APU2, both of which are out of stock, unfortunately. I found one store which has them in stock, but shipping costs would be higher than the costs of the module, so it's quite expensive...

bdd, 2020-07-20:
@silentcreek 
> I found two places that offer a module for the APU2, both of which are out of stock, unfortunately.

Maintainers of this repository @piotr-kleins and @miczyg1 work for 3mdeb. They do offer TPM modules and looks like they still have them in stock. https://shop.3mdeb.com/product/tpm2/

Because they ship from Poland, flat cost to California was a bit steep at €60, so I asked around among a couple coworkers and we made a group buy.

mkopec, 2022-02-16:
Looks like the CCP can be used as an entropy source on the APU2, not sure how much entropy it actually generates though:

```
cat /sys/devices/virtual/misc/hw_random/rng_available
ccp-1-rng
```

Also there's some related documentation at OpenWRT wiki: https://openwrt.org/toh/pcengines/apu2#cryptographic_hardware

This topic seems worth additional investigation

miczyg1, 2022-02-17:
I am not sure it is still possible. We have removed PSP Secure OS from PSP firmware a long time ago. It may not be functional anymore

silentcreek, 2022-02-21:
@mkopec I just checked on Debian 11, and while it says that ccp-1-rng is the "current" rng, /dev/hwrandom does not provide any entropy. If I check with hexdump, I only get "ffff" back. I also recall trying installing rngd which looks for /dev/hwrng and is supposed to test the quality of the entropy it gets during initialization. On my APU2 rngd fails to start complaining that no hardware rng/entropy source is available. So, I guess it's really not usable.

@miczyg1 That makes me wonder, though, whether the ccp kernel module should be blacklisted entirely on APU2 systems...

nicklowe, 2022-02-22:
> I am not sure it is still possible. We have removed PSP Secure OS from PSP firmware a long time ago. It may not be functional anymore

Why was it removed? It seems useful to have the CCP functions available if reasonably possible, at least an option to enable them.

nicklowe, 2022-02-22:
Seems from https://review.coreboot.org/c/coreboot/+/31075 but I am not sure it fully considered this use case.

miczyg1, 2022-02-22:
But I think it found the culprit quickly... It seems the RNG still works despite the firmware being stripped (which is good). For Linux the workaround is as follows:

```
# ensure ccp is the rng source
$ cat /sys/devices/virtual/misc/hw_random/rng_current 
ccp-1-rng
$ setpci -s 8.0 48.L=0x0000003d
# check if the value has been set in PCI and does not return 00000000
$ setpci -s 8.0 48.L
0000003d
# read from hwrng
$ dd if=/dev/hwrng count=1K iflag=count_bytes 2> /dev/null |xxd                                                                                                                               
00000000: bbfa 075e d6ea 4458 6984 f213 d3c3 d5c3  ...^..DXi.......
00000010: 3833 04ff ddc9 48ba 32db ac49 5054 e36a  83....H.2..IPT.j
00000020: 255d 66e5 b8ba a82e 35d9 8ca6 0fda cf07  %]f.....5.......
00000030: 0370 e1ca 5fd2 2e6e bb49 7447 0b1f 2b35  .p.._..n.ItG..+5
00000040: 1504 c3d1 28be 5786 3b6d ebbe 12ee 8328  ....(.W.;m.....(
00000050: 24b4 5f90 9739 547c af58 68ed 7083 19a5  $._..9T|.Xh.p...
00000060: 52a2 5c36 dbf6 23cc 8053 8c8c c18b cca9  R.\6..#..S......
00000070: 512d 2a2c e908 44fa 0664 69fe 73da f113  Q-*,..D..di.s...
00000080: d029 32a9 e61a 0a35 42b2 4ed2 9a8c 8a89  .)2....5B.N.....
00000090: f92e 553d 741f e415 385a f102 6b04 fdae  ..U=t...8Z..k...
000000a0: 3b61 4d3b cf49 5ee3 38be 0ab5 7dcd 84f3  ;aM;.I^.8...}...
000000b0: 14e1 9c30 367f 541c c92a 9bfb 08ce 4604  ...06.T..*....F.
000000c0: c3b9 a760 1d07 1b81 2339 6b67 47f3 53c7  ...`....#9kgG.S.
000000d0: 8361 7559 a790 9259 20ba 83ed 38c7 1088  .auY...Y ...8...
000000e0: ae9b 9511 7e7e 910e d95c 1cce d0e5 d4db  ....~~...\......
000000f0: 0d23 b2bb 7feb 1253 db21 d85a 2207 1420  .#.....S.!.Z".. 
00000100: 8ad6 2281 1b10 1132 fda4 5a1e 0937 d317  .."....2..Z..7..
00000110: 77eb cd9d 1557 2d2d beef 650c f3e1 83a1  w....W--..e.....
00000120: 832a a9c5 45df 7d43 6606 7808 b72e db9e  .*..E.}Cf.x.....
00000130: 653b 5578 4798 2610 2f4b dbc5 2ec5 8056  e;UxG.&./K.....V
00000140: 1de5 80b6 a0a6 f8e5 e88f ab63 d69a bdd5  ...........c....
00000150: 19bf 0604 1447 6212 68ad b033 5282 b117  .....Gb.h..3R...
00000160: 68fd 3a19 32b3 c6c4 7ea7 d37b 5d4a 568a  h.:.2...~..{]JV.
00000170: 4af8 a72c c774 1dc2 0bf8 1c86 9225 5705  J..,.t.......%W.
00000180: eebe f474 b4b6 2c29 be90 b85e e142 58ef  ...t..,)...^.BX.
00000190: c36f 5933 4fcc 969c 3712 5448 1c7b 559a  .oY3O...7.TH.{U.
000001a0: 8098 6cbe 6636 8890 f681 455a 6f7e 0b95  ..l.f6....EZo~..
000001b0: fa92 7afe 6595 3ab9 dca3 2aea eab2 bc8b  ..z.e.:...*.....
000001c0: dbf5 3efa d328 5b04 2b51 9a9f 8521 3da5  ..>..([.+Q...!=.
000001d0: 2a94 1c3f 51e2 f75b d3d1 da6c 7ca0 6edc  *..?Q..[...l|.n.
000001e0: 2ebb 6db2 5cfe 5c8e 45b4 9cb0 a5c4 cee4  ..m.\.\.E.......
000001f0: c1da 5ab0 d544 c082 e8e0 bcdc 5e47 3b28  ..Z..D......^G;(
00000200: 5e7e a816 4801 f1e9 d01a 72f1 dd50 1b2e  ^~..H.....r..P..
00000210: 9a56 c64c c17a eb4a e37d f28e 5d13 09f6  .V.L.z.J.}..]...
00000220: 1e2b 5ba7 1ec3 b3b6 7cff bfb4 1fba 0613  .+[.....|.......
00000230: 2dbf 29af dd0d d85b 4658 eb4b 7a38 2840  -.)....[FX.Kz8(@
00000240: 9a00 599b 1a53 0c79 da31 48bc abdf 45f7  ..Y..S.y.1H...E.
00000250: cd70 bfcb 0ce8 c946 e8f0 cf4b 0d8d e35e  .p.....F...K...^
00000260: 7cba 09c5 7249 a057 4ca0 db31 aba1 5293  |...rI.WL..1..R.
00000270: d51a 8efb 288d b013 3de8 89c9 dd98 4fb7  ....(...=.....O.
00000280: c5ac 82c2 5e76 b3c2 57c8 fe3e 394e 11c0  ....^v..W..>9N..
00000290: 0da7 d0ff 140c 0c94 7903 f468 8b2d 69fc  ........y..h.-i.
000002a0: 9b73 6026 3fe0 d6f2 8068 9514 7b04 2f9c  .s`&?....h..{./.
000002b0: 7f89 c966 8378 bb52 28a3 0c6d c429 dbaa  ...f.x.R(..m.)..
000002c0: fe99 613e 0a02 b4af 4d18 05de 862d ad1f  ..a>....M....-..
000002d0: 2805 ce52 82b5 7dd9 56d3 c853 2973 b2b9  (..R..}.V..S)s..
000002e0: 8e66 fdb1 aae3 12ff 3c1e 042f 1fff 682d  .f......<../..h-
000002f0: 1d45 1072 873a 3883 2d91 c36d 176a c488  .E.r.:8.-..m.j..
00000300: 5d3d 5bc9 0775 7ddb ff28 8265 13d9 8158  ]=[..u}..(.e...X
00000310: a4d0 04c2 8442 773d f3fe 2b4e 7383 6a41  .....Bw=..+Ns.jA
00000320: 7dbc 9a1f feb7 66cc 7d82 4502 4aef f80e  }.....f.}.E.J...
00000330: 9924 92b4 895d 405e 75f8 a8d2 cfe5 7e6b  .$...]@^u.....~k
00000340: 06d2 9347 de6e 3068 0508 93b6 2da8 8d46  ...G.n0h....-..F
00000350: d0f9 b770 f8d5 d24c cfcd d1d3 5c72 dac4  ...p...L....\r..
00000360: c672 99c8 9081 3f72 7a4c ad08 b406 6707  .r....?rzL....g.
00000370: b370 7527 5d08 d235 c55c 84ed 47ea 1434  .pu']..5.\..G..4
00000380: e13f 77ac f062 728c 7e4b 4d2a 3290 3ea5  .?w..br.~KM*2.>.
00000390: 99e0 a79d 13cc e9f0 7c06 7450 0ef2 9a74  ........|.tP...t
000003a0: 44e4 a4ee 1ba1 8c14 d62e 0d5c 729c afa6  D..........\r...
000003b0: 4c4a 54bc d9f1 c9fe 6213 b6e6 f931 33a2  LJT.....b....13.
000003c0: f647 958b a0e8 cbf8 9580 d4f7 34bc adc3  .G..........4...
000003d0: c614 1e75 5993 34a6 c109 949b 37dd d0fd  ...uY.4.....7...
000003e0: c3c0 7080 cd8a b454 1bb7 445f da5c c84f  ..p....T..D_.\.O
000003f0: dc4a 686e 349d ad27 b3bf 70d4 30f9 27cb  .Jhn4..'..p.0.'.
```
Happy "entropying". In BSD `pciconf` can probably give the same results although I don't know what is the RNG devnode there,

We can issue a fix in our firmware releases for it.

pietrushnic, 2022-02-22:
@miczyg1 wow, those are great news. It took us almost 4 years. This fix has great impact on whole firewall/VPN ecosystem and should be correctly highlighted in documentation. I think our validation also should pick it up, so maybe some *BSD and Linux distros  can leverage that and squeeze more from the platform.

pietrushnic, 2022-02-22:
@bdd buzzkill is needed when facts say so. Definitely I was happy to see action after almost 4 years, but as you wrote this is not so big thing. Speaking about quality of this RNG I'm not an expert, but reading `ent` documentation seem that Chi square distribution is quite good for AMD CCP in comparison to `urandom`. For firewall/VPN probably the only thing here would be CPU offloading which can give small improvement for that old SoC in comparison to devices not using hwrng.

silentcreek, 2022-02-22:
@miczyg1 I can confirm, with your workaround rngd starts just fine now and seems to accept /dev/hwrng as a valid entropy source. Well done. Just out of curiosity, what does your setpci workaround actually change?

nicklowe, 2022-02-22:
I think this is fantastic as a good source of high quality entropy. Well done, and thank you.

miczyg1, 2022-02-23:
@silentcreek I would need to find public references to tell you what it does specifically. It comes from confidential BKDG. Here you may find partial information what the register contains (but not all bits are defined):  https://github.com/coreboot/coreboot/blob/master/src/soc/amd/stoneyridge/include/soc/southbridge.h#L149

mkopec, 2022-03-31:
Closing as resolved by https://github.com/pcengines/coreboot/pull/505

closed 2022-03-31 by mkopec