pcengines/apu2-documentation #117: [4.8.0.4] APU2 Release Date: Thu Jan 1 1970

Veldkornet, 2018-09-05:
After upgrading from 4.8.0.3 to 4.8.0.4, the BIOS release date does not sow the correct date (or someone forgot to add it ;) )

[image]

pietrushnic, 2018-09-05:
@Veldkornet thanks for reporting that. We will add this to regression testing and make sure it will never happen again.

@krystian-hebel @miczyg1 FYI

pietrushnic, 2018-09-05:
@Veldkornet what tool you use to display data?

Veldkornet, 2018-09-05:
It’s just the pfSense gui, which issues the following command:

‘/bin/kenv -q smbios.bios.reldate’

Strangely, if I do this I get the following:

‘20183108’

So it looks like it is set... but I’m not sure if maybe the format changed? Unfortunately I don’t have another system to check. 

I can say that pfSense then puts it together as follows:

<?=gettext("Release Date: ");?><strong><?= date("D M j Y ",strtotime($biosdate[0]));?></strong><br/>

krystian-hebel, 2018-09-05:
@Veldkornet are you sure that it showed correct date in 4.8.0.3? We did change the format, but it was between 4.8.0.2 and 4.8.0.3. Now it should be consistent with mainline coreboot.

pietrushnic, 2018-09-05:
@Veldkornet IMO this is a bug. SMBIOS tables should comply to the particular format. We implemented that incorrectly because of assumtion that sign of life will have the same date format as what we have in `Release Date` in SMBIOS. Many thanks for pointing it out.

[image]

miczyg1, 2018-10-05:
This PR should have fixed the issue: https://github.com/pcengines/coreboot/pull/197

miczyg1, 2018-10-22:
@Veldkornet v4.8.0.5 is released and the bug is fixed. Please have a try: https://pcengines.github.io/#mr-15
We will fix it for legacy release too this month.

Veldkornet, 2018-10-24:
Looks good

```
/bin/kenv -q smbios.bios.reldate
10/04/2018
```

[image]