pcengines/apu2-documentation #120: APU1 goes straight to OS output in serial console [4.8.0.4]

40b0a3cd, 2018-10-07:
When flashing an APU1 with 4.8.0.4, no F12 menu is displayed, and there is no apparent way to go into BIOS configuration.

The CMOS pin was shorted to clear any problematic settings, and I also pushed S1 during boot. Flashing with an older BIOS verison (one that I backed up) restored configuration functionality.

40b0a3cd, 2018-10-07:
@pietrushnic Thanks for the quick response.

That's a typo - I also have an apu2d4 that works great, and I believe F12 is the key to see the boot options. I tried replicating this on the apu1 since I never got a prompt. The problem board is an apu1d4, and here's what the console outputs on boot:

`            PC Engines apu1
coreboot build 20183108
BIOS version v4.8.0.3
_<pause>_
>> OpenBSD/amd64 BOOT 3.34
boot>`

Also, when I was referring to the CMOS reset, I meant shorting the J5 jumper as referenced here: https://pcengines.ch/pdf/apu1.pdf

The issue is that both an older firmware on the apu1, and 4.8.0.4 firmware on the apu2 pause to allow the user to modify bios options.

One last thing to note is that the BIOS version displays as "4.8.0.3" even though I flashed with the "apu1_v4.8.0.4.rom" file. I'm not sure if this makes any difference. The flash was done twice just to verify I hadn't gone crazy.

pietrushnic, 2018-10-07:
@40b0a3cd IIRC problem with version is know, @miczyg1 please confirm.

miczyg1, 2018-10-08:
@40b0a3cd the bug with version is known. I have also had problems with serial console on apu1 and I am aware that 4.8.0.4 causes this. 4.8.0.5 will not have this issue.

miczyg1, 2018-10-22:
@40b0a3cd have You got any problems with apu1 4.8.0.5? 4.8.0.5 has been already released:
https://pcengines.github.io/#mr-15

40b0a3cd, 2018-10-23:
@miczyg1 My apu1 worked as expected on 4.8.0.5. I was able to get into the BIOS/boot menus as expected now. Thanks!

closed 2018-10-23 by 40b0a3cd