pcengines/apu2-documentation #122: [4.8.0.5] Enabling ECC on PC Engines platforms

Veldkornet, 2018-10-24:
I noted in the release notes of 4.8.0.5 the following:
> ECC exclusion range, ECC now works properly. For details see also [3mdeb blog post](https://3mdeb.com/firmware/enabling-ecc-on-pc-engines-platforms/) about enabling ECC on PC Engines apu platforms.

However, I've had a look at that blog post, and if I'm not mistaken, it only explains how to test it in MemTest86, not specifically how to enable it.

So I just wanted to check, is there something specific that I need to do to enable ECC as suggested in the release notes? Or should it just work in the latest firmware?

pietrushnic, 2018-10-24:
@Veldkornet maybe the message is not clear, but this works by default since v4.8.0.5. Previously it didn't work because of exclusion range was set incorrectly. More to that there was no reliable way to validate that ECC works.

@krystian-hebel have you got anything to add?

krystian-hebel, 2018-10-24:
@Veldkornet it should work OOTB, we find no reason to not use ECC when it is available. However we may think about disabling ECC Injection at some point, if such a need arises.

Veldkornet, 2018-10-24:
Okay, I thought as much.

From the release notes, it sounded as if you need to follow the guide to enable it. But if it should just be working, then no worries ;)

soder10, 2018-12-12:
Is there any chance this ECC feature will be back-ported to legacy?

pietrushnic, 2018-12-13:
@miczyg1 @krystian-hebel I believe this is a valid request. Any possible obstacles for that?

miczyg1, 2019-02-26:
ECC has been enabled on legacy too in [v4.0.23](https://github.com/pcengines/coreboot/blob/coreboot-4.0.x/CHANGELOG.md#fixed)

closed 2019-02-26 by miczyg1