pcengines/apu2-documentation #124: Forum maintenance long overdue

doktornotor, 2018-11-03:
For lack of a better way, filing this as an issue here. Guys, I know this is apparently not your priority, your hardware is great, but the forum is unusable. Either clean it up and archive it, or replace the zombie software with something that's not spammers' paradise.

E.g., 16/20 threads on the APU forum is spam. Not a productive communication platform for anyone. 👎 

[image]

doktornotor, 2018-11-03:
Another user complaint [here](http://pcengines.info/forums/?page=post&id=2A0C534A-8A16-4264-9DA1-C5C77D3DC7D3&fid=DF5ACB70-99C4-4C61-AFA6-4C0E0DB05B2A) - of course completely lost in the heaps of spam.

pietrushnic, 2018-11-03:
@doktornotor I'm really sorry about your experience. TBH I had the same feeling and tried to change that. Please note that most people operating this GitHub account are not PC Engines employees.

Key issues with the transition to a new forum:
1. There is no way to export from the current engine
2. We choose WoltLab as next forum engine and set it up here: http://forum.anschlag.ch/ but DNS was not yet redirected

I will reach PC Engines and ask how and if we can proceed.

doktornotor, 2018-11-04:
>Please note that most people operating this GitHub account are not PC Engines employees.

Yes, pretty much what I suspected... just really did not find a better place to post since the forum is definitely unusable now.  Would be nice to have a usable place to discuss things with other users of the HW. 

>I will reach PC Engines and ask how and if we can proceed.

Thanks for relaying the issue to where it belongs.

soder10, 2019-02-05:
3 months pass, and clearly zero change. Owners of pcengines dont care as long as you people still buy their product, and they have revenue. People vote with your wallet next time!

soder10, 2020-09-22:
Another 1,5 year, and the pcengines.info is still a cesspool :)

seamusdemora, 2020-09-23:
Not to excuse the dismal state of the forum, but in my experience PC Engines still delivers good technical service.

pietrushnic, 2020-09-27:
@soder10 @seamusdemora we are really sorry about that, but 3mdeb cannot do anything with that. We tried and we stuck. What you can do is sent complaints to PC Engines directly. @michaelsteinmann please let me know if you have a better way to solve the problem.

soder10, 2020-09-28:
@pietrushnic Dont get me wrong, my post is not pointing towards 3mdeb (we have clarified the difference between the 2 companies approx. 2 yrs ago), but thats the only place where there is activity happening, and somebody reading and responding to complaints about that OTHER company. I mean even much smaller companies or organizations can have a proper website with a forum to collect feedback from their products. If pcengines cannot handle a forum portal, why dont they move to facebook at least? The pcengines.info site is a half-dead spamcollector, and if people post their complaints on that site, pcengines clearly doesnt give a crap.
If the official pcengines forum is like that, I wonder how their "paid" support works really? Or is there any paid support option from that OTHER company at all? Are they hiding behind their distributors for a reason? They literally have 1 single product, assembled in 3-4 different PCB (but technically those are all like 99% similar, so not really 3-4 really different product), they outsourced the software development to 3mdeb, and they cannot even run a proper portal to handle community support. Thats the terrible truth about their existence.

pietrushnic, 2020-09-28:
@soder10, of course, I have to act in defense of PC Engines. They did for Open Source Firmware more than most companies using coreboot and there is no second so small company with such commitment to openness (schematics, open-source firmware etc.). I see a couple of options:

- we switch to form on anschlag domain mentioned above
- we trying to contact PC Engines support email and convince the CEO to improve the forum communication, we can even employ resellers since they would have a better impact
- we move communication on GitHub (e.g. here)
- we choose another platform e.g. reddit
- we add forum to https://pcengines.github.io :thinking: I wonder if it is even possible

The key problem I see with alternatives to the official forum is to convince and inform the community that there are alternatives.

doktornotor, 2020-09-30:
> @soder10 @seamusdemora we are really sorry about that, but 3mdeb cannot do anything with that. We tried and we stuck. What you can do is sent complaints to PC Engines directly.

Do you have some direct **working** contact that you are able to share? (I mean. a real person in charge, not a dead mailbox flooded with spam that noone is reading.)

Thanks.

pietrushnic, 2020-09-30:
@doktornotor I know that PC Engines team read `support<at>pcengines<dot>ch`

dpieski, 2020-12-23:
It looks like the `pcengines.info/forums` website is down (or currently not working).

Another option @pietrushnic  than those you listed is to enable the new "Discussions" feature in this repo. That essentially makes a forum in the repo without cluttering up Issues.

Strykar, 2020-12-24:
@pietrushnic TBH, as long as the forum is usable and easy to migrate and manage, I don't really really care which one is used. But I do wish to know if you guys evaluated [Discourse](https://www.discourse.org/)?

Strykar, 2020-12-24:
> Do you have some direct **working** contact that you are able to share? (I mean, a real person in charge, not a dead mailbox flooded with spam that noone is reading.)
> 
`support@pcenginesdotch`

michaelsteinmann, 2020-12-24:
Thanks for letting us know ...

pietrushnic, 2021-06-23:
I hope with this announcement everyone would be happy and we can close this issue: https://forum.pcengines.dev/

We will market new forum soon.

closed 2021-06-23 by pietrushnic

soder10, 2021-06-24:
@pietrushnic just please make sure it does not become the same spam collector forum as the old one!

pietrushnic, 2021-06-24:
@soder10 we will try our best. At least we have admin privileges here and more modern engine.