pcengines/apu2-documentation #125: BIOS write protect option can't be enabled

gnustomp, 2018-11-12:
On apu2d, the `Enable BIOS write protect` option can't be enabled. It is always "Currently Disabled" after selecting the option. This occurs regardless of whether the WP pins are shorted or not.

Tested on mainline 4.8.0.5 and legacy 4.0.20.

pietrushnic, 2018-11-12:
@miczyg1 are there any regression tests related to that?

miczyg1, 2018-11-12:
@pietrushnic we do not have regression for it, because it requires shorting WP pin in order to verify the feature. One would have to remove the jumper after successful verification manually, so that platform can be flashed again later.

@gnustomp I have been testing this feature on v4.8.0.5 for sure. And did not see any issues. Just put a jumper on WP pin and then lock the SPI in sortbootorder. In my case it showed `Enabled` state after saving changes and booting to setup again. I also could not flash any other BIOS with flashrom. It should yield `Block protection could not be disabled!` and output erase errors.

gnustomp, 2018-11-12:
It looks like in `sortbootorder/sortbootorder.c`, `spi_wp_toggle` is never set to `1`/`true`. I am not sure why.

The SPI chip is a W25Q64.V according to flashrom.

miczyg1, 2018-11-12:
@gnustomp since I could not reproduce Your issue, please provide which platform are You using? 
The SPI chip part is correct.

The protection is implemented in such way, that enabling the protection causes all Block Protection bits to be set (protect whole SPI content) and additionally set SRP0 bit. Please before setting the protection in sortbootorder, place the jumper on pins 1-2 on J2(apu2) or J3(apu3/4). It will ensure that once write protection is set, nothing will change that state midway.

gnustomp, 2018-11-12:
This is an apu2d2. I have tried using the jumper before setting the option.

I think the code for setting status registers for Winbond chips is incorrect. From the datasheet, I can only tell there are 2 status registers, and only 1 "Write Status Register" command, but the code has 3 status registers with individual write commands.

Interestingly, I managed to get the board into a state where it is no longer able to write the SPI chip regardless of whether the WP jumper is short or not. (erase failed every time)

`sortbootorder` still thinks WP is off, and from what I can tell SRP0 and BP[0-2] are 0.

I have a Raspberry Pi that I can use to try to fix this - is it safe to apply external 3.3V to the SPI chip while on the board, or do I need a creative solution for powering the chip?

miczyg1, 2018-11-12:
@gnustomp note that the code for Winbond chips was ported from coreboot with intention to support all Winbond chips. If there are parts that have 3 status registers, it should be handled. W25Q64.V has only 2 status registers, but it does not affect the process, especially that it works for me perfectly and cannot reproduce Your problem on 3 different platforms (tested with apu2/3/4 on v4.8.0.5). 

Are You using also the mainline firmware version?

miczyg1, 2018-11-12:
I have looked on v4.0.20 legacy and I am experiencing what You described. However mainline is not affected. We will fix it in v4.0.22 then.

gnustomp, 2018-11-13:
The CMP bit somehow set, so I fixed that. I can enable SRP0 and BP[0-2] with an external programmer, and with the WP jumper it will prevent writing or changing status registers with the external programmer.

However, when I boot coreboot, BP[0-2] are unset, even if WP is short. SRP0 is still set, so status registers can't be changed, so WP can't be enabled in `setbootorder`. If I unset SRP0 externally, on the next boot, SRP0 is set again. Again, WP doesn't change this.

If SRP0 is set but WP is not short, attempting to enable WP option sets WEL and BUSY. I'm not sure why SRP0 is always set when coreboot boots. This may be a side effect of the following issue.

I modified `sortbootorder` to use the correct command for the W25Q64FV, and now it works. The `0x01` "Write Status Register" command expects two bytes for the values of SR1 and SR2, but `sortbootorder` only gives one byte. I think this is what caused an unexpected value to be written to SR2 earlier.

What is the factory-default value of the SR-2 registers? On this board they are currently `0x3b`.
In particular, what should LB[1-3] be?

I note that the W25Q128FV has 3 Status Registers with individual write commands. This isn't compatible with W25Q64FV.

gnustomp, 2018-11-13:
This is strange, as the SPI chip has W25Q64JV printed on it...

miczyg1, 2018-11-13:
@gnustomp 

1. When I was checking the issue yesterday, I was not able to check which part was soldered on the board, so probably I had different chips on these 3 platforms. Given that some platforms work well, some not. 

2. You are right, is is not mentioned directly, but writing SR1 and SR2 is possible with WRSR instruction on JV, fortunately. 

3. PC Engines firmware is open source. Feel free to submit PR. Contribution is welcome.

4. Datasheet does not say anything about HW WP, thus You won't get BIOS write protect setting persistent across boots. After power down the SRL bit will be cleared.

5. Yes this is some inconsistency. By looking on the datasheets it is not possible to implement unified procedure for those 2 chips, because JV part has no SRP0. So the function `is_flash_locked` can not possibly return real state of flash lock for both parts.

I have contacted board vendor regarding SPI chips, hope to get explanation about SPI chip differences.

Regarding LB bits, these are ont-time-programmable, so reverting them to factory state is not possible.

gnustomp, 2018-11-13:
Thanks for inquiring about this.

With my changes, HW write protection (SRP0) with the jumper is working as expected on the \*JV chip, even though this is undocumented. Writes are not allowed while the header is short, and are allowed if it is open. This is non-volatile and persists after power down.

I think this function is retained on \*JV for backwards compatibility, but undocumented. The datasheet does say, regarding the WRSR command:
>The W25Q64JV is also backward compatible to Winbond’s previous generations of serial flash memories, in which the Status Register-1&2 can be written using a single “Write Status Register-1 (01h)” command

I think it's possible to support at least all W25Q64\*V chips by using SR[12] and the WRSR command only. Since the SRL bit isn't used, we can ignore SR3. This may also apply to other W25 chips too.

I'll do a bit more testing with the board I have, and I'll send a PR to sortbootorder.

miczyg1, 2018-11-13:
@gnustomp that is great! Really appreciate Your effort. Looking forward to see the PR then.

miczyg1, 2018-11-22:
Ok i have tested the change and it is working. Already merged and will be released with v4.8.0.7 and v4.0.22.

closed 2018-11-22 by miczyg1