pcengines/apu2-documentation #130: Compex WLE600VX PCIe not detect with v4.9.0.1

ShinjiLE, 2019-02-07:
With V4.9.0.1 the card is not detected anymore.
Neither at boot nor with lspci the card is shown now . The previous version (V4.8.0.7) works well  .

the output of lspci and V4.8.0.7 and kernel 'Linux carter 4.19.0-2-amd64 #1 SMP Debian 4.19.16-1 (2019-01-17) x86_64 GNU/Linux' looks like this:

05:00.0 Network controller: Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter
        Flags: bus master, fast devsel, latency 0, IRQ 66
        Memory at f7600000 (64-bit, non-prefetchable) [size=2M]
        Expansion ROM at f7800000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 2
        Capabilities: [50] MSI: Enable+ Count=1/8 Maskable+ 64bit-
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [140] Virtual Channel
        Capabilities: [160] Device Serial Number 00-00-00-00-00-00-00-00
        Kernel driver in use: ath10k_pci
        Kernel modules: ath10k_pci

pietrushnic, 2019-02-07:
@ShinjiLE deinitely looks like regression probably introduced by some changes in mainline coreboot. @miczyg1 do we have WLE600VX? Ideally we should bisect the issue, if we can't do that on our side, then maybe let's produce set of bisection binaries and send to @ShinjiLE for testing.

miczyg1, 2019-02-07:
@pietrushnic We have only WLE200NX and WLE900VX. 
During regression tests apu2 did not detect WLE900VX on v4.9.0.1, which may be the case with WLE600VX too. But on v4.9.0.2, which will be released soon, the test passed for WLE900VX. Hope it will work with v4.9.0.2. We will try to investigate the issue with v4.9.0.1 @ShinjiLE

ShinjiLE, 2019-02-14:
Yes in V4.9.0.2 is it fixed.
Thx

closed 2019-02-14 by ShinjiLE