pcengines/apu2-documentation #134: APU3 - Internal USB headers not detected

rickvanderzwet, 2019-02-20:
Upgrading the firmware on a APU3 board did reveal an interesting regression, devices connected to the internal USB2 headers on the APU3 are not detected by the OS only. The last known working firmware is v4.0.23.

I have tested apu3_v4.6.10 & apu3_v4.8.0.7 and apu3_v4.9.0.1 and all failed. None of these firmware versions detects devices connected to the internal USB headers. System is running Linux 4.14.94+ #0 SMP Thu Aug 16 07:51:15 2018 x86_64 GNU/Linux.

Method to reproduce:
  - Upgrade firmware to version described.
  - Connect USB device to internal USB header
  - Boot system
  - Watch output of `lsusb` of `dmesg` 

I would love to write and TDBF report, how-ever I am unable to find out where test cases for 'regression testing' are coded, thus unfamiliar with the coding style of such test cases. If some pointers could be given I will spend time and effort in making it an official regression test case.

pietrushnic, 2019-02-20:
@rickvanderzwet we have to try to reproduce that on our side.  Our sets are written in Robot Framework, but at this point, we don't want to open source that. We can reveal a detailed description of procedures, but implementation was mostly made by us without financial involvement of our client, the code is generic across many platforms and we look to monetize it as Open Source Firmware regression test suite or similar product.

Anyway if we confirm this but we would be glad to work on the fix and add required regression testing.

miczyg1, 2019-02-21:
@rickvanderzwet have You enabled EHCI0 in sortbootorder?

rickvanderzwet, 2019-02-21:
@miczyg1 Not sure on EHCI0 in sortbootorder, I have to check. I will  have access to the hardware next monday again. The BIOS settings are default.

@pietrushnic thanks for the details. I was triggered by [Bug Fix Development](https://github.com/pcengines/apu2-documentation/blob/master/README.md#featurebug-fix-developmentl) section, hence my question. Still wondering if there is an reference image available, which I should use for bug reporting.

pietrushnic, 2019-02-21:
@rickvanderzwet no problem, this is more for maintaining team since no other has access to test development environment. You are right that reference image would be great - @miczyg1 that's very cool idea:

1. Boot bug reporting OS from 3mdeb https server - we could have there automated bug reporting including e.g. cbmem console and paltform hardware configuration dump
2. the same as 1 but boot from image flashed to USB stick

miczyg1, 2019-02-26:
@rickvanderzwet have You tried the sortbootorder EHCI0 option? Did it help?

rickvanderzwet, 2019-03-04:
> @rickvanderzwet have You tried the sortbootorder EHCI0 option? Did it help?

Thanks for the suggestion. I have enabled EHCI0 and flashed image apu3_v4.9.0.2.rom. This will make the internally connected devices visible again. 

Since flashing the BIOS will clear all BIOS settings, it will cause my units not to be reachable except from physical serial access are there methods available to set BIOS options (again) via CLI as well?

miczyg1, 2019-03-04:
@rickvanderzwet I hope You have preserved the correct order of operations:

flash apu3_v4.9.0.2.rom -> enable EHCI0 in sortbootorder.

Otherwise, the setting will be wiped out.

The units will be reachable via serial and SSH if You have configured so. The only issue here may be the iPXE. If You are booting Your platform via iPXE, it won't be reachable, because iPXE is not enabled by default. Whole set of sortbootorder [options](https://github.com/pcengines/sortbootorder/blob/master/README.md#settings-description)
and their [default values](https://github.com/pcengines/sortbootorder/blob/master/README.md#default-settings)

rickvanderzwet, 2019-03-04:
> @rickvanderzwet I hope You have preserved the correct order of operations:
>
> flash apu3_v4.9.0.2.rom -> enable EHCI0 in sortbootorder.

I am not sure how-to perform the "enable EHCI0 in sortbootorder" over the CLI at all, hence my question.

> Otherwise, the setting will be wiped out.

I surprises me a little to found out BIOS settings being wiped after an BIOS upgrade, which also breaks an remote upgrade flow for example. In my case I do not have physical access the units, as such settings BIOS settings by connecting an serial cable it not possible. 

Could "enable EHCI0 in sortbootorder." and other BIOS settings also be set via CLI tooling like flashrom or only via serial BIOS access?

miczyg1, 2019-03-04:
We are working on a partial image upgrade, to avoid settings wipeout.

Serial BIOS access is the only official way. These are BIOS settings and shouldn't be modified outside BIOS. We can change the default value of EHCI0 to enabled in v4.9.0.3 if it satisfies Your needs.

rickvanderzwet, 2019-03-04:
> We can change the default value of EHCI0 to enabled in v4.9.0.3 if it satisfies Your needs.

Works fine for me at apu3 board now, how-ever it was disabled with good reasons I guess, so it might introduce issues on other places...

pietrushnic, 2019-03-04:
@miczyg1 please do not change default values in firmware. What is needed is either:
1. Tool (flashrom?) to modify binary before flashing to contain required values, so nobody will stuck updating remotely
2. Partial update to now overwrite boot order and other configuration values, is this fmap feature?

@rickvanderzwet we are aware of this limitation and agree that clean up of configurations bad thing. We working on fixing that, I'm not sure if we can address it in this cycle, but by the next release, it should be there.

miczyg1, 2019-03-04:
@pietrushnic I was thinking about 2nd solution. But I guess the first one will be needed too.

miczyg1, 2019-03-21:
The issue with not persistent runtime configuration across firmware udpates will be tracked here:
https://github.com/pcengines/coreboot/issues/211
We will update the statuses there and inform about work on binary modification tool. I am closing the issue for now as the problem has been solved.

closed 2019-03-21 by miczyg1