pcengines/apu2-documentation #145: Microcode patching fails on APU2

j75, 2019-05-06:
At the 3rd step (`Build v4.8.0.7 image`) it spits out the following message:
```
./build.sh release v4.8.0.7 apu2
Cloning into 'release/coreboot'...
remote: Counting objects: 375, done
...
Build coreboot for apu2
mkdir: cannot create directory 'build': Permission denied
...
mkdir: cannot create directory 'build': Permission denied
Error: Expected config file (/home/coreboot/coreboot/.config) not present.
Please specify a config file or run 'make menuconfig' to
generate a new config file.
Makefile:135: recipe for target 'real-all' failed
```
At the 5th step:
```
./build.sh dev-build `pwd`/release/coreboot apu2 menuconfig

Dev-build coreboot mainline
Build custom coreboot for apu2
mkdir: cannot create directory 'build': Permission denied
mkdir: cannot create directory 'build': Permission denied
mkdir: cannot create directory 'build': Permission denied
Assembler messages:
Fatal error: can't create build/util/kconfig/mconf.o: No such file or directory
/home/coreboot/coreboot/util/kconfig/Makefile:349: recipe for target 'build/util/kconfig/mconf.o' failed
make: *** [build/util/kconfig/mconf.o] Error 1
```

What should be done to solve these issues?

Firefishy, 2019-05-06:
Seem likely a local permission issue. Did you maybe git clone using sudo?

j75, 2019-05-07:
No, I am working as a regular user (but with admin rights).
BTW should I use `v4.8.0.7` (as stated in the documentation) or a more recent one? (`v9....`) ?

j75, 2019-05-07:
I think I know what's happening: the docker image uses a `coreboot` user, and I am **not** `coreboot`!
However, this could be overcame with the docker option `-u root` => and now we have another error:
```
...
SUCCESS: Wrote 182 bytes to file 'build/fmap.fmap' (and generated header)
The sections containing CBFSes are: COREBOOT
    CC         bootblock/lib/cbfs.o
    CC         bootblock/lib/cbmem_console.o
    CC         bootblock/lib/delay.o
    CC         bootblock/lib/fmap.o
    CC         bootblock/lib/gcc.o
    CC         bootblock/lib/halt.o
    CC         bootblock/lib/hexdump.o
    CC         bootblock/lib/libgcc.o
    CC         bootblock/lib/memchr.o
    CC         bootblock/lib/memcmp.o
    CC         bootblock/lib/prog_loaders.o
    CC         bootblock/lib/prog_ops.o
    CP         bootblock/lib/program.ld
    CC         bootblock/lib/reset.o
    CC         bootblock/lib/timestamp.o
    CC         bootblock/lib/version.o
    CC         bootblock/southbridge/intel/common/reset.o
    LINK       cbfs/fallback/bootblock.debug
build/bootblock/drivers/uart/uart8250io.o: In function `uart_init':
/home/coreboot/coreboot/src/drivers/uart/uart8250io.c:113: undefined reference to `check_com2'
src/arch/x86/Makefile.inc:120: recipe for target 'build/cbfs/fallback/bootblock.debug' failed
make: *** [build/cbfs/fallback/bootblock.debug] Error 1
```

j75, 2019-05-07:
BTW the `build.sh` script doesn't make happy **shellcheck**:
```
In build.sh line 52:
    if [ $major -ge 4 ]; then
         ^-- SC2086: Double quote to prevent globbing and word splitting.

In build.sh line 53:
        if [ $minor -eq 0 -a $patch -le 16 ]; then
             ^-- SC2086: Double quote to prevent globbing and word splitting.
                          ^-- SC2166: Prefer [ p ] && [ q ] as [ p -a q ] is not well defined.
                             ^-- SC2086: Double quote to prevent globbing and word splitting.
...
```

pietrushnic, 2019-05-07:
@j75 @Firefishy thank you for being interested in our work. There are some corner cases not handled correctly - definitely, we have to fix that. Please note that Debian, toolchain, and coreboot as continuously moving targets do not simplify things. Recently we faced a situation where some combination of coreboot-sdk and coreboot code do not work together.

@miczyg1 @krystian-hebel if you can elaborate and help with above issues it would be great.

miczyg1, 2019-05-08:
The version does not matter. The version stated in the documentation was chosen as the first compatible version for microcode patching. Any newer version is also ok and should work.

Regarding permissions, the user should be in the docker group. The containers create a coreboot user with UID:GID 1000:1000. So if You user on the host does not match these IDs, these errors are likely to happen. This was done to avoid using root user in the container and creation of files owned by root.

@j75 if YOu think the code quality is not the best, fell free to submit patches.

@Firefishy first of all, the mkdir problems with creatign build directory should not be ignored. That means coreboot did nto even start to build, because it had no place to put build binaries. The latter problems from switching to root user may be related to the failed build from earlier steps. Switching user like that most likely broke whole coreboot directory. I advise starting from cloning coreboot again. `make distclean` may also be useful, but I do not guarantee it will clean everything as it should in this case. Also do not switch versions and do not build on previously built coreboot. Only recompilation of the same version/revision is allowed.  After switching version `make distclean` is required. When You change boards `make distclean` is also required. The error with `check_com2` looks to me like permission issues or version switching problem.

Checkout the CI to see how the build log should look alike:
https://gitlab.com/pcengines/coreboot/-/jobs/189781915

It should be accessible publicly. Link to Gitlab CI is also placed on pcengines.github.io

j75, 2019-05-09:
>Regarding permissions, the user should be in the docker group.

I am.

>I advise starting from cloning coreboot again. 

Ok, I shall try to build **coreboot** outside docker.

krystian-hebel, 2019-05-09:
> Ok, I shall try to build **coreboot** outside docker.

Please don't. It will take a couple of hours just to compile crosscompilers, assuming that all required packages are even available in the host system. What you can try is to clone the repository from within the docker container.

miczyg1, 2019-06-11:
@j75 have Your problem been solved? If not, can we assist You still somehow?

j75, 2019-08-10:
No, but I left it aside for a while... I shall try to look again next week.
Anyhow, I've checked and I am in the docker group:
```
uid=1602(...),....,136(docker)...
```

krystian-hebel, 2019-08-11:
@j75 right, but uid is 1602. One solution is to change the owner of `coreboot` directory and its subdirectories (root permissions required):

```
chown -R 1000 coreboot
```

More elegant solution would be to make `build` directory and change only its owner/permissions. This way git can still be used on host system. Unfortunately, payloads are built inside their directories, so those need write access too.

Perhaps there is a way to remap uids of bind-mounted directories when starting Docker, but I'm not aware of it. There is `-u` option in `docker run`, it _may_ work, but I have no way of testing it at the moment for full build with different host uid. Try to add `-u 1602` to all of the `docker run` lines in `build.sh`, before image names (`pcengines/pce-fw-builder*`).

Out of curiosity, what is Your host system and how many users does it have? That uid seems a bit strange to me.

j75, 2019-08-11:
>That uid seems a bit strange to me.

Not to me! I've decided once to start my users ids from 1600 and that's all! ;-)
I think I have tried with the `-u` option and it didn't work, but I can't remember why - anyhow, soon I'll start working on that again.

j75, 2019-08-20:
OK, I managed to build the firmware (or it seems so!)
The only salient modification was adding ` -u root` to `docker run` calls (I have also added a `set -e` at the beginning of the build script because I like to see the script stopped as soon as an error is encountered).
There were also some other strange error messages during various building phases (not in the `docker run` but in cloning various modules, I may give you further details if you are interested in)
but in the end the result was OK.

However, I still have one more question: according to the doc from https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md the last step is
>The firmware image can be found in release/coreboot/build

What is actually the name of this firmware file? is it *release/coreboot/build/coreboot.rom* ?

pietrushnic, 2019-08-20:
@j75 yes, exactly it is `coreboot.rom`

j75, 2019-08-26:
Anyhow, on my APU2 system I noticed that performing `flashrom -r...` produces messages like
> Found Winbond flash chip "W25Q64.V" (8192 kB, SPI).
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).

Would it be possible to flash it though ?

I also noticed on https://pcengines.github.io/ that there are newer firmware (the recommended being the v4.10.0.x.  series) do they have the Meltdown patch ?
Also, when trying to rebuild the firmware, although abiding strictly on the documentation (`./build.sh release v4.8.0.7 apu2`) I noticed that there are newer versions that may be used (v4.9.0.7 for instance) do you still recommend **v4.8.0.7**?

Many thanks in advance!

miczyg1, 2019-08-26:
@j75 the flashrom message is related to a flag that describes the chip features:
https://github.com/flashrom/flashrom/blob/v1.1-rc1/flashchips.c#L15602
The FEATURE_OTP indicates the presence of security registers that are programmable via different instruction set and can also be locked permanently. These registers do not contain any firmware related code or data in PC Engines firmware or official coreboot, but it is possible to write to these registers if they are not locked.

We do not include the microcode patches yet. Still pinging AMD for permission to publicly include the newest microcode.

The documentation shows only arbitrary version which can be used to patch the microcode. The v4.8.0.7 is the first one supporting the patching IIRC. You are free to use any newer version.

j75, 2019-08-27:
OK, so I finally flashed the new firmware, I rebooted.... and now I have

> $ dmesg | grep microcode
> [    7.602890] microcode: CPU0: patch_level=0x07030106
> [    7.607830] microcode: CPU1: patch_level=0x07030106
> ...

I tried to the same operations but specifying a **v4.9.0.7** release and it fails just after the last build:
```
$  ./build.sh dev-build $PWD/release/coreboot apu2 CPUS=$(nproc)
...
rm bin/8086157b.rom.zbin bin/8086157b.rom.bin bin/version.8086157b.rom.o bin/8086157b.rom.zinfo
make[2]: Leaving directory '/home/coreboot/coreboot/payloads/external/iPXE/ipxe/src'
Created CBFS (capacity = 8388056 bytes)
    CBFS       apu/amdfw
    CBFS       AGESA
    CBFS       fallback/romstage
    CBFS       bootorder
    CBFS       fallback/ramstage
    CBFS       config
    CBFS       revision
    CBFS       spd.bin
    CBFS       fallback/dsdt.aml
    CBFS       fallback/payload
    CBFS       payload_config
    CBFS       payload_revision
    CBFS       bootorder_map
    CBFS       bootorder_def
    CBFS       etc/boot-menu-key
    CBFS       etc/boot-menu-wait
    CBFS       etc/boot-menu-message
    CBFS       img/memtest
    CBFS       img/setup
    CBFS       genroms/pxe.rom
    CBFS       coreboot.rom
    SeaBIOS    Add sercon-port file
make: *** [Makefile.inc:1079: build/coreboot.rom] Error 1
```
Should I use another version or do something else?

miczyg1, 2019-08-27:
The builder does not support switching between versions. You should build a release version of v4.9.0.7 and then dev build with changed config file.

j75, 2019-08-27:
I made myself misuderstood - of course I started with
`./build.sh release v4.9.0.7 apu2`
then I copied cpu00730F01_ver07030106_2018-02-09_88EDFAA0.bin in **release/coreboot/**, then I started `./build.sh dev-build $PWD/release/coreboot apu2 menuconfig  ` and finally `./build.sh dev-build $PWD/release/coreboot apu2 CPUS=$(nproc)` => here I have encountered the error
> make: *** [Makefile.inc:1079: build/coreboot.rom] Error 1

which is very strange because there is no other error before!

miczyg1, 2019-08-28:
Okay, this is a bug in a v4.9.0.7 then. As an alternative You may also specify different option to load microcode. Choose to load external microcode binaries instead of experimental option of AGESA (the option is in the same place) . Then pass the microcode blob file name as usual.

I got some patches merged upstream that let loading microcode without experimental path and are included in v4.9.0.7 already.

j75, 2019-08-28:
Yes! It worked! Now when booting I have the message
> coreboot build 20190907

I suppose it is correct (although we are still in August 2019... but the previous version displayed `coreboot build 20180312`) and, more important, I have the **dmesg** message `microcode: CPU0: patch_level=0x07030106`

closed 2019-08-28 by j75

miczyg1, 2019-08-28:
coreboot builds are timeless in order to be reproducible. The build date is taken from the last commit of the current branch you are building from. It is expected to have the binary built in the "past".

j75, 2019-08-28:
Never mind - the important thing is that it works!
Anyhow, thank you very much for your help!