pcengines/apu2-documentation #146: Installing pfSense does not work by provided guide on apu2d4

cytopia, 2019-05-10:
I was following the [pfSense-install-guide.md](https://github.com/pcengines/apu2-documentation/blob/master/docs/pfSense-install-guide.md).

* USB Boot stick is prepared with `pfSense-CE-memstick-serial-2.4.4-RELEASE-p1-amd64.img`
* Booting up [apu2d4](https://pcengines.ch/apu2d4.htm)
* Connecting to it via Putty
* Selecting `USB MSC Crive JetFlash Transcent 2GB 8.07`
* Message appears: `Booting from Hard Disk...`
* Nothing happens:

[image]

What am I missing here that is not shown in the guide?

## Settings

### Version information
```
PCEngines apu2
coreboot build 20170228
4080 MB ECC DRAM
SeaBIOS (version rel-1.10.0.1)
```

### Putty settings
```
Serial device: /dev/ttyUSB0
Flow Control:  None
Speed:         115200
Data bits:     8
Stop bits:     1
```

pietrushnic, 2019-05-13:
@cytopia thank you for reporting this issue. @miczyg1 who can handle that?

miczyg1, 2019-06-11:
@artur-rs could You verify whether there are any problems with the installation guide?

miczyg1, 2019-06-11:
@cytopia I suspect that You are using very old firmware version (judging by displayed SeaBIOS version). The guide should work with SeaBIOS version at least 1.11.0.x. Please consider updating the firmware, because the older version You have, the more trouble with debugging and problem-solving we have. The build system as wel as components have evolved significantly and brought support of many features. Our guides mostly refer to the recent firmware releases (up to 1 year old approximately).

miczyg1, 2019-06-11:
@cytopia we have checked the procedure on our side and we have no such issues. What we tried:

- burned the same pfSense image on USB stick
- booted installer on apu2_v4.9.0.6 and apu2_v4.0.7 (which I assume You are using according to build date in the sign-of-life string)
- tested platform apu2c4 (we do not have apu2d4 yet)

The only differences were : USB stick , apu2c4 instead of apu2d4 and minicom instead of putty. The installer should work without any modifications in BIOS. We have successfully launched the installer on both firmware versions. I suspect there might be a problem with:
- putty on BSD (rather unlikely)
- USB transmission problem (also unlikely since it is USB 2.0 stick You have used I assume)
- incorrectly burned installer image (have You tried booting the pfSense on other machines with the same stick?)

NanoCaiordo, 2019-07-07:
I'd strongly suggest you to update APU firmware as I believe you are using the one that came shipped with.

A part fixes, compatibility and improvements new firmwares switched SATA controller mode from IDE to AHCI mode which is required for your SSD drive, you might also have more chances to get the USB 3.0 port working.

After, and only after,  you update your firmware have a look at #154

cytopia, 2019-11-04:
Thanks for the support!

closed 2019-11-04 by cytopia