pcengines/apu2-documentation #156: apu2d4 cannot be powered off when Huawei ME909s-120 is installed

jbzld2, 2019-07-23:
With an apu2d4 board setup with Debian Buster (10.0), Linux 4.19.0-5-amd64 and firmware v4.9.0.6 , the board cannot be powered off with a command such as `shutdown -h now` when a Huawei ME909s-120 LTE modem is installed. When the system is shut down, the leftmost LED switches off but then the 3 LEDs blink back on after approximately 10 secs.

When I remove the modem from mPCIe 2 , that same shutdown command works as it should.

It is expected that this APU will be a backup router and will be powered on by WoL only when needed.

Any idea on what is going on?

krystian-hebel, 2019-07-25:
I believe @miczyg1 noticed something similar

miczyg1, 2019-07-25:
@jbzld2 have a look at https://pcengines.ch/schema/apu2c.pdf page 8 J13 (mPCIe2 connector).
There is a PE_WAKE signal (active low) which is connected to pin 1 to the module. If the module grounds this pin, the PE_WAKE will boot the platform up again after poweroff. The same mechanism is used on wakeonlan. The NIC emits low signal on PE_WAKE which results in platform bring up. If YOu have any means to check the pin 1 state, verify please it is 3.3V there (high). This is the most probable scenario I am thinking of now.

charles-martel55, 2021-11-16:
Hello,

I have the same issue with a Quectel EP06. Is it possible to add an option to disable this pin in BIOS ?

metux, 2023-02-21:
> Hello,
> 
> I have the same issue with a Quectel EP06. I tested the pin, the module ground it all the time.
> 
> Is it possible to add an option to disable this pin in BIOS ?

If somebody really understand the deep black magic of the pinmux, there *might* be some chance
(*if* this pin can be configured by the pinmux and one finds out the right control bit).

When writing the linux apu board driver, I kept myself away from that, since I couldn't find any *reliable* information on that specific SoC.