pcengines/apu2-documentation #161: Issue with apuled.ko

pfsenseuser, 2019-09-06:
I made a test on a spare Apu2 and the apuled.ko is loading but has no effect :-( ( in /dev/led/ are only the igb0 and igb1 the led1, led2, led3 dont appear anymore)
Flashing back the bios 4.0.27 and everything is fine. I guess in Bios v4.10.0.0 its the "enable basic ACPI support for GPIOs" that kills the apuled function ...

miczyg1, 2019-09-10:
@pfsenseuser temporary solution for now: https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md#known-issue

pfsenseuser, 2019-09-10:
> @pfsenseuser temporary solution for now: https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md#known-issue

Thank you very much!

miczyg1, 2020-01-07:
Duplicate of https://github.com/pcengines/coreboot/issues/329
Closing.

closed 2020-01-07 by miczyg1