pcengines/apu2-documentation #164: Missing signature file for v4.10.0.1 firmware

drduh, 2019-09-13:
https://3mdeb.com/open-source-firmware/pcengines/apu4/apu4_v4.10.0.1.SHA256.sig
https://3mdeb.com/open-source-firmware/pcengines/apu3/apu3_v4.10.0.1.SHA256.sig

pietrushnic, 2019-09-14:
@drduh yes, we know about this problem. It looks like not a lot of people use signatures verification since it's been a while from publication date.

We have to generate new keys for 4.10 release.

miczyg1, 2019-09-15:
@drduh Yes, it was mentioned in the release blog post available on the release site. The process if enrolling new signing key is in progress and signatures, should appear soon.

taiwei0, 2019-09-18:
If possible, could you please publish key fingerprint on GitHub or other place for easy verification.

taiwei0, 2019-09-18:
> If possible, could you please publish key fingerprint on GitHub or other place for easy verification.

Sorry, I had not seen that this can be found here:

https://github.com/3mdeb/3mdeb-secpack/tree/master/canaries/pcengines

miczyg1, 2019-09-18:
Signatures are available now and special note has been added to the release page: https://github.com/pcengines/pcengines.github.io/pull/49

closed 2019-09-18 by miczyg1