pcengines/apu2-documentation #168: Temperatures no longer work in 4.10.0.3

Veldkornet, 2019-11-14:
After updating to 4.10.0.3, it seems as though the temperature monitors no longer work.

In pfSense, it would usually be shown in the GUI. Or from SSH witch` sysctl -a | grep temperature`.
But now, it's just blank.

pietrushnic, 2019-11-14:
@Veldkornet interesting regression, @artur-rs ideally if we would add test for that.

kevindd992002, 2019-11-14:
I have the same issue, btw. I have another box at 4.10.0.2 that still had the temps shown.

pfsenseuser, 2019-11-15:
> 
> 
> I have the same issue, btw. I have another box at 4.10.0.2 that still had the temps shown.

I can confirm this too! :-)

pietrushnic, 2019-11-15:
@Veldkornet @kevindd992002 @pfsenseuser we appreciate pfSense community engagement and that you care about firmware. We will look into that as soon as possible, but definitely I cannot promise it would be earlier than 10/12/2019.

miczyg1, 2019-12-10:
Fixed in v4.11.0.1. coreboot community changed the definitions of processors in ACPI to be compliant with ACPI specification, so now pfSense/OPNSense APCI compliance is under a big question mark. We will investigate the topic further, for now we kept the non-compliant ACPI definitions of processors.

closed 2019-12-10 by miczyg1