pcengines/apu2-documentation #173: doc: How to find current firmware version and update from freebsd?

fwolfst, 2020-04-18:
Lately our apu2x4 doesn't deliver what I expect. I believe the problem is on the other side, however the issue put me in update-rage and the last thing left would be a firmware update of the board.
I struggle to find out which firmware I am currently running on (also https://pcengines.ch/howto.htm#bios doesnt mention any trick to do so). I guess `dmidecode` is the tool of choice (available both in Linux and FreeBSD)? If so, this could be mentioned in the documentation.

Also, as far as I understand, I can install `flashrom` in a running pfSense and do the update live, right? Of course I have to reboot after that. If that is true, a hint on how to install flashrom on FreeBSD/pfSense would be great. And also a note if its really a good/bad idea.

PS: I am really happy with the products (and honestly **love** the pcengine homepage) and advertise them wherever I can.

fwolfst, 2020-04-18:
I guess the freebds / flashrom-part it boils down to https://teklager.se/en/knowledge-base/apu-bios-upgrade/ .

It is unlikely that I will time find to craft a PR with doc updates.

miczyg1, 2020-04-18:
@fwolfst no problem, we can take care of that.

miczyg1, 2020-07-15:
@fwolfst we have updated the documentation, I hope it helps: https://github.com/pcengines/apu2-documentation/blob/master/docs/firmware_flashing.md

closed 2020-07-15 by miczyg1

fwolfst, 2020-07-15:
Nice, thanks! I am no power user (like, flashing APUs is not my day-to-day job) but will notice next time I have to lay my hands on the baby.

TheDom42, 2020-07-16:
I have a related question, running OPNsense on an APU.2E4.
I'm still on the BIOS flashed when the board was delivered (v4.0.29) and I would like to switch from legacy to mainline (if that is possible).
Does the upgrade process described under https://github.com/pcengines/apu2-documentation/blob/master/docs/firmware_flashing.md take into consideration the legacy/mainline path? Do I get to choose which version from the mainline path is flashed or is it just using the most recent?
Is it even possible to switch over to mainline from a legacy build using `flashrom` or would it be safer to use an image built using the image builder? If so: would that mean I would need to reinstall OPNsense?

I'm in a comparable situation as @fwolfst (i.e. no power user when flashing APU boards) and I would like to do the process as safe as possible.

Thanks in advance and thank you for building such a great product!

miczyg1, 2020-07-16:
@TheDom42 migrating from legacy to mainline is safe unless you are flashing mainline older than v4.9.0.4. For newer versions the update should be seamless.

The binaries we provide on https://pcengines.github.io are tested and verified, so there is no particular need to build one yourself. Typically the most recent version is the best choice. There is no need to reinstall the operating system. What you may experience is for example:
https://github.com/pcengines/apu2-documentation/issues/175 - then you have to disable the core performance boost
https://github.com/pcengines/coreboot/issues/329 - then follow instructions for workaround https://github.com/pcengines/apu2-documentation/blob/master/docs/gpios.md#known-issues

The mainline firmware is much more featureful. If you need such functionalities as TPM 2.0, IOMMU etc. it is advised to switch to  mainline, because we do not currently plan to backport everything to legacy.

TheDom42, 2020-07-16:
Thank you for the swift response. I will have a look if and when I have the time to afford the downtime/troubleshooting capacity as this is the main FW/Router for my home network, so bricking it or having constant reboots like the ones mentioned in #175 are not really optimal (btw. will the #175 bug be fixed in an upcoming firmware release or is that something too OS-specific or hardware related?)