pcengines/apu2-documentation #175: Kernel panic APU2C4

sikysikov, 2020-05-04:
Hi, I am getting kernel panic "spin lock held too long" after firmware update from 4.0.7 to v4.11.0.5 or v4.0.30 on APU2C4 running pfSense 2.4.5. On 4.0.7 all running fine, after update I get 2-3 reboots per day. Do I need to do any steps before update? Any help?

miczyg1, 2020-05-04:
@sikysikov hi, have you followed https://github.com/pcengines/apu2-documentation/blob/master/docs/firmware_flashing.md#corebootrom-flashing ?
Did you try power cycle or at least cold reset the platform as described in above link?

sikysikov, 2020-05-04:
Hi, nope, just the warm reboot. Sorry I missed this guide. Will try to flash the latest mainline, proceed with cold reboot and come back here to confirm. Thanks.

sikysikov, 2020-05-06:
I am getting system hangs or reboots without any error log on v4.11.0.6. Now testing v4.0.30.

sikysikov, 2020-05-10:
Still getting reboots on v4.0.30 and even on v4.0.25. Reverting back to v4.0.9 which seems to be rock solid so far.

miczyg1, 2020-05-11:
@sikysikov what is the operating system you use? Have you any dumps from these panics?

sikysikov, 2020-05-12:
Hi, I am running pfSense 2.4.5. At first update to v4.11.0.6 without power cycle I got some error reports but after re-flash with power cycle I got just reboots or system hangs without errors. I do not know which files of error report of pfSense you would need, see the end of msgbuf.txt (but please keep in mind that this report was after flashing without power cycle):
`spin lock 0xffffffff82ca2f40 (smp rendezvous) held by 0xfffff8000cc35620 (tid 100134) too long
timeout stopping cpus
panic: spin lock held too long
cpuid = 1
KDB: enter: panic`

miczyg1, 2020-05-15:
@sikysikov could you please try disabling Core Performance Boost and test the uptime?

sikysikov, 2020-05-21:
@miczyg1 any guide how can I do that? Thanks

miczyg1, 2020-05-22:
@sikysikov just enter Payload [setup] on boot and disable Core Performance Boost option:
https://github.com/pcengines/sortbootorder#example-menu-view

miczyg1, 2020-06-05:
@sikysikov any updates?

sikysikov, 2020-06-15:
@miczyg1 sorry I did not have any chance to get to the device yet. I will keep you updated.

sikysikov, 2020-06-19:
@miczyg1 I finally updated device to latest mainline v4.12.0.1 and got reboot right after few hours. I disabled Core Performance Boost then and it is running fine for 21hours now. I see Core Performance is disabled due higher CPU temperature but syctl still showing this:

`dev.cpu.0.temperature: 67.0C
dev.cpu.0.cx_method: C1/hlt C2/io
dev.cpu.0.cx_usage_counters: 56812286 0
dev.cpu.0.cx_usage: 100.00% 0.00% last 923us
dev.cpu.0.cx_lowest: C1
dev.cpu.0.cx_supported: C1/1/0 C2/2/400
dev.cpu.0.freq_levels: 1400/-1 1200/-1 1000/-1
dev.cpu.0.freq: 1400
dev.cpu.0.%parent: acpi0
dev.cpu.0.%pnpinfo: _HID=none _UID=0
dev.cpu.0.%location: handle=\_PR_.P000
dev.cpu.0.%driver: cpu
dev.cpu.0.%desc: ACPI CPU`

miczyg1, 2020-06-19:
@krystian-hebel any comments on that?

krystian-hebel, 2020-06-19:
```
dev.cpu.0.freq_levels: 1400/-1 1200/-1 1000/-1
dev.cpu.0.freq: 1400
```

It looks like whatever code obtains this info does not account for possible boost levels, and a different numbering between hardware and software P-states. If that is the case, the kernel panic is somewhat expected, as the kernel probably believes it works at 1400 MHz while in fact it works at only 1000 MHz, and such assumptions may be used for calculating TSC frequency. If at some point it waits for a device using a different clock source there is a possibility of an unexpected timeout happening.

@sikysikov could you check if "stable" coreboot version also reports 1.4 GHz here?

This may still be caused by some other issue, like IOMMU, PCIe power saving features, P-states of different cores or non-optimal interrupt routing. The first two can be toggled in sortbootorder so it may be a good idea to test it, too. Remember that those settings are reset to defaults after flashing a different coreboot version.

sikysikov, 2020-06-23:
Hi @krystian-hebel , did you mean to use legacy version by mentioning "stable" coreboot?

miczyg1, 2020-06-23:
If the BSD system reports boosted frequencies that may mean:
```
hint.p4tcc.0.disabled=1
hint.acpi_throttle.0.disabled=1
hint.acpi_perf.0.disabled=1
```
is configured as documented here: https://github.com/pcengines/apu2-documentation/blob/master/docs/apu_CPU_boost.md#bsd

sikysikov, 2020-06-23:
@miczyg1 I added that to boot/loader.conf as mentioned right after flashing. Should I remove that with Core Performance Boost disabled?

krystian-hebel, 2020-06-23:
@sikysikov I meant the version that didn't result in kernel panics so 4.0.9 or older

sikysikov, 2020-06-23:
> @miczyg1 I added that to boot/loader.conf as mentioned right after flashing. Should I remove that with Core Performance Boost disabled?

Removing "hint.." from boot/loader.conf I got

`dev.cpu.0.temperature: 67.1C
dev.cpu.0.cx_method: C1/hlt C2/io
dev.cpu.0.cx_usage_counters: 182761 0
dev.cpu.0.cx_usage: 100.00% 0.00% last 1041us
dev.cpu.0.cx_lowest: C1
dev.cpu.0.cx_supported: C1/1/0 C2/2/400
dev.cpu.0.freq_levels: 1000/882 800/724 600/543
dev.cpu.0.freq: 1000
dev.cpu.0.%parent: acpi0
dev.cpu.0.%pnpinfo: _HID=none _UID=0
dev.cpu.0.%location: handle=\_PR_.P000
dev.cpu.0.%driver: cpu
dev.cpu.0.%desc: ACPI CPU`

miczyg1, 2020-07-06:
@sikysikov now it looks correctly. The reported frequencies are non-boosted 1000, 800, 600. If it is running well now and you consider it resolved, please close the issue.

sikysikov, 2020-07-20:
@miczyg1 it is running well without CPU boost. Should I take that as final solution? I guess so as nobody else has such problem.

miczyg1, 2020-08-03:
For your particular issue, yes. If CPU boost would work well for everybody, we wouldn't make it runtime configurable.

closed 2020-08-03 by miczyg1