pcengines/apu2-documentation #176: After APU2 BIOS update from v4.9.0.4 to v4.12.0.1, memory speed droped x2

vespian, 2020-06-14:
I just upgraded from v4.9.0.4 to v4.12.0.1 and the dmidecode shows that memory now works at just 667 MT/s instead of 1333 MT/s. Why?

I have attached full outputs of dmidecode, before and after the  upgarde.

Thanks in advance for any information !

[dmidecode.post-update.txt](https://github.com/pcengines/apu2-documentation/files/4776487/dmidecode.post-update.txt)
[dmidecode.pre-update.txt](https://github.com/pcengines/apu2-documentation/files/4776488/dmidecode.pre-update.txt)

pfsenseuser, 2020-06-16:
I have a APU2C0 board running also bios v4.12.0.1 (pfsense) and i saw too that dmidecode says now 667 instead of 1333 but I think that is only a fault in the measuring algorithm because I couldn't observe a decrease of the overall performance of the system. (half speed should make a BIG difference). 
Just my opinion, I maybe wrong ...

miczyg1, 2020-06-17:
The DMI tables are constructed based on the information from AMD AGESA. It may be incorrectly constructed by AGESA itself. But it doesn't necessarily mean that the memory speed is halfed. Do you experience performance drop?

pfsenseuser, 2020-06-17:
Hello,

I made a test on a Apu2C0 Board once with bios v4.12.0.1 (dmidecode: Speed: 667 MT/s) and once with bios v4.0.30 (dmidecode: Speed: 1333 MT/s) and I cannot see any performance decrease. I used memtest86 for the test.
(Sorry cannot upload screenshots it says: 
"Something went really wrong, and we can’t process that file.")

miczyg1, 2020-06-29:
Please have a look at section 7.18.4 of SMBIOS specification: https://www.dmtf.org/sites/default/files/standards/documents/DSP0134_3.1.1.pdf

It looks like I did a mistake in v4.12.0.1 when creating the memory tables using AGESA provided. In v4.9.0.4 I have put the total speed of the memory (as per newer specification than 3.0.0) which is the memory clock speed (666MHz) multiplied by 2 (as per DDR - double data rate). Multiplied by 2, because DDR memory issues two transitions per clock cycle.

To sum up, value of 666MT/s reported by AGESA complies to SMBIOS specification 3.0.0 and older. I will prepare a fix for that.

miczyg1, 2020-08-04:
The issue has been fixed in v4.12.0.3

closed 2020-08-04 by miczyg1