pcengines/apu2-documentation #177: USB3 does not survive soft reboot

vespian, 2020-06-14:
This is an issue I had since a long time already. I have a USB dongle attached to the board, but once I issue `reboot` underlinux, system will not detect it again. Turning the board off and on helps to resolve situation. In dmesg I can find, compared to healthy boot:
```diff
 xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 2
-xhci_hcd 0000:00:10.0: hcc params 0x014040c3 hci version 0x100 quirks 0x0000000000000410
-usb usb2: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.03
-usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
-usb usb2: Product: xHCI Host Controller
-usb usb2: Manufacturer: Linux 5.3.18dfs.2 xhci-hcd
-usb usb2: SerialNumber: 0000:00:10.0
-hub 2-0:1.0: USB hub found
-hub 2-0:1.0: 2 ports detected
-xhci_hcd 0000:00:10.0: xHCI Host Controller
-xhci_hcd 0000:00:10.0: new USB bus registered, assigned bus number 3
-xhci_hcd 0000:00:10.0: Host supports USB 3.0 SuperSpeed
-usb usb3: We don't know the algorithms for LPM for this host, disabling LPM.
-usb usb3: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.03
-usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
-usb usb3: Product: xHCI Host Controller
-usb usb3: Manufacturer: Linux 5.3.18dfs.2 xhci-hcd
-usb usb3: SerialNumber: 0000:00:10.0
-hub 3-0:1.0: USB hub found
-hub 3-0:1.0: 2 ports detected
+xhci_hcd 0000:00:10.0: Host halt failed, -110
+xhci_hcd 0000:00:10.0: can't setup: -110
+xhci_hcd 0000:00:10.0: USB bus 2 deregistered
+xhci_hcd 0000:00:10.0: init 0000:00:10.0 fail, -110
+xhci_hcd: probe of 0000:00:10.0 failed with error -110
```

This issue is independent of the BIOS version and kernel version - I expirience it since ~3 years already.

miczyg1, 2020-06-16:
USB3 was always causing problems on this platform. We are also fighting it for years. But it is like 
looking for a needle in a haystack. This is more like hardware bug in the xHCI controller.

We have multiple USB detection tests in our regression suite, which are based on Linux. We do 50 cycles of detection after power cycle, warm boot and reboot. And yes there are problems sometimes in SeaBIOS, but Linux worked quite well all this time.

Could you please provide more information about the configuration? Which Linux distro, which apu, which USB dongle?

vespian, 2020-06-16:
Hi! 

Thanks for a reply. In the attached archive you will find the verbose output of lspci, lsusb, dmesg and dmidecode. The problem persisted on Centos7(kernel 3.10), Ubuntu 19.10(kernel 5.3) and Ubuntu 20.04(my current distro, kernel 5.4). The board is apu2c4.

Also, I wanted to say that I am very happy with the board and will be basing my future DIY routers on the PCengines boards. Awesome work!

[debug_info.zip](https://github.com/pcengines/apu2-documentation/files/4788658/debug_info.zip)

ukleinek, 2021-06-25:
I encounter this on two different APUs, too. The details are:

 - Debian 10.10, kernel 4.19.181-1, APU2+4GB
 - Debian 11/testing, kernel 5.10.40-1, APU2+4GB

A workaround is to boot with no USB devices connected and then after boot plug them in one after another.

In both cases there are several HUBs with several devices (usb-UART-adapters, thumbdrives ...)

`lsusb -t` looks as follows on the Debian 10 machine:

<pre>
$ lsusb -t
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 14, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 7, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 4: Dev 12, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
        |__ Port 4: Dev 12, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
        |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 10, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 4: Dev 19, If 0, Class=Vendor Specific Class, Driver=, 480M
            |__ Port 2: Dev 15, If 1, Class=Vendor Specific Class, Driver=, 480M
            |__ Port 2: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 480M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 6, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 17, If 0, Class=Hub, Driver=hub/1p, 480M
                |__ Port 1: Dev 22, If 0, Class=Mass Storage, Driver=usb-storage, 480M
            |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 2: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 16, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 2: Dev 20, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 3: Dev 24, If 0, Class=Hub, Driver=hub/3p, 480M
                |__ Port 1: Dev 28, If 0, Class=Vendor Specific Class, Driver=, 12M
            |__ Port 4: Dev 26, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
        |__ Port 3: Dev 13, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 18, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 4: Dev 116, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 23, If 1, Class=Vendor Specific Class, Driver=, 480M
            |__ Port 2: Dev 23, If 0, Class=Vendor Specific Class, Driver=, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 10, If 0, Class=Hub, Driver=hub/3p, 480M
                |__ Port 1: Dev 11, If 0, Class=Vendor Specific Class, Driver=, 12M
                |__ Port 3: Dev 12, If 0, Class=Mass Storage, Driver=usb-storage, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 5: Dev 4, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 6: Dev 5, If 0, Class=Hub, Driver=hub/1p, 480M
            |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 7: Dev 6, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
</pre>