pcengines/apu2-documentation #181: APU2D4 (and others?) Linux 1.4ghz boost?

anonymous-one, 2020-07-25:
Its been quite some time since the 1.4ghz boost was added.

As of July 2020, we are unable to check if we are even boosting to this frequency in Linux.

Could we get this sorted? And if it at all possible, add 1.4ghz to the list of possible scaling freqnecies (eg: so we can force the APU2 to run at 1.4ghz 24/7)?

I am currently running the performance scaling governor and am cannot tell if by doing so I am giving up the 1.4ghz boost or not.

miczyg1, 2020-08-04:
You may check that by running benchmark on a single core, using dd command or memtest payload with and without boost. This is the only way to detect whether the boost is working. This is all we have:

https://github.com/pcengines/apu2-documentation/blob/master/docs/apu_CPU_boost.md
https://blog.3mdeb.com/2019/2019-02-14-enabling-cpb-on-pcengines-apu2/

Also the CPU documentation states that boosted processor states must be hidden from operating system. Given that we do not look for more proofs.

netstx, 2020-08-31:
Is the documentation as far as enabling CPU Boost in *BSD (I assume the sysctl variables are from FreeBSD) still up to date?

miczyg1, 2020-09-01:
@netstx the CPU boost is enabled for all operating systems because it is operating system agnostic. YUou may disable CPU boost only via setup menu in sortbootorder. The sysctl vars should be up to date. Note these variables do not have to be changed. The CPU boost will work either way. It is only needed to read the 1400MHz value out of the system. Except that it is almost useless.

CPU boost works on apu2 series platforms, that is apu2*, apu3*, apu4* and apu5*.

netstx, 2020-09-01:
@miczyg1 Ok, I understand now. Thanks for the detailed response!

netstx, 2021-04-15:
I can confirm that OpenBSD detects the same cpu frequency of 998MHz with or without CPU boost enabled.