pcengines/apu2-documentation #188: Change default settings

jailbird777, 2020-12-15:
Since persistent configuration between version upgrades isn't done yet, I'm looking at using pce-fw-builder to build my normal settings into the ROM itself.  That saves me from hauling a laptop and serial cable to each unit when I update them.

It would also be nice to document maybe how to do this, sort of like the COM2 document.

Is the process as simple as:

$ ./build.sh release v4.... apu2
$ vi release/coreboot/src/mainboard/pcengines/apu2/bootorder_def
[ Flip the '0' or '1' after an option depending on if you want it off or on]
$ ./build.sh dev-build $PWD/release/coreboot apu2

Or is there more to it then just editing the bootorder_def?

Thanks!

miczyg1, 2020-12-15:
@jailbird777 in order to get your setting working you should (also) edit `release/coreboot/src/mainboard/pcengines/apu2/variants/apu2/bootorder`. `bootorder_def` holds default values to which the sortbootorder setup payload will reset setting when pressed `R`.

jailbird777, 2020-12-27:
Perfect, that's exactly what I needed.  Thanks! :)

closed 2020-12-27 by jailbird777