pcengines/apu2-documentation #192: USB issues on the new FreeBSD-12.2

sniku, 2021-02-21:
APU2/3/4 + FreeBSD-12.2 + USB memory don't like each other very much.

Messages such as below appear in large numbers on this hardware+OS combination. 
```
Root mount waiting for: CAM
Root mount waiting for: CAM
Root mount waiting for: CAM
(probe0:umass-sim0:0:0:0): INQUIRY. CDB: 12 00 00 00 24 00                                                                                              
(probe0:umass-sim0:0:0:0): CAM status: CCB request completed with an error                                                                              
(probe0:umass-sim0:0:0:0): Retrying command, 3 more tries remain   
```

Since pfSense 2.5.0 is based on FreeBSD-12.2, and a USB stick is required to install this OS, this problem is somewhat painful. Most of my USB sticks don't work for installing the new pfSense. Some of them, however, work after printing several of these retry messages. 

Here's a recording of a successful installation and a successful boot from SSD. https://asciinema.org/a/392645

I would appreciate any info/workarounds for this issue.

pietrushnic, 2021-02-21:
@sniku thanks for reporting this issue.

@macpijan @pkonkol we definitely have to update our test suites to cover the most recent FreeBSD release. Can we schedule that for March release?

pkonkol, 2021-02-22:
There is a similar issue in pcengines/coreboot: https://github.com/pcengines/coreboot/issues/461. We can add that to the test suites in march.