pcengines/apu2-documentation #194: Tianocore payload: silent resets

Yuubari, 2021-03-29:
I built v4.13.0.3 with Tianocore payload for my apu4d4 according to the instructions. There are no issues during build time and the firmware seems to work, but the machine silently resets daily (and I have not noticed it resetting more often than once a day). I am suspecting some kind of a watchdog; is there anything to poke for logs or whatnot?

The reset happens abruptly, with no OS shutdown.

miczyg1, 2021-03-29:
No clues for now. We haven't given much extensive testing to Tianocore payload with apu2. We will try to check and test the uptime with this configuration and let you know the results.

pietrushnic, 2021-03-29:
@Yuubari please note any workaround Tianocore payload for PC Engines is out of the maintenance package that 3mdeb provides to PC Engines. There are a couple of ways to work around that:

- Since there is some commercial interest in UEFI for PC Engines we working on [Dasharo](https://dasharo.com), through that product we will provide support for UEFI-based firmware for PC Engines. This would be pay-as-you-go (per release) commercial effort or sponsorware.
- Convince PC Engines to support 3rd line of firmware, we already support legacy and mainline, it would be another mainline firmware but with UEFI payload
- Contribute to the repositories so this can be supported, this is quite a lot of work to be able to develop support and test it

kpfleming, 2021-04-03:
Thank you for reporting this @Yuubari! Just yesterday I was setting up two new systems and had built Tianocore-enabled firmware so that I could use GPT instead of MBR... chatting about it in IRC with some smart people one of them pointed out this issue, so I've switched back to SeaBIOS (and GRUB is able to boot from GPT even from a BIOS).

sqozz, 2021-05-20:
> I built v4.13.0.3 with Tianocore payload for my apu4d4 according to the instructions. There are no issues during build time and the firmware seems to work, but the machine silently resets daily (and [...]

I see the same behaviour with a Linux payload so this doesn't seem to be UEFI specific

miczyg1, 2021-05-20:
@sqozz this is rather a payload specific issue. I think I know what may cause this. We keep the runtime configuration in SeaBIOS related files, so if another payload is used, the watchdog is accidentally set to max value. If the Linux kernel has no working driver for it, the watchdog will kick resets periodically.

sqozz, 2021-05-20:
> @sqozz this is rather a payload specific issue. I think I know what may cause this. We keep the runtime configuration in SeaBIOS related files, so if another payload is used, the watchdog is accidentally set to max value. If the Linux kernel has no working driver for it, the watchdog will kick resets periodically. I will fix it for the next release

Thanks! I'm just about to compile a kernel with the sp5100_tco watchdog driver. Do you know if this is the correct one?

pietrushnic, 2021-05-20:
https://github.com/pcengines/pce-fw-builder/issues/41

miczyg1, 2021-05-20:
@sqozz yes, that's the correct driver

Yuubari, 2021-05-20:
Does this watchdog default apply to UEFI payload as well? I *think* FreeBSD does not have a driver for the SP5100, so could this be the reason my system was resetting daily?

miczyg1, 2021-05-20:
That's exactly what I am saying.

sqozz, 2021-05-22:
Thanky @miczyg1 with the sp5100_tco driver I'm able to run the system continuously for 2 days now. @pietrushnic pointing to Dasharo unfortunately doesn't help much for private persons who run an APU board. I'm almost certain it would cost me more then the whole machine…

pietrushnic, 2021-05-22:
@sqozz over time more and more people may be interested in Tianocore payload support, what means it can be crowdfunded. We consider Dasharo sponsorware model, which may be not so expensive for regular buyer. Unfortunately, 3mdeb cannot afford maintaining the same quality as for regular releases for Tianocore payload, eventually we have to pay salaries. I hope you understand my point.

TBH in long run most of the projects should be driven by demand. We hope to create ecosystem around Dasharo that can help sponsor OSF projects by multiple parties including small subscription fees from community and adequately shared commitment from business. What do you think about that? Do you think Open Source Firmware can bring value ~5USD/month for individual user?