pcengines/apu2-documentation #196: M.2 / NGFF 4G Modems in an APU2

lentinj, 2021-04-07:
FWIW, I've had success using a Huawei ME906s M.2 4G modem in my APU2D with BIOS v4.13.0.1, using a "M.2 Key B to Mini PCIE PCI-E Adapter Converter for 3G/4G/5G" available from the usual sources (with a free screwdriver!). This includes routing the SIM signals so I can use the APU2 SIM slot. Given M.2 is a lot more common than mPCIe nowadays, thought this might be useful to someone.

Please add to ``mpcie_modules.md`` if you think it belongs there, or equally feel free to close if it doesn't.

miczyg1, 2021-04-07:
@lentinj if you don't mind sharing some photos of this setup? We can gladly add it to the documentation

lentinj, 2021-04-07:
Sure:

[image]

Just the adapter, The end of it needs to be snapped off to fit, but it's designed to do so.

[image]

With a modem in it, and "Sma Female Bulkhead To U.Fl (Ipex/Ipx) MHF-4" cables attached. SIM card sits in the usual slot at the front.

miczyg1, 2021-06-23:
https://github.com/pcengines/apu2-documentation/pull/198

closed 2021-06-23 by miczyg1