pcengines/apu2-documentation #199: Building Protectli Firmware

Ben-Mathews, 2021-07-05:
I know this repo is specifically for PCEngines products, but I am hoping someone can help me build firmware for a Protectli FW4B.  The instructions provided by Protectli don't reference the versions or repos to use, and the versions provided for download by Protectli (e.g. 4.12.0.5) map to tags in this repo, not tags in the coreboot repo, so I assume that their builds come from here.  The one time I tried loading an image created from the coreboot repo yielded a bricked device.  

When I try to build using the docker image and `make menuconfig`, targeting the Protectli FW4B, I get the following error:

```
build/bootblock/drivers/uart/uart8250io.o: In function `uart_init':
/home/coreboot/coreboot/src/drivers/uart/uart8250io.c:98: undefined reference to `check_com2'
make: *** [src/arch/x86/Makefile.inc:92: build/cbfs/fallback/bootblock.debug] Error 1
coreboot@ec09eae5fd08:~/coreboot$ git status
On branch release
Your branch is up to date with 'origin/release'.
```

I get this on both the current release branch and on the 4.12.0.5 tag.  I see this referenced [here](https://github.com/pcengines/apu2-documentation/issues/145) as a possible permissions issue, but it's not clear to me how to resolve that.  

I've also seen the [pce_fw_builder](https://github.com/pcengines/pce-fw-builder) repo, but that seems targeted at PC Engines hardware.

Any help building for a Protectli device would be greatly appreciated.

miczyg1, 2021-07-06:
> I know this repo is specifically for PCEngines products, but I am hoping someone can help me build firmware for a Protectli FW4B. The instructions provided by Protectli don't reference the versions [...]

@Ben-Mathews unfortunately this is not true. Protectli coreboot builds for FW2B/FW4B and FW6A/B/C v4.12.0.x does not come from here.

> When I try to build using the docker image and `make menuconfig`, targeting the Protectli FW4B, I get the following error:
>
> ```
> build/bootblock/drivers/uart/uart8250io.o: In function `uart_init': [...]
PC Engines coreboot repository is solely targeted on PC Engines platforms, other platform builds will not simply work due to PC Engines specific features we maintain here.
> 
> I've also seen the [pce_fw_builder](https://github.com/pcengines/pce-fw-builder) repo, but that seems targeted at PC Engines hardware.
> 
> Any help building for a Protectli device would be greatly appreciated.

I will see what I can do and I will get back to you with new info.

miczyg1, 2022-04-11:
Closing as it is not relevant to PC Engines.

closed 2022-04-11 by miczyg1