pcengines/apu2-documentation #203: after update apu2 to Versie:v4.14.0.4 openvpn client do not start anymore

heuvep, 2021-09-19:
After the firmware update to Versie:v4.14.0.4 on my pfsense apu2 box openvpn client do not start any more and also it can not see if there are newer version pfsense.

heuvep, 2021-09-19:
I did the update  this way as a always do 
flashrom -w /tmp/apu2_v4.14.0.4.rom -p internal:boardmismatch=force

heuvep, 2021-09-19:
[image]

heuvep, 2021-09-19:
as you can see it can not check for update at the versie tab (left)

miczyg1, 2021-09-20:
@heuvep do you possibly have any logs from the openvpn client service:
- from a previous BIOS version?
- when openvpn client is started manually on current BIOS version?

Do they differ? Sharing `dmesg` output from both situations would be helpful (you may strip it from any data you consider private).

heuvep, 2021-09-20:
problem is solved. default route was changed after reboot
Op ma., sep. 20, 2021 om 10:09, Michał Żygowski  schreef:
 @heuvep (https://github.com/heuvep) do you possibly have any logs from the openvpn client service:
	* from a previous BIOS version? 
	* when openvpn client is started manually on current BIOS version? 

 Do they differ? Sharing dmesg output from both situations would be helpful (you may strip it from any data you consider private).

miczyg1, 2021-09-20:
@heuvep glad to hear. Closing

closed 2021-09-20 by miczyg1