pcengines/apu2-documentation #251: 2 x APU2E4 unstable with CPB enabled.

MyGithubUser01, 2021-10-18:
Hi all,

I'm having some serious stability issues with APU2E4 and CPB with BIOS 4.13.0.1 and 4.13.0.5
This is brand new hardware which was believed to be "DOA" but the replacement I got had the exact same issue.
After disabling CPB the system appears to be stable and has an uptime of a record high 4 days and going.

Operating system tested OPNsense 21.1 and 21.1.5.
I've tried booting from msata, sd card and USB but it gives me the same issue.
I've also tried multiple power adapters.
The CPU Temperature is typically in the range 54-56c and the system isn't even connected to any network just the console cable.

The system has been very unstable and is core dumping every 4-12 hours. BIOS 4.13.0.1, but did see similar issues when testing 4.13.0.5.

From the logs/console I see the following:

FreeBSD/amd64 (OPNsense.localdomain) (ttyu0)
login: MCA: Bank 1, Status 0x9400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR ICACHE L1 IRD error
MCA: Address 0x282060
[HBSD SEGVGUARD] [/usr/local/bin/python3 (5880)] Suspension expired.
 -> pid: 5880 ppid: 1302 p_pax: 0xa50<SEGVGUARD,ASLR,NOSHLIBRANDOM,NODISALLOWMAP32BIT>

And:
"root@OPNsense:/var/db/rrd # MCA: Bank 1, Status 0xd400000000000151
MCA: Global Cap 0x0000000000000106, Status 0x0000000000000000
MCA: Vendor "AuthenticAMD", ID 0x730f01, APIC ID 0
MCA: CPU 0 COR OVER ICACHE L1 IRD error
MCA: Address 0xffff80d1ff60"

Let me know if additional details are required.
Broken hardware, bios bug, OPNsense/HardenedBSD compatibility issues?

kurtselbach, 2022-01-06:
> After a power-outage, my AP2E4 started to misbehave again, randomly locking up.
> 
> Had to check if CPB for some reason had been re-enabled, and for sure, it was Enabled again.
> 
> I'll report back if it is still stable.

14 days later, still stable?

mkopec, 2022-05-17:
Hi @toredash , did you experience any more lock-ups since disabling CBP?

poetiquette, 2022-05-20:
Apologies for a "me too" comment but unfortunately for me, the described symptoms in this issue are also somewhat occurring with my APU2E4. Alas, I'm long past the warranty period as I've purchased mine in the summer of 2020. I have not tested with Linux since I am mainly running pfSense on this unit. This occurs with 2.6.0 and a few previous versions.

I've only experienced it randomly crashing and restarting a few times, but since this unit is operating as a firewall for my home connection, I'd rather turn off CPB to make the unit stable again than deal with the instability and MCA errors. If at all possible, I would gladly appreciate a fix to this issue as I could use the extra performance to handle bursty traffic flows since I have a gigabit internet connection. I am willing to turn CPB on again offer my help in debugging the problem

As mentioned just above, I also see the MCA errors with CPB enabled but unfortunately I don't have the logs anymore (I've stumbled across this issue randomly when reading the documentation for something unrelated), but they appear very similar and the error seemed to have been associated with CPU2 in my case. With CPB off, I do not see them ever appear in the syslog. I've since then done a fresh install and why I don't have the logs anymore.

I do however, still see errors in the logs of pfSense that are seemingly related to the firmware and the first i210 NIC and I don't know if it's related.

```
May 19 22:43:04 	kernel 		igb2: netmap queues/slots: TX 4/1024, RX 4/1024
May 19 22:43:04 	kernel 		igb2: Ethernet address: <REDACTED>
May 19 22:43:04 	kernel 		igb2: Using MSI-X interrupts with 5 vectors
May 19 22:43:04 	kernel 		igb2: Using 4 RX queues 4 TX queues
May 19 22:43:04 	kernel 		igb2: Using 1024 TX descriptors and 1024 RX descriptors
May 19 22:43:04 	kernel 		igb2: NVM V0.6 imgtype5
May 19 22:43:04 	kernel 		igb2: <Intel(R) I210 Flashless (Copper)> port 0x3000-0x301f mem 0xd0200000-0xd021ffff,0xd0220000-0xd0223fff irq 36 at device 0.0 on pci3
May 19 22:43:04 	kernel 		pci3: <ACPI PCI bus> on pcib3
May 19 22:43:04 	kernel 		pcib3: <ACPI PCI-PCI bridge> irq 27 at device 2.4 on pci0
May 19 22:43:04 	kernel 		igb1: netmap queues/slots: TX 4/1024, RX 4/1024
May 19 22:43:04 	kernel 		igb1: Ethernet address: <REDACTED>
May 19 22:43:04 	kernel 		igb1: Using MSI-X interrupts with 5 vectors
May 19 22:43:04 	kernel 		igb1: Using 4 RX queues 4 TX queues
May 19 22:43:04 	kernel 		igb1: Using 1024 TX descriptors and 1024 RX descriptors
May 19 22:43:04 	kernel 		igb1: NVM V0.6 imgtype5
May 19 22:43:04 	kernel 		igb1: <Intel(R) I210 Flashless (Copper)> port 0x2000-0x201f mem 0xd0100000-0xd011ffff,0xd0120000-0xd0123fff irq 32 at device 0.0 on pci2
May 19 22:43:04 	kernel 		pci2: <ACPI PCI bus> on pcib2
May 19 22:43:04 	kernel 		pcib2: <ACPI PCI-PCI bridge> irq 26 at device 2.3 on pci0
May 19 22:43:04 	kernel 		igb0: netmap queues/slots: TX 4/1024, RX 4/1024
May 19 22:43:04 	kernel 		igb0: Ethernet address: <REDACTED>
May 19 22:43:04 	kernel 		igb0: Using MSI-X interrupts with 5 vectors
May 19 22:43:04 	kernel 		igb0: Using 4 RX queues 4 TX queues
May 19 22:43:04 	kernel 		igb0: Using 1024 TX descriptors and 1024 RX descriptors
May 19 22:43:04 	kernel 		igb0: NVM V0.6 imgtype5
May 19 22:43:04 	kernel 		igb0: <Intel(R) I210 Flashless (Copper)> mem 0xd0000000-0xd001ffff,0xd0020000-0xd0023fff irq 28 at device 0.0 on pci1
May 19 22:43:04 	kernel 		pci1: <ACPI PCI bus> on pcib1
May 19 22:43:04 	kernel 		pcib1: failed to allocate initial I/O port window: 0x1000-0x1fff
May 19 22:43:04 	kernel 		pcib1: <ACPI PCI-PCI bridge> irq 25 at device 2.2 on pci0
May 19 22:43:04 	kernel 		pci0: <base peripheral, IOMMU> at device 0.2 (no driver attached)
May 19 22:43:04 	kernel 		pci0: <ACPI PCI bus> on pcib0
May 19 22:43:04 	kernel 		pcib0: could not evaluate _ADR - AE_NOT_FOUND
May 19 22:43:04 	kernel 		pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0 
```

The relevant lines/errors:

`May 19 22:43:04 	kernel 		pcib1: failed to allocate initial I/O port window: 0x1000-0x1fff`

and

`May 19 22:43:04 	kernel 		pcib0: could not evaluate _ADR - AE_NOT_FOUND`

Again, I don't know if the above lines/errors are relevant to the instability issue with CPB at hand.

toredash, 2022-05-20:
> Hi @toredash , did you experience any more lock-ups since disabling CPB?

No, my device has been stable since CPB was disabled.

daduke, 2022-06-08:
FTR I've also had bad stability issues with an apu6 that seem to be solved by disabling CPB. Maybe CPB shouldn't be enabled by default?

toredash, 2022-09-04:
Update: My device is still stable after several months.

toredash, 2022-11-01:
> Perfect, thanks for the update. Got two unused APU2E4 in a drawer...
>
> Which BIOS version are you running now?

v4.14.0.6

mdickers47, 2023-03-21:
FWIW, I have a very old APU2 that became extremely unstable with CPB and Linux 6.2.5.  It generates a lot of different "null pointer dereference," "unable to handle page fault," and "soft lockup" panics.  It lasts no more than a few hours per reboot, and sometimes only a few minutes.  Disabling CPB in the BIOS seems to have solved it.

Here is one of the common panics:

```
[ 4238.591613] BUG: kernel NULL pointer dereference, address: 0000000000000003
[ 4238.598622] #PF: supervisor write access in kernel mode
[ 4238.603856] #PF: error_code(0x0002) - not-present page
[ 4238.609046] PGD 106842067 P4D 106842067 PUD 103a51067 PMD 0 
[ 4238.614744] Oops: 0002 [#1] PREEMPT SMP NOPTI
[ 4238.619132] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 6.2.5-arch1-1 #1 fcf70e9d97e045884ea945a3d5b5ff73b06f7a27
[ 4238.629245] Hardware name: PC Engines apu2/apu2, BIOS v4.19.0.1 01/31/2023
[ 4238.636132] RIP: 0010:psi_group_change+0x2f/0x400
[ 4238.640906] Code: 41 57 48 63 c6 49 89 ff 41 56 41 55 41 54 41 89 cc 55 53 48 83 ec 20 48 8b 5f 30 48 03 1c c5 c0 da eb ab 4c 89 04 24 83 03 01 <44> 89 4c 24 10 48 89 44 24 08 f6 c2 10 0f 85 ea 02 00 00 f6 c1 10
[ 4238.659677] RSP: 0018:ffff9b4e800d3dc0 EFLAGS: 00010002
[ 4238.664929] RAX: 0000000000000003 RBX: ffffbb4e7fd81dc0 RCX: 0000000000000010
[ 4238.672104] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff8c0a40d61800
[ 4238.679258] RBP: 0000000000000003 R08: 000003dadfbf0c5d R09: 0000000000000001
[ 4238.686442] R10: 0000000000000001 R11: 0000000000000100 R12: 0000000000000010
[ 4238.693593] R13: 0000000000000003 R14: ffff8c0a46408000 R15: ffff8c0a40d61800
[ 4238.700744] FS:  0000000000000000(0000) GS:ffff8c0a6ad80000(0000) knlGS:0000000000000000
[ 4238.708852] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 4238.714618] CR2: 0000000000000003 CR3: 0000000106eb0000 CR4: 00000000000406e0
```

toredash, 2025-11-25:
For fun:

v4.14.0.6 is still stable with CPB disabled. Uptime > 311 days now