pcengines/apu2-documentation #283: Kernel panic with ath10k

dseomn, 2021-12-05:
**PCEngines firmware version**
v4.14.0.5

**APU variant**
apu2e4

**OS and OS Version**
Debian GNU/Linux 10 (buster)

**Affected component(s), peripheral(s) or functionality**
The entire system crashes.

**Brief summary**
The kernel panics occasionally. The watchdog still works though, so it gets rebooted and then works.

**How reproducible**
I think it happens a bit less than once per week.

**How to reproduce**
I haven't noticed any pattern of when it happens.

**Expected behavior**
The kernel shouldn't panic.

**Actual behavior**
The kernel panics. Here's a log from the serial console the most recent time it happened:
```
[86737.652272] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-110)
[259531.353847] wlp5s0: failed to remove key (0, 14:c1:4e:b7:b5:d2) from hardware (-110)
[432016.150262] wlp1s0.guest.h2: failed to remove key (2, ff:ff:ff:ff:ff:ff) from hardware (-11)
[518728.502333] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-110)
[1005496.119589] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-11)
[1295278.270050] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-11)
[1296309.161775] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-110)
[1471067.906042] wlp1s0.lan.l3: failed to remove key (0, a0:af:bd:e9:cc:c6) from hardware (-11)
[1555503.955474] wlp5s0: failed to remove key (0, 14:c1:4e:b7:b5:d2) from hardware (-110)
[1589501.336457] wlp1s0: failed to remove key (0, 14:c1:4e:b7:b5:d2) from hardware (-11)
[1589502.759769] BUG: unable to handle kernel NULL pointer dereference at 0000000000000fc0
[1589502.767792] PGD 0 P4D 0 
[1589502.770522] Oops: 0002 [#1] SMP NOPTI
[1589502.774376] CPU: 3 PID: 19198 Comm: kworker/3:0 Tainted: G        W   E     4.19.0-18-amd64 #1 Debian 4.19.208-1
[1589502.784721] Hardware name: PC Engines apu2/apu2, BIOS v4.14.0.5 10/11/2021
[1589502.791874] Workqueue: events_freezable ieee80211_restart_work [mac80211]
[1589502.798856] RIP: 0010:_raw_spin_lock_bh+0x17/0x30
[1589502.803741] Code: 0f 1f 44 00 00 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 0f 1f 44 00 00 65 81 05 10 e2 cd 4b 00 02 00 00 31 c0 ba 01 00 00 00 <f0> 0f b1 17 85 c0 75 01 c3 89 c6 e8 d9 33 99 ff 66 90 c3 66 0f 1f
[1589502.822666] RSP: 0018:ffff8eca2ab83cc0 EFLAGS: 00010246
[1589502.828074] RAX: 0000000000000000 RBX: 0000000000000fc0 RCX: 0000000000000000
[1589502.835389] RDX: 0000000000000001 RSI: ffff8eca27e1f4c8 RDI: 0000000000000fc0
[1589502.842702] RBP: ffff8eca27e1f3a8 R08: 0000000000000000 R09: 0000000000000000
[1589502.850019] R10: ffff8ec935c7e0f8 R11: 00000000000001ff R12: ffffffffc0b66030
[1589502.857332] R13: 0000000000000000 R14: ffff8eca26dc0ff8 R15: ffff8eca27e1e8c0
[1589502.864650] FS:  0000000000000000(0000) GS:ffff8eca2ab80000(0000) knlGS:0000000000000000
[1589502.872915] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1589502.878841] CR2: 0000000000000fc0 CR3: 00000001291ec000 CR4: 00000000000406e0
[1589502.886148] Call Trace:
[1589502.888780]  <IRQ>
[1589502.891026]  ath10k_wmi_tx_beacons_iter+0x24/0x140 [ath10k_core]
[1589502.897253]  ? ath10k_wmi_cmd_send_nowait+0xa0/0xa0 [ath10k_core]
[1589502.903622]  __iterate_interfaces+0x72/0x110 [mac80211]
[1589502.909136]  ieee80211_iterate_active_interfaces_atomic+0xd/0x10 [mac80211]
[1589502.916313]  ath10k_wmi_op_ep_tx_credits+0x20/0x40 [ath10k_core]
[1589502.922533]  ath10k_htc_process_trailer+0x1af/0x280 [ath10k_core]
[1589502.928853]  ath10k_htc_rx_completion_handler+0x97/0x130 [ath10k_core]
[1589502.935592]  ? ath10k_htc_process_trailer+0x280/0x280 [ath10k_core]
[1589502.942051]  ath10k_pci_process_rx_cb+0x174/0x1b0 [ath10k_pci]
[1589502.948112]  ath10k_ce_per_engine_service+0x93/0x100 [ath10k_core]
[1589502.954508]  ath10k_ce_per_engine_service_any+0x75/0xb0 [ath10k_core]
[1589502.961136]  ath10k_pci_napi_poll+0x44/0x100 [ath10k_pci]
[1589502.966724]  net_rx_action+0x149/0x3b0
[1589502.970666]  __do_softirq+0xde/0x2d8
[... 215 lines cut ...]
[1589504.523352]  </IRQ>
[1589504.525663] RIP: 0010:ath10k_add_interface+0x79/0x1180 [ath10k_core]
[1589504.532195] Code: 89 e1 31 c0 48 c7 83 48 01 00 00 00 00 00 00 48 83 e7 f8 48 c7 83 a8 03 00 00 00 00 00 00 48 29 f9 81 c1 68 02 00 00 c1 e9 03 <f3> 48 ab 48 8b 83 30 01 00 00 48 85 c0 74 0c 48 8d 50 18 48 89 50
[1589504.551120] RSP: 0018:ffffb16bc0a5fd60 EFLAGS: 00000212 ORIG_RAX: ffffffffffffffda
[1589504.558868] RAX: 0000000000000000 RBX: ffff8eca27e1f3a8 RCX: 000000000000001f
[1589504.566183] RDX: ffff8eca237c0000 RSI: ffff8eca27e1f3a8 RDI: ffff8eca27e1f660
[1589504.573498] RBP: ffff8eca26dc0780 R08: 0000000000200000 R09: 0000000000000000
[1589504.580814] R10: 000000000000001f R11: ffff8eca2605a100 R12: ffff8eca27e1f4f0
[1589504.588128] R13: ffff8eca26dc2540 R14: ffff8eca24609f00 R15: ffff8eca26dc15a0
[1589504.595528]  drv_add_interface+0x4c/0x130 [mac80211]
[1589504.600775]  ieee80211_reconfig+0x226/0x1420 [mac80211]
[1589504.606187]  ? rcu_exp_wait_wake+0x240/0x240
[1589504.610639]  ? try_to_del_timer_sync+0x4d/0x80
[1589504.615338]  ieee80211_restart_work+0xbb/0xe0 [mac80211]
[1589504.620834]  process_one_work+0x1a7/0x3a0
[1589504.625026]  worker_thread+0x30/0x390
[1589504.628874]  ? create_worker+0x1a0/0x1a0
[1589504.632983]  kthread+0x112/0x130
[1589504.636397]  ? kthread_bind+0x30/0x30
[1589504.640245]  ret_from_fork+0x22/0x40
[1589504.644006] ---[ end trace 90a3696deb151c83 ]---
```

**Solutions you've tried**
The watchdog timer turns this from a major issue into a tolerable one.

miczyg1, 2021-12-06:
@dseomn as per our documentation of mPCIe modules: https://github.com/pcengines/apu2-documentation/blob/master/docs/mpcie_modules.md
you may want to:
- try passing `amd_iommu=off` to the kernel cmdline OR
- disable the IOMMU in the apu BIOS setup

Then see if it helps

dseomn, 2021-12-07:
Thanks for the link, I just added `amd_iommu=off`. I'll update here if it panics again.

dseomn, 2021-12-17:
It panicked again, this time with `amd_iommu=off`.

metux, 2023-02-21:
Null pointer dereference: looks like a kernel bug in ath driver - please report to the corresponding kernel maintainer.
In your case - Debian kernel - the right place is the Debain Kernel Team.

Note that HW vendors usually neither have the capacity nor the knowledge to give SW or OS support.