pcengines/apu2-documentation #285: Watchdog timer documentation

jowrjowr, 2021-12-13:
**PCEngines firmware version**
v4.12.0.1

**APU2 variant**
apu4d4

**Affected component(s), peripheral(s) or functionality (if applicable)**
watchdog timer

**Brief summary**

Some documentation issues that have caused, let's say, "issues".

1. Make sure that the bios does not allow any non-zero setting below 30 seconds. There is no seemingly valid reason to ever do this, and a misunderstanding (LIKE I JUST MADE) bricks the device.
2. Adjust the [firmware documentation](https://github.com/pcengines/apu2-documentation/blob/master/docs/theory-of-operation.md) to include the details from the [firmware docs on the subject](https://pcengines.github.io/firmware/2019/06/10/PC-Engines-Firmware-v4-9-0-6.html)

Specifically, this bit:

`By default the timeout is set to 0 seconds (disabled state). To enable watchdog, enter setup menu and toggle watchdog option. You will be prompted to specify the timeout after which the platform should reset. Options description WARNING do not set short timeouts! It may lead to a reset loop and brick Your platform. Please take into consideration that platform boot time and OS boot time also counts to the overall timeout time, so set at least few minutes timeout to still be able to enter setup menu to disable the watchdog! The operating system has to support the watchdog, otherwise the platform will constantly reboot. For Linux OSes the driver is sp5100_tco, however it conflicts with i2c_piix4 leaving the watchdog driver unloaded completely. One has to properly blacklist the i2c_piix4 driver in order to get the watchdog working.`

3. Make it clear that settings like this are written to flash and that a J10 reset will not help.
4. Additionally and more generally, make it clear on what settings can potentially brick the device and make it clear that there's capacity for self-ownage.

**Additional context**

The APU4D4 schematics make reference to watchdog enable on the J3 header, but this doesn't seem to help. It is inconsistent with the manual saying those pins are simply reserved. Following up with support on that point to see if there's any way to unbrick this without having to pay the stupid tax and order the unbrick tool (the tax was just paid as the answer looks like "no").

miczyg1, 2021-12-13:
@jowrjowr I am so sorry to hear that. `misunderstanding (LIKE I JUST MADE)` - what exactly happened?

The disadvantage of a big documentation is that the information tends to be scattered or not structured well for everyone to move smoothly throughout documents. Here you may find detailed documentation about runtime configuration: https://github.com/pcengines/sortbootorder/blob/master/README.md#settings-description
it also mentions the warning about watchdog small timeout.

Of course we will apply your suggestion to mark the options which can potentially brick the device.

J3 header watchdog jumper connects the Super I/O Nuvoton NCT5104D watchdog output to the reset pin of the processor. It would be possible to use it, however the NCT5104D watchdog driver is not so common in the distros, so it would not work well (as there would be no software to feed the watchdog regularly).

macpijan, 2021-12-13:
There is also a safeguard since the firmware version v4.14.0.4: https://pcengines.github.io/#mr-50 which prevents watchdog timer to be set below 60s https://github.com/pcengines/coreboot/commit/a80d65899010c4c92f7bca14130ca1aaee2e9414

IgorBagnucki, 2022-01-25:
There is created pull request related to the issue that should add missing watchdog information to the documentation. [https://github.com/pcengines/apu2-documentation/pull/291](https://github.com/pcengines/apu2-documentation/pull/291)

miczyg1, 2023-07-05:
> > J3 header watchdog jumper connects the Super I/O Nuvoton NCT5104D watchdog output to the reset pin of the processor.
> 
> @miczyg1 Does that mean the `Nuvoton NCT5104D` watchdog only has an effect when pin 5-6 is shorted?

Yes

>  And am I correct to assume the BIOS watchdog enable setting only enables the `sp5100_tco`?

Yes

TimelessNL, 2023-07-05:
Thanks for you response 👍 

Do you by any chance know if the `NCT5104D` is reset during a warm reboot? And therefor won't trigger on a timeout?

miczyg1, 2023-07-05:
It will probably get reset. Also the watchdog timer stops whenever it expires

TimelessNL, 2023-07-05:
Ah that's unfortunate, I was hoping it would not.
Probably because `LRESET` being triggered by the LPC right?

>Also the watchdog timer stops whenever it expires

True, but in that case it should fire a pulse on `WDTOUT` and reset the board when it hangs in the BIOS I suppose.