pcengines/apu2-documentation #287: APU1: the hard disk is not visible in the Seabios Boot Menu after a coldboot.

damiankaras, 2021-12-30:
**PCEngines firmware version**

v4.15.0.2

**APU variant**

apu1d

**OS and OS Version**

**Affected component(s), peripheral(s) or functionality**

hard disk

**Brief summary**

After the coldboot, Hard-Disk is not visible in Seabios Boot Menu. It was tested with two disks.

**How reproducible**

Everytime

**How to reproduce**

Steps to reproduce the behavior:
1. Make a coldboot.
2. Enter the Seabios Boot Menu
3. Check the visibility of the hard disk.

**Expected behavior**

The hard disk should be visible in the Boot Menu after a coldboot.

**Actual behavior**

The hard disk is not visible in the Boot Menu after a coldboot.

**Screenshots**

**Additional context**

**Solutions you've tried**

Change between two disks. The results are the same.

IgorBagnucki, 2022-03-14:
Fixed in release [v4.15.0.3](https://pcengines.github.io/#mr-55)

damiankaras, 2022-03-14:
I confirm.

closed 2022-03-14 by damiankaras