pcengines/apu2-documentation #289: APU2E4 crashing frequently

bylienator, 2022-01-10:
**PCEngines firmware version**
v4.14.0.6

**APU2 variant**
APU2E4

**Question text**
Hi,

I purchased an APU2E4 in march 2021 and while it initially worked fine it has started crashing more and more in the recent months.

I've tried multiple combinations of OS's (OpenWRT, OPNsense, OpenBSD, Debian) and BIOS versions (mainline and legacy) but all tend to crash. Some crashes happen during boot while others can be triggered by running some task to generate load (stress-ng, openssl speed, stress, sha256sum, ...).

Running Memtest86+ reports no errors.

CPB has been disabled as I've read multiple reports pointing to this as the culprit.

Temperature probes reports values between 50 and 60 degrees C.

I've tried running the same tests with the internal mSATA drive removed and running the OS from USB but the results are the same.

I've replaced the PSU with a different, but higher rated one (12v/4A), with the same barrel jack but also the same results.

I'm at my wits' end here and don't know how to proceed further. The APU2E4 is replaced by a different router at the moment.
Are there any definitive load tests with specific BIOS version/settings and OS's/tools that should definitely give a positive result so that I can verify the hardware?

bylienator, 2022-01-12:
OpenBSD crashes:
```
During boot
kernel: double fault trap, code=0
Stopped at      rw_exit_write+0x64:     popq    %r11

kernel: protection fault trap, code=0
Stopped at      pmap_tlb_shootpage+0x10c:       movl    0x2c8(%r13),%eax
```
```
openssl speed
kernel: page fault trap, code=0
Stopped at      usertrap+0x140: leave
    TID    PID    UID     PRFLAGS     PFLAGS  CPU  COMMAND
*459145  60239      0    0x100003          0    2K openssl
ffffffff81bd32f0(uvm_fault(0xffffffff821d0090, 0xfffffffffffffffe, 0, 1) -> e
kernel: page fault trap, code=0
Stopped at      db_read_bytes+0x3f:     movq    0(%rdi),%rax
    TID    PID    UID     PRFLAGS     PFLAGS  CPU  COMMAND
*459145  60239      0    0x100003          0    2K openssl
db_read_bytes(fffffffffffffffe,8,ffff800022631b38) at db_read_bytes+0x3f
db_get_value(fffffffffffffffe,8,0) at db_get_value+0x3f
db_stack_trace_print(ffffffff81bd32f0,0,e,ffffffff81ee463b,ffffffff81e07f70) at
 db_stack_trace_print+0x2fd
db_trap(6,0) at db_trap+0xeb
db_ktrap(6,0,ffff800022631d40) at db_ktrap+0x112
kerntrap(ffff800022631d40) at kerntrap+0xa2
alltraps_kern_meltdown() at alltraps_kern_meltdown+0x7b
end of kernel
end trace frame: 0x6, count: 8
```
```
stress -c 4 -m 2 -v
kernel: stack fault trap, code=0
Stopped at      0x578b4900000028e8:     kernel: protection fault trap, code=0
Stopped at      db_read_bytes+0x70:     movzbl  0(%rdi,%rcx,1),%eax
```

OPNsense crashes:
```
during install
panic: vm_fault_hold: fault on nofault entry, addr: 0xfffffe0025b5b000
cpuid = 2
time = 1641999242
__HardenedBSD_version = 1200059 __FreeBSD_version = 1201000
version = FreeBSD 12.1-RELEASE-p19-HBSD #0  f692db99f3c(stable/21.7)-dirty: Mon
 Aug  2 13:29:29 CEST 2021
    root@sensey:/usr/obj/usr/src/amd64.amd64/sys/SMP
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe0025b5b470
vpanic() at vpanic+0x1a2/frame 0xfffffe0025b5b4c0
panic() at panic+0x43/frame 0xfffffe0025b5b520
vm_fault_hold() at vm_fault_hold+0x26cf/frame 0xfffffe0025b5b670
vm_fault() at vm_fault+0x60/frame 0xfffffe0025b5b6b0
trap_pfault() at trap_pfault+0x164/frame 0xfffffe0025b5b710
trap() at trap+0x29f/frame 0xfffffe0025b5b820
calltrap() at calltrap+0x8/frame 0xfffffe0025b5b820
--- trap 0xc, rip = 0xfffffe0025b5b8e0, rsp = 0xfffffe0025b5b8f0, rbp = 0xfffff
e0025b5b940 ---
??() at 0xfffffe0025b5b8e0/frame 0xfffffe0025b5b940
g_io_deliver() at g_io_deliver+0x265/frame 0xfffffe0025b5b9a0
g_disk_done() at g_disk_done+0x11c/frame 0xfffffe0025b5b9e0
xpt_done_process() at xpt_done_process+0x682/frame 0xfffffe0025b5ba20
xpt_done_td() at xpt_done_td+0x116/frame 0xfffffe0025b5ba70
fork_exit() at fork_exit+0x83/frame 0xfffffe0025b5bab0
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe0025b5bab0
--- trap 0, rip = 0, rsp = 0, rbp = 0 ---
KDB: enter: panic
[ thread pid 7 tid 100044 ]
Stopped at      kdb_enter+0x3b: movq    $0,kdb_why
```
```
openssl speed
pid 20498 (openssl), jid 0, uid 0: exited)
Segmentation fault (core dumped)
```
```
stress-ng --cpu 4 --verbose
Fatal double fault
rip 0x252e97 rsp 0x17ea8d55 rbp 0x6e42604c2ed0
rax 0x17ea8d753f9074b6 rdx 0x17ea8d7500000000 rbx 0xfffffffffffa7b00
rcx 0x26da74b6 rsi 0x6e42604c2fe0 rdi 0x26da3290
r8 0 r9 0x400 r10 0x8408
r11 0xfffe0fffffffffff r12 0x6e42604c3020 r13 0x29b44018568
r14 0x70 r15 0x26b1c0 rflags 0x202
cs 0x43 ss 0x3b ds 0x3b es 0x3b fs 0x13 gs 0x1b
fsbase 0x29b4321a8d0 gsbase 0xffffffff82230a00 kgsbase 0
cpuid = 2; apic id = 02
panic: double fault
cpuid = 2
time = 1642001348
__HardenedBSD_version = 1200059 __FreeBSD_version = 1201000
version = FreeBSD 12.1-RELEASE-p19-HBSD #0  f692db99f3c(stable/21.7)-dirty: Mo1
    root@sensey:/usr/obj/usr/src/amd64.amd64/sys/SMP
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe0001690db0
vpanic() at vpanic+0x1a2/frame 0xfffffe0001690e00
panic() at panic+0x43/frame 0xfffffe0001690e60
dblfault_handler() at dblfault_handler+0x1de/frame 0xfffffe0001690f30
Xdblfault() at Xdblfault+0xc6/frame 0xfffffe0001690f30
--- trap 0x17, rip = 0x252e97, rsp = 0x17ea8d55, rbp = 0x6e42604c2ed0 ---
KDB: enter: panic
[ thread pid 40810 tid 100224 ]
Stopped at      kdb_enter+0x3b: movq    $0,kdb_why
```

bylienator, 2022-01-13:
OpenWrt 21.02.1 crashes:
```
openssl speed
[  373.383068] ------------[ cut here ]------------
[  373.387886] WARNING: CPU: 2 PID: 0 at 0xffffffff81103214
[  373.393539] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADy
[  373.454003] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.154 #0
[  373.460229] Hardware name: PC Engines apu2/apu2, BIOS v4.14.0.6 11/04/2021
[  373.467489] RIP: 0010:0xffffffff81103214
[  373.471671] Code: 05 49 d2 f0 7e 48 c7 80 d0 00 00 00 00 00 00 00 48 89 d0 65 48 03 05 33 d2 f0 7e ba 02 00 00 09
[  373.491532] RSP: 0018:ffffc90000083ed0 EFLAGS: 00010087
[  373.497047] RAX: ffff88812ad205c0 RBX: 0000000000000002 RCX: 4000000000000000
[  373.504620] RDX: 00000000000205c0 RSI: 0000000000000e33 RDI: 0000000000000006
[  373.512244] RBP: ffffc90000083ed0 R08: 000000000000000a R09: 000000000001f1c0
[  373.519962] R10: 0000000000000000 R11: 0000000000000100 R12: ffffffff8209bb80
[  373.527355] R13: ffff888129ad3400 R14: ffffffff82055680 R15: 0000000000000002
[  373.534937] FS:  0000000000000000(0000) GS:ffff88812ad00000(0000) knlGS:0000000000000000
[  373.543379] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  373.549409] CR2: 000000000044fc40 CR3: 0000000128aa2000 CR4: 00000000000406e0
[  373.556897] Call Trace:
[  373.559486]  0xffffffff810d58d6
[  373.562775]  0xffffffff810d5b08
[  373.566109]  0xffffffff81043885
[  373.569436]  0xffffffff810000d4
[  373.572743] ---[ end trace 94a20389bc851814 ]--
```
```
stress-ng --cpu 4 --verbose
[  128.812540] PANIC: double fault, error_code: 0x0
[  128.817493] CPU: 2 PID: 3856 Comm: stress-ng Not tainted 5.4.154 #0
[  128.824169] Hardware name: PC Engines apu2/apu2, BIOS v4.14.0.6 11/04/2021
[  128.831473] RIP: 0010:0xffffffff81a01321
[  128.835668] Code: 48 0f b3 1c 25 16 f2 01 00 eb 05 49 0f ba ee 3f 41 0f 22 de e9 ab f7 ff ff fc 56 48 8b 74 24 05
[  128.855533] RSP: 0000:fffffe0000068000 EFLAGS: 00010046
[  128.861073] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  128.868648] RDX: 0000000000016bc0 RSI: ffffffff81a01158 RDI: fffffe0000068088
[  128.876179] RBP: fffffe0000068078 R08: 0000000000000000 R09: 0000000000000000
[  128.883611] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  128.891285] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  128.898939] FS:  00007f33932b7d48(0000) GS:ffff88812ad00000(0000) knlGS:0000000000000000
[  128.907600] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  128.913558] CR2: fffffe0000067ff8 CR3: 000000012926e000 CR4: 00000000000406e0
[  128.921107] Call Trace:
[  128.923675]  <ENTRY_TRAMPOLINE>
[  128.927012]  ? 0xffffffff81a00ae0
[  128.930474]  ? 0xffffffff81053c81
[  128.933890]  ? 0xffffffff81054099
[  128.937484]  0xffffffff81a01184
[  128.940921] RIP: 0010:0xffffffff81053c81
[  128.945084] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  128.965155] RSP: 0000:fffffe0000068138 EFLAGS: 00010046
[  128.970730] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  128.978323] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068158
[  128.986061] RBP: fffffe0000068148 R08: 0000000000000000 R09: 0000000000000000
[  128.993553] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  129.001091] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  129.008737]  ? 0xffffffff81a00ae0
[  129.012221]  ? 0xffffffff81054099
[  129.015685]  0xffffffff81a01184
[  129.018989] RIP: 0010:0xffffffff81053c81
[  129.023094] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  129.042951] RSP: 0000:fffffe0000068208 EFLAGS: 00010046
[  129.048564] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  129.056353] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068228
[  129.063798] RBP: fffffe0000068218 R08: 0000000000000000 R09: 0000000000000000
[  129.071511] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  129.079082] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  129.086778]  ? 0xffffffff81a00ae0
[  129.090239]  ? 0xffffffff81054099
[  129.093719]  0xffffffff81a01184
[... 113 lines cut ...]
[  129.906575] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  129.914115] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068b18
[  129.921724] RBP: fffffe0000068b08 R08: 0000000000000000 R09: 0000000000000000
[  129.929421] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  129.936899] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  129.944463]  ? 0xffffffff81a00ae0
[  129.947950]  ? 0xffffffff81054099
[  129.951467]  0xffffffff81a01184
[  129.954781] RIP: 0010:0xffffffff81053c81
[  129.959047] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  129.978760] RSP: 0000:fffffe0000068bc8 EFLAGS: 00010046
[  129.984304] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  129.991953] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068be8
[  129.999620] RBP: fffffe0000068bd8 R08: 0000000000000000 R09: 0000000000000000
[  130.007291] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  130.014842] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  130.022420]  ? 0xffffffff81a00ae0
[  130.025903]  ? 0xffffffff81054099
[  130.029465]  0xffffffff81a01184
[  130.032828] RIP: 0010:0xffffffff81053c81
[  130.037099] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  130.057112] RSP: 0000:fffffe0000068c98 EFLAGS: 00010046
[  130.062742] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  130.070359] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068cb8
```

bylienator, 2022-01-13:
What stands out, now that I've collected the different crashes, is that in most of them cpu 2 is mentioned.

So I went ahead and disabled cpu 2 in OpenWrt:
`echo 0 > /sys/devices/system/cpu/cpu2/online`

After that it seems that the crashes stop, running a stress-ng on OpenWrt in normal circumstances crashes almost immediately but now keeps on running until stopped.

Could this particular core (cpu 2) of the AMD GX-412TC soc be malfunctioning?

miczyg1, 2022-01-13:
@bylienator thank you for a detailed report. From what I have read you have extensively tried multiple versions and options that are known to work. At this point there is very little I can advise to be honest given that the platform was working well at the beginning. Interesting though that cpu2 caused the system to work stable. Could you maybe provide a dmesg from the OpenWRT? Maybe we could find something unusual there.

bylienator, 2022-01-13:
OpenWrt 21.02.1 dmesg
```
[    0.000000] Linux version 5.4.154 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16325-88151b8303)) #0 SMP Sun Oct 24 09:01:35 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=c58d873d-02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000cfe7ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cfe80000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000012effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000012f000000-0x000000012fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0 present.
[    0.000000] DMI: PC Engines apu2/apu2, BIOS v4.14.0.6 11/04/2021
[    0.000000] tsc: Fast TSC calibration failed
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] last_pfn = 0x12f000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-back
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00D0000000 mask FFF0000000 uncachable
[    0.000000]   1 base 00E0000000 mask FFE0000000 uncachable
[    0.000000]   2 base 012F000000 mask FFFF000000 uncachable
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] TOM2: 000000012f000000 aka 4848M
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] last_pfn = 0xcfe80 max_arch_pfn = 0x400000000
[    0.000000] Using GB pages for direct mapping
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F3830 000024 (v02 COREv4)
[    0.000000] ACPI: XSDT 0x00000000CFE950E0 000074 (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: FACP 0x00000000CFE96E10 000114 (v06 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: DSDT 0x00000000CFE95280 001B87 (v02 COREv4 COREBOOT 00010001 INTL 20200925)
[    0.000000] ACPI: FACS 0x00000000CFE95240 000040
[    0.000000] ACPI: SSDT 0x00000000CFE96F30 0001F8 (v02 COREv4 COREBOOT 0000002A CORE 20200925)
[    0.000000] ACPI: MCFG 0x00000000CFE97130 00003C (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: TPM2 0x00000000CFE97170 00004C (v04 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: APIC 0x00000000CFE971C0 00007E (v03 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: HEST 0x00000000CFE97240 0001D0 (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: SSDT 0x00000000CFE97410 0048A6 (v02 AMD    AGESA    00000002 MSFT 04000000)
[    0.000000] ACPI: SSDT 0x00000000CFE9BCC0 0007C8 (v01 AMD    AGESA    00000001 AMD  00000001)
[    0.000000] ACPI: DRTM 0x00000000CFE9C490 00007C (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: HPET 0x00000000CFE9C510 000038 (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.000000] ACPI: Reserving FACP table memory at [mem 0xcfe96e10-0xcfe96f23]
[    0.000000] ACPI: Reserving DSDT table memory at [mem 0xcfe95280-0xcfe96e06]
[    0.000000] ACPI: Reserving FACS table memory at [mem 0xcfe95240-0xcfe9527f]
[    0.000000] ACPI: Reserving SSDT table memory at [mem 0xcfe96f30-0xcfe97127]
[    0.000000] ACPI: Reserving MCFG table memory at [mem 0xcfe97130-0xcfe9716b]
[    0.000000] ACPI: Reserving TPM2 table memory at [mem 0xcfe97170-0xcfe971bb]
[    0.000000] ACPI: Reserving APIC table memory at [mem 0xcfe971c0-0xcfe9723d]
[    0.000000] ACPI: Reserving HEST table memory at [mem 0xcfe97240-0xcfe9740f]
[    0.000000] ACPI: Reserving SSDT table memory at [mem 0xcfe97410-0xcfe9bcb5]
[    0.000000] ACPI: Reserving SSDT table memory at [mem 0xcfe9bcc0-0xcfe9c487]
[    0.000000] ACPI: Reserving DRTM table memory at [mem 0xcfe9c490-0xcfe9c50b]
[... 450 lines cut ...]
[    9.459394] igb 0000:03:00.0: Intel(R) Gigabit Ethernet Network Connection
[    9.466842] igb 0000:03:00.0: eth2: (PCIe:2.5Gb/s:Width x1) 00:0d:b9:5a:08:52
[    9.474504] igb 0000:03:00.0: eth2: PBA No: FFFFFF-0FF
[    9.480034] igb 0000:03:00.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[    9.492693] ixgbe: Intel(R) 10 Gigabit PCI Express Network Driver - version 5.1.0-k
[    9.500693] ixgbe: Copyright (c) 1999-2016 Intel Corporation.
[    9.513469] i2c /dev entries driver
[    9.534865] e1000e: Intel(R) PRO/1000 Network Driver - 3.2.6-k
[... cut ...]
[   11.947126] device eth1 entered promiscuous mode
[   11.985408] 8021q: adding VLAN 0 to HW filter on device eth2
[   11.992496] br-lan: port 2(eth2) entered blocking state
[   11.998171] br-lan: port 2(eth2) entered disabled state
[   12.003941] device eth2 entered promiscuous mode
[   12.041345] 8021q: adding VLAN 0 to HW filter on device eth0
[   14.811196] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   15.034890] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

bylienator, 2022-01-13:
> @bylienator thank you for a detailed report. From what I have read you have extensively tried multiple versions and options that are known to work. At this point there is very little I can advise to [...]

In the beginning I had some similar issues, like being unable to reliably install a new OS, that seemed to be mitigated by turning CPB off so I didn't give it further thought. After that the unit ran fine for more than 6 months after which intermittent issues were beginning to occur which were not immediately clear as it's a headless setup without permanent console monitoring. It's only in the last weeks that it began getting worse and it's fuction as a router started to become problematic by multiple crashes/reboots every day.

I've been playing with stress-ng and the taskset parameter, which allows to set the CPU affinity and enable/disable CPU cores, and the issue seems to reliably occur when using all 4 cores:
```
root@OpenWrt:/# stress-ng --taskset 0,1,2,3 --cpu 4 --timeout 5m --verbose
stress-ng: debug: [3122] system: Linux OpenWrt 5.4.154 #0 SMP Sun Oct 24 09:01:35 2021 x86_64
stress-ng: debug: [3122] RAM total: 3.8G, RAM free: 3.8G, swap free: 0.0
stress-ng: debug: [3122] 4 processors online, 4 processors configured
stress-ng: info:  [3122] dispatching hogs: 4 cpu
stress-ng: debug: [3122] cache allocate: using cache maximum level L2
stress-ng: debug: [3122] cache allocate: shared cache buffer size: 2048K
stress-ng: debug: [3122] starting stressors
stress-ng: debug: [3123] stress-ng-cpu: started [3123] (instance 0)
stress-ng: debug: [3123] stress-ng-cpu using method 'all'
stress-ng: debug: [3124] stress-ng-cpu: started [3124] (instance 1)
stress-ng: debug: [3124] stress-ng-cpu using method 'all'
stress-ng: debug: [3122] 4 stressors started
stress-ng: debug: [3125] stress-ng-cpu: started [3125] (instance 2)
stress-ng: debug: [3125] stress-ng-cpu using method 'all'
stress-ng: debug: [3126] stress-ng-cpu: started [3126] (instance 3)
stress-ng: debug: [3126] stress-ng-cpu using method 'all'
[   34.598551] stress-ng[3123]: segfault at ffffffff00d599a9 ip 00000000004377df sp 00007ffc4c53b7e0 error 7 in stress-ng[409000+a5000]
[   34.611231] Code: c1 e2 08 c1 f9 04 00 8c 17 40 29 ca 00 45 84 db 0f 84 93 00 00 00 45 84 c0 0f 84 8a 00 00 00 4b 8d 14 52 c1 f8 04 48 c1 e2 08 <00> 84 17 40 29 ca 00 48 81 c6 00 03 00 00 4c 89 d1 0f b6 06 4c 8d
stress-ng: debug: [3122] process [3123] (stress-ng-cpu) terminated on signal: 11
stress-ng: debug: [3122] process [3123] terminated
stress-ng: debug: [3125] stress-ng-cpu: exited [3125] (instance 2)
stress-ng: debug: [3126] stress-ng-cpu: exited [3126] (instance 3)
stress-ng: debug: [3124] stress-ng-cpu: exited [3124] (instance 1)
stress-ng: debug: [3122] process [3124] terminated
stress-ng: debug: [3122] process [3125] terminated
stress-ng: debug: [3122] process [3126] terminated
stress-ng: info:  [3122] unsuccessful run completed in 2.06s
stress-ng: fail:  [3122] cpu instance 0 corrupted bogo-ops counter, 8 vs 0
stress-ng: fail:  [3122] cpu instance 0 hash error in bogo-ops counter and run flag, 1053922665 vs 0
stress-ng: fail:  [3122] metrics-check: stressor metrics corrupted, data is compromised

root@OpenWrt:/# stress-ng --taskset 0,1,2,3 --cpu 4 --timeout 5m --verbose
stress-ng: debug: [3185] system: Linux OpenWrt 5.4.154 #0 SMP Sun Oct 24 09:01:35 2021 x86_64
stress-ng: debug: [3185] RAM total: 3.8G, RAM free: 3.8G, swap free: 0.0
stress-ng: debug: [3185] 4 processors online, 4 processors configured
stress-ng: info:  [3185] dispatching hogs: 4 cpu
stress-ng: debug: [3185] cache allocate: using cache maximum level L2
stress-ng: debug: [3185] cache allocate: shared cache buffer size: 2048K
stress-ng: debug: [3185] starting stressors
stress-ng: debug: [3186] stress-ng-cpu: started [3186] (instance 0)
stress-ng: debug: [3186] stress-ng-cpu using method 'all'
stress-ng: debug: [3187] stress-ng-cpu: started [3187] (instance 1)
stress-ng: debug: [3187] stress-ng-cpu using method 'all'
stress-ng: debug: [3185] 4 stressors started
stress-ng: debug: [3188] stress-ng-cpu: started [3188] (instance 2)
stress-ng: debug: [3188] stress-ng-cpu using method 'all'
stress-ng: debug: [3189] stress-ng-cpu: started [3189] (instance 3)
stress-ng: debug: [3189] stress-ng-cpu using method 'all'
[   47.735882] BUG: kernel NULL pointer dereference, address: 0000000000000be6
[   47.743455] #PF: supervisor read access in kernel mode
[   47.749150] #PF: error_code(0x0000) - not-present page
[   47.754670] PGD 127539067 P4D 127539067 PUD 12951f067 PMD 0
[   47.760765] Oops: 0000 [#1] SMP NOPTI
[   47.764750] CPU: 2 PID: 1086 Comm: kworker/2:2 Not tainted 5.4.154 #0
[   47.771620] Hardware name: PC Engines apu2/apu2, BIOS v4.14.0.6 11/04/2021
[   47.778887] Workqueue:  0x0 (events)
[   47.782663] RIP: 0010:0xffffffff81024b0b
[   47.786813] Code: 00 00 66 39 c1 74 02 8e e0 89 d6 bf 00 01 00 c0 48 c1 ea 20 e8 26 a8 02 00 66 90 44 0f b7 ab e6 08 00 00 4c 8b b3 f0 08 00 00 <41> 0f b7 84 24 e6 08 00 00 66 41 83 fd 03 0f 87 80 01 00 00 4d 85
[   47.806789] RSP: 0018:ffffc9000225bea8 EFLAGS: 00010017
[   47.812359] RAX: 0000000000000000 RBX: ffff88812903ce00 RCX: 000000000000010f
[   47.820073] RDX: 0000000000000003 RSI: 0000000000000007 RDI: 0000000000000004
[   47.827588] RBP: ffffc9000225bec8 R08: 00000000ffff2b01 R09: 0000000000000113
[   47.835258] R10: 0000000000000110 R11: 0000000080000001 R12: 0000000000000300
[... 129 lines cut ...]
stress-ng: debug: [3443] starting stressors
[... cut ...]
stress-ng: debug: [3443] process [3446] terminated
stress-ng: debug: [3447] stress-ng-cpu: exited [3447] (instance 3)
stress-ng: debug: [3443] process [3447] terminated
stress-ng: info:  [3443] unsuccessful run completed in 9.74s
stress-ng: fail:  [3443] cpu instance 0 corrupted bogo-ops counter, 608 vs 0
stress-ng: fail:  [3443] cpu instance 0 hash error in bogo-ops counter and run flag, 3781793961 vs 0
stress-ng: fail:  [3443] metrics-check: stressor metrics corrupted, data is compromised
```

bylienator, 2022-01-17:
At first glance it's starting to look like a hardware issue but I'd like to be sure before I start an RMA to avoid discussions with the vendor about reproducability as it seems random in nature. Sometimes crashes happen as soon as bootup other times only after generating load.

Any further tests (specific BIOS versions/tools/OS's) I could run? Maybe something comparable to internal PC Engines testing (if applicable) so I can determine if the hardware is definitely faulty?

gbmaster, 2022-01-24:
Hello
I have an APU2D4 and I'm suffering exactly the same issue since a while and it got worse in the last months.

The machine becomes completely unresponsive on OpenBSD 7.0 and the following output is printed on the console:

```
kernel: double fault trap, code=0                                                                                                                                                                                                                                                         
Stopped at      mi_switch+0x20b:        callq   splassert_check+0x4                                                                                                                                                                                                                       
ddb{3}>
```

or similar messages. After this, it is impossible to interact with the serial console and the only way to get the machine back is to unplug and re-plug power.

If I can somehow help in investigating this issue, I'm available

bylienator, 2022-01-24:
Could you also try to run the "openssl" or "stress" commands on your OpenBSD install to see if similar crashes can be triggered?
```
openssl speed
stress -c 4 -m 2 -v
```
About a week ago I've send out an email to support@pcengines.ch for additional information on this issue but I haven't had a reply yet.

My hunch is that this is a hardware issue but I'd like to have some sort of formal procedure to verify this on my end before starting an RMA procedure. For example any test(s) PC Engines uses internally to verify similar cases would be helpful to validate a hardware issue and hopefully eliminate dicussions with the vendor regarding RMA.

For now this device is a paperweight on my desk because there's no way I'm able to use this reliably.
I'd also like to now what to do regarding this issue before march 2022 as the first year warranty period with the vendor will elapse after that.

gbmaster, 2022-01-24:
Both commands ran smoothly without causing the crash, so far

pietrushnic, 2022-02-02:
@sulewskiprzemyslaw I would appreciate if validation could try to reproduce above results and see, if this happen on our hardware. Also it would be great to have above stress/performance tests added to transparent validation suite.

@macpijan cc

TimelessNL, 2022-02-08:
I also have a couple of APU2/3 boards laying around with frequent restart/freeze issues for quite some time(#244). And since some people suggested to disable CBP I though lets give it another try, but unfortunately didn't make any difference.

But just for giggles I tried the oldest BIOS I could find which is [v4.0.9](https://pcengines.github.io/#lr-1) for the APU2 and [v4.0.10](https://pcengines.github.io/#lr-2) for the APU3. And I was amazed that the APU2 was (_and still is_) running after 8 days (_previously didn't make it past 3_) and the APU3 was stable for +4 days (_previously crashed every day_) until I flashed [v4.0.33](https://pcengines.github.io/#lr-25) that froze after 6 hours up-time.

So... mmh, true this is only a limited test on 2 boards. But the results seem somewhat promising. I'm curious if you people can give v4.0.9(10) a try and see what happens.

bylienator, 2022-02-10:
I've downgraded the BIOS to the oldest version as suggested:
```
PC Engines apu2
coreboot build 20170531
BIOS version v4.0.9
4080 MB ECC DRAM

SeaBIOS (version rel-1.10.2.1)
```
But running stress-ng on OpenWRT 21.02.1 exhibits the same behaviour:
```
stress-ng --cpu 4 --verbose
stress-ng: debug: [3186] system: Linux OpenWrt 5.4.154 #0 SMP Sun Oct 24 09:01:35 2021 x86_64
stress-ng: debug: [3186] RAM total: 3.8G, RAM free: 3.8G, swap free: 0.0
stress-ng: debug: [3186] 4 processors online, 4 processors configured
stress-ng: info:  [3186] defaulting to a 86400 second (1 day, 0.00 secs) run per stressor
stress-ng: info:  [3186] dispatching hogs: 4 cpu
stress-ng: debug: [3186] cache allocate: using cache maximum level L2
stress-ng: debug: [3186] cache allocate: shared cache buffer size: 2048K
stress-ng: debug: [3186] starting stressors
stress-ng: debug: [3187] stress-ng-cpu: started [3187] (instance 0)
stress-ng: debug: [3187] stress-ng-cpu using method 'all'
stress-ng: debug: [3188] stress-ng-cpu: started [3188] (instance 1)
stress-ng: debug: [3188] stress-ng-cpu using method 'all'
stress-ng: debug: [3186] 4 stressors started
stress-ng: debug: [3189] stress-ng-cpu: started [3189] (instance 2)
stress-ng: debug: [3189] stress-ng-cpu using method 'all'
stress-ng: debug: [3190] stress-ng-cpu: started [3190] (instance 3)
stress-ng: debug: [3190] stress-ng-cpu using method 'all'
[  197.558957] PANIC: double fault, error_code: 0x0
[  197.563956] CPU: 2 PID: 3190 Comm: stress-ng Not tainted 5.4.154 #0
[  197.570548] Hardware name: PC Engines apu2/apu2, BIOS v4.0.9 05/31/2017
[  197.577737] RIP: 0010:0xffffffff81a01321
[  197.582035] Code: 48 0f b3 1c 25 16 f2 01 00 eb 05 49 0f ba ee 3f 41 0f 22 de e9 ab f7 ff ff fc 56 48 8b 74 24 05
[  197.601939] RSP: 0000:fffffe0000068000 EFLAGS: 00010046
[  197.607574] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  197.615133] RDX: 0000000000016bc0 RSI: ffffffff81a01158 RDI: fffffe0000068088
[  197.622739] RBP: fffffe0000068078 R08: 0000000000000000 R09: 0000000000000000
[  197.630421] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  197.638021] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  197.645744] FS:  00007fe03fafbd48(0000) GS:ffff88811ad00000(0000) knlGS:0000000000000000
[  197.654413] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  197.660494] CR2: fffffe0000067ff8 CR3: 00000001174dc000 CR4: 00000000000406e0
[  197.668067] Call Trace:
[  197.670688]  <ENTRY_TRAMPOLINE>
[  197.674173]  ? 0xffffffff81a00ae0
[  197.677698]  ? 0xffffffff81053c81
[  197.681129]  ? 0xffffffff81054099
[  197.684638]  0xffffffff81a01184
[  197.687985] RIP: 0010:0xffffffff81053c81
[  197.692190] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  197.712206] RSP: 0000:fffffe0000068138 EFLAGS: 00010046
[  197.717729] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  197.725259] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068158
[  197.732825] RBP: fffffe0000068148 R08: 0000000000000000 R09: 0000000000000000
[  197.740635] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  197.748430] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  197.756065]  ? 0xffffffff81a00ae0
[  197.759548]  ? 0xffffffff81054099
[  197.763124]  0xffffffff81a01184
[  197.766468] RIP: 0010:0xffffffff81053c81
[  197.770791] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  197.790783] RSP: 0000:fffffe0000068208 EFLAGS: 00010046
[  197.796406] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  197.804050] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068228
[  197.811662] RBP: fffffe0000068218 R08: 0000000000000000 R09: 0000000000000000
[  197.819267] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  197.826946] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  197.834825]  ? 0xffffffff81a00ae0
[  197.838390]  ? 0xffffffff81054099
[... 151 lines cut ...]
[  198.925730] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  198.933455]  ? 0xffffffff81a00ae0
[  198.937009]  ? 0xffffffff81054099
[  198.940596]  0xffffffff81a01184
[  198.943932] RIP: 0010:0xffffffff81053c81
[  198.948257] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  198.968250] RSP: 0000:fffffe0000068e38 EFLAGS: 00010046
[  198.973902] RAX: 0000000000016bc0 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  198.981608] RDX: 0000000000016bc0 RSI: 0000000000000000 RDI: fffffe0000068e58
[  198.989251] RBP: fffffe0000068e48 R08: 0000000000000000 R09: 0000000000000000
[  198.996826] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000016bc0
[  199.004507] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  199.012147]  ? 0xffffffff81a00ae0
[  199.015799]  ? 0xffffffff81054099
[  199.019332]  0xffffffff81a01184
[  199.022689] RIP: 0010:0xffffffff81053c81
[  199.026841] Code: 04 0f 85 41 ff ff ff 41 b8 02 00 00 00 b9 07 00 00 00 4c 89 e2 48 89 de e8 fc f8 ff ff eb a1 09
[  199.046798] RSP: 0000:fffffe0000068f08 EFLAGS: 00010046
[  199.052283] RAX: ffffffff00101150 RBX: 0000000000000000 RCX: ffffffff81a00ae0
[  199.059694] RDX: ffffffff00101150 RSI: 0000000000000010 RDI: fffffe0000068f28
[  199.067261] RBP: fffffe0000068f18 R08: 0000000000000000 R09: 0000000000000000
[  199.074724] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffff00101150
[  199.082375] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  199.090012]  ? 0xffffffff81a00ae0
```

bylienator, 2022-02-10:
Again seems to point at CPU 2 but don't know how to definitely confirm a hardware issue in a way that warrants an RMA.

bylienator, 2022-02-26:
I've requested an RMA through the reseller.

Sadly still no answer from PC Engines support via mail ([support@pcengines.ch](mailto:support@pcengines.ch)).

mkopec, 2022-05-27:
Closing as resolved by RMA. I can reopen, if it turns out that the new hardware also crashes

closed 2022-05-27 by mkopec