pcengines/apu2-documentation #296: SFP configuration cannot be set using `ethtool`

frederictobiasc, 2022-03-23:
**PCEngines firmware version**

apu6_v4.14.0.6.rom

**APU variant**

APU6B4

**OS and OS Version**

Tested on
Arch Linux with Linux 5.16.14
and also on OpenWRT 21.02.2

**Affected component(s), peripheral(s) or functionality**

SFP port

**Brief summary**

Dear pcengines team,
When trying to manually set the parameters on the SFP port via ethtool, this does not work. Instead, an error message is displayed. This causes that my GPON SFP modules cannot be configured offline, since they only enable auto negotiation if GPON fibre uplink is present.

**How reproducible**

1

**How to reproduce**

Steps to reproduce the behavior:
1. *Carefully* connect an SFP without auto negotiation enabled
2. Enter `ethtool -s enp1s0 port fibre speed 1000 duplex full autoneg off` as root

**Expected behavior**

No error message and successful setup.

**Actual behavior**

Error message and auto-negotiation still enabled
```
[root@rtr ~]# ethtool -s enp1s0 port fibre speed 1000 duplex full autoneg off
netlink error: link settings update failed
netlink error: Operation not supported
[root@rtr ~]# ethtool -s enp1s0 speed 1000 duplex full autoneg off
[root@rtr fw]# ethtool enp1s0
Settings for enp1s0:
        Supported ports: [ FIBRE ]
        Supported link modes:   1000baseKX/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseKX/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Auto-negotiation: on
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: no
[root@rtr ~]# ethtool -m enp1s0
        Identifier                                : 0x03 (SFP)
        Extended identifier                       : 0x04 (GBIC/SFP defined by 2-wire interface ID)
        Connector                                 : 0x01 (SC)
        Transceiver codes                         : 0x00 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x00
        Transceiver type                          : Ethernet: 1000BASE-LX
        Encoding                                  : 0x03 (NRZ)
        BR, Nominal                               : 1200MBd
        Rate identifier                           : 0x00 (unspecified)
        Length (SMF,km)                           : 20km
        Length (SMF)                              : 20000m
        Length (50um)                             : 0m
        Length (62.5um)                           : 0m
        Length (Copper)                           : 0m
        Length (OM3)                              : 0m
        Laser wavelength                          : 1310nm
        Vendor name                               : ZYXEL___________
        Vendor OUI                                : 00:00:00
        Vendor PN                                 : PMG3000-D20B____
        Vendor rev                                : V1.0
        Option values                             : 0x00 0x1a
        Option                                    : RX_LOS implemented
        Option                                    : TX_FAULT implemented
        Option                                    : TX_DISABLE implemented
        BR margin, max                            : 0%
        BR margin, min                            : 0%
        Vendor SN                                 : S214125500938___
        Date code                                 : 150525
```
I also tried different speed, different combination of options, but it seems that no settings are possible at all.

**Screenshots**

N/A

**Additional context**

Works on other devices with SFPs

**Solutions you've tried**

yurisyb, 2023-06-19:
Any update on this please

pietrushnic, 2023-06-19:
https://github.com/pcengines/apu2-documentation/issues/319