pcengines/apu2-documentation #299: [APU.3D4] SIM missing after firmware update

computersalat, 2022-04-17:
**PCEngines firmware version:**
v4.16.0.2

**APU2 variant:**
APU3D4

**Question:**
After Firware Upgrade the SIM is not detected anymore.

```
mmcli -m 0
  -----------------------------
  General  |         dbus path: /org/freedesktop/ModemManager1/Modem/0
           |         device id: c233bea49daf74b6a7d142455188d4790e530ac3
  -----------------------------
  Hardware |      manufacturer: Quectel
           |             model: EC25
           | firmware revision: EC25EFAR06A11M4G
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 862348059479360
  -----------------------------
  System   |            device: /sys/devices/pci0000:00/0000:00:13.0/usb4/4-1/4-1.3
           |           drivers: option1, qmi_wwan
           |            plugin: Quectel
           |      primary port: ttyUSB2
           |             ports: ttyUSB0 (qcdm), ttyUSB2 (at), wwan0 (unknown), ttyUSB3 (at)
  -----------------------------
  Status   |             state: failed
           |     failed reason: sim-missing
           |       power state: on
           |    signal quality: 0% (cached)
  -----------------------------
  Modes    |         supported: allowed: 2g, 3g, 4g; preferred: none
           |           current: allowed: any; preferred: none
  -----------------------------
  IP       |         supported: ipv4, ipv6, ipv4v6
```

In BIOS I did only enable PXE.
Any Idea how to fix ?
Or do I need to switch to a `specific` Firmware.

miczyg1, 2022-04-24:
@computersalat what was your previous firmware version before upgrade?

mkopec, 2022-04-25:
@computersalat I have just tested a Quectel EC25-E in an apu3 board (Debian 11, firmware 4.16.0.2). When I placed the SIM card in the left SIM slot (when looking from the front of the device) I had the same error in mmcli (`sim-missing`), but it started working when I moved the SIM card to the right slot.

Please check if moving the SIM card to the right slot helps.

computersalat, 2022-04-25:
> @computersalat I have just tested a Quectel EC25-E in an apu3 board (Debian 11, firmware 4.16.0.2). When I placed the SIM card in the left SIM slot (when looking from the front of the device) I had [...]

SIM is here:
```
J15 (mPCIe2) | USB, SIM1 | only LTE modules
```
That is the slot where the SIM worked before firmware update (left one)
I will try the other slot ...

computersalat, 2022-04-25:
> SIM is here:
>
> ```
> J15 (mPCIe2) | USB, SIM1 | only LTE modules
> ```
>
> That is the slot where the SIM worked before firmware update (left one) I will try the other slot ...

so SLOT changed after firmware upgrade ... nice gadget.
@mkopec  Thank yoiu :-)
I tried it while the box was running even with a restart of ModemManger without success ... strange
Did I miss something ?

mkopec, 2022-04-25:
Glad it worked for you. Could be that an old firmware version had the SIM slots reversed. We'll check the documentation and see if it needs updating

> I tried it while the box was running even with a restart of ModemManger without success ... strange
> Did I miss something ?

Could be that the modem needs a reset after changing the SIM card, like some phones do.

miczyg1, 2022-04-25:
@mkopec there is a simswap chip o the apu devices. The selected SIM is controlled via a GPIO from SoC. It may be possible that the GPIO changed its default value in coreboot between some versions.

mkopec, 2022-04-26:
Added an explantation for the SIM slot order: #300

closed 2022-04-26 by mkopec