pcengines/apu2-documentation #304: ECC not recognised in APU6a

PrzemyslawBanasiak13, 2022-05-26:
**PCEngines firmware version**
4.16.0.4

**APU variant**
APU6a

**OS and OS Version**
Linux debian 5.10.0-14-amd64 #1 SMP Debian 5.10.113-1 (2022-04-29) x86_64

**Affected component(s), peripheral(s) or functionality**
Ability to confirm ECC presence using `dmidecode`

**Brief summary**
ECC not recognized by `dmidecode`

**How reproducible**
Every time (100%)

**How to reproduce**

Steps to reproduce the behavior:
1. Boot and login to Debian OS from iPXE
2. Use the command `dmidecode -t memory`
3. From output read line containing `Error Correction Type: ...`   

**Expected behavior**
The output should contain: `Error Correction Type: Multi-bit ECC`

**Actual behavior**
The output contains: `Error Correction Type: None`

**Screenshots**

**Additional context**

**Solutions you've tried**
Retesting several times.

miczyg1, 2022-05-26:
1. We have apu6b in the rack.
2. I don't know if the ECC is supported. If the board has 4GB RAM then maybe yes.
3. If ECC is supported the board should have 9 memory chips in total: 4 on the top and 5 on the bottom. If there are 4 on the top and 4 on the bottom ECC is not supported.
4. Check SOL string if ECC is displayed

PrzemyslawBanasiak13, 2022-05-26:
Problem occurs also in APU4

PrzemyslawBanasiak13, 2022-05-27:
Problem occurs also in APU3

mkopec, 2022-05-27:
Looks like machines at `10.174` and `10.176` both have 2GB of RAM and thus do not support ECC. I wonder why the test passed before, but fails now?

PrzemyslawBanasiak13, 2022-05-27:
The test was passing when 2GB ram was recognized and Error Correction Type was None. The problem was caused by the test environment and is fixed now.

closed 2022-05-27 by PrzemyslawBanasiak13