pcengines/apu2-documentation #309: GPIO stuff has no script... for sim swap/wifi enable (script included also includes coldboot)

Tristan79, 2022-07-16:
**The problem you're addressing (if any)**

GPIO stuff has no script... for sim swap/wifi enable/coldboot

**Describe the solution you'd like**

Include below script in GPIO section which encapsulate the gpio commands possible

**Where is the value to a user, and who might that user be?**

So they don't have to script it again...

**Describe alternatives you've considered**

Making a github repo... but currently I have no puff to maintain it... maybe when I have the auto firmware downloader/updater
finished... but that could be months or never.

**Additional context**

There is a lot of information scattered around which could be very useful.
Wish you poured it in one location... and link it from you frontpage.

```
#!/bin/bash
echo "apuctl v1 2022 (c) MonkeyCat (code.monkeycat.nl)"
echo
echo "license:     public domain
echo
echo "info         only run on apu2 and higher boards"
echo
echo "todo         there is no checking if it is an actually apu"
echo "             sim not work/show up for apu2"
echo "             probably check if there is an actual wifi card in the mpcie slots"
echo "             should look up the user if it has write access to the gpio"
echo

if [[ "$#" == "1" ]]; then
    if [ "$EUID" -ne 0 ]; then
        echo "suggestion   please run as root or as someone with write access to the gpio"
        echo
    fi

    sim=$(cat /sys/class/gpio/gpio410/value)
    wifi1=$(cat /sys/class/gpio/gpio391/value)
    wifi2=$(cat /sys/class/gpio/gpio392/value)

    if [[ "$1" == "status" ]]; then
        if [[ "$sim" == "0" ]]; then
            echo "sim slot one is selected"
        else
            echo "sim slot two is selected"
        fi
        if [[ "$wifi1" == "1" ]]; then
            echo "wifi mpcie2 on"
        else
            echo "wifi mpcie2 off"
        fi
        if [[ "$wifi2" == "1" ]]; then
            echo "wifi mpcie3 on"
        else
            echo "wifi mpcie3 off"
        fi
        exit
    fi

    echo 386 > /sys/class/gpio/export
    echo 387 > /sys/class/gpio/export
    echo 391 > /sys/class/gpio/export
    echo 392 > /sys/class/gpio/export
    echo 410 > /sys/class/gpio/export
    echo out > /sys/class/gpio/gpio386/direction
    echo out > /sys/class/gpio/gpio387/direction
    echo out > /sys/class/gpio/gpio391/direction
    echo out > /sys/class/gpio/gpio392/direction
    echo out > /sys/class/gpio/gpio410/direction
    echo 1 > /sys/class/gpio/gpio386/active_low
    echo 1 > /sys/class/gpio/gpio387/active_low
    echo 1 > /sys/class/gpio/gpio391/active_low
    echo 1 > /sys/class/gpio/gpio392/active_low
    echo 1 > /sys/class/gpio/gpio410/active_low

    if [[ "$1" == "on" ]]; then
        echo 1 > /sys/class/gpio/gpio391/value
        echo 1 > /sys/class/gpio/gpio392/value
    elif [[ "$1" == "off" ]]; then
        echo 0 > /sys/class/gpio/gpio391/value
        echo 0 > /sys/class/gpio/gpio392/value
    elif [[ "$1" == "one-on" ]]; then
        echo 1 > /sys/class/gpio/gpio391/value
    elif [[ "$1" == "one-off" ]]; then
        echo 0 > /sys/class/gpio/gpio391/value
    elif [[ "$1" == "two-on" ]]; then
        echo 1 > /sys/class/gpio/gpio392/value
    elif [[ "$1" == "two-off" ]]; then
        echo 0 > /sys/class/gpio/gpio392/value
    elif [[ "$1" == "sim-one" ]]; then
        echo 0 > /sys/class/gpio/gpio410/value
    elif [[ "$1" == "sim-two" ]]; then
        echo 1 > /sys/class/gpio/gpio410/value
    elif [[ "$1" == "one-reset" ]]; then
        echo 1 > /sys/class/gpio/gpio386/value
        sleep 1
        echo 0 > /sys/class/gpio/gpio386/value
    elif [[ "$1" == "two-reset" ]]; then
        echo 1 > /sys/class/gpio/gpio387/value
        sleep 1
        echo 0 > /sys/class/gpio/gpio387/value
    elif [[ "$1" == "coldboot" ]]; then
        for i in s u; do echo $i | sudo tee /proc/sysrq-trigger; sleep 15; done
        echo -ne "\xe" | dd of=/dev/port bs=1 count=1 seek=$((0xcf9))
    fi

else
    echo "Usage: apuctl command"
    echo
    echo " on             enable mpcie wifi transmission"
    echo " off            disable mpcie wifi transmission"
    echo " one-on         enable mpcie2 wifi transmission"
    echo " one-off        disable mpcie2 wifi transmission"
    echo " two-on         enable mpcie3 wifi transmission"
    echo " two-off        disable mpcie3 wifi transmission"
    echo " sim-one        select simcard one"
    echo " sim-two        select simcard two"
    echo " one-reset      reset mpcie2"
    echo " two-reset      reset mpcie3"
    echo " coldboot       cold reboot"
    echo
fi
```

Monkeycat-nl, 2022-07-17:
**Auto firmware downloader and updater**: https://github.com/Monkeycat-nl/APUFirmware
**Hardware wifi switch and simcard slot selector** https://github.com/Monkeycat-nl/APUCtl

For the apu tools aficionados...

macpijan, 2022-08-02:
Hello @Tristan79 @Monkeycat-nl and thans for these ideas.
I can see there is a need for such script among the PC Engines user base, so we might
adapt it to the PC Engines github and documentation.

Would you like to elaborate on how you tested this script and which cards did you use?

Tristan79, 2022-08-02:
apu4c4, 2x apu4d4...  3x vlx600 wifi cards...

you should probably update in your docs which wifi card works... the recommended wifi card (200/600/900) is NOT available due to the chip shortage... 

the metal box...sd slot cutout and atleast two more wifi antenna mounting circles would be nice...

do these work? https://nl.aliexpress.com/item/1005003173274741.html (with beter search 10 buck cheaper...)

also if i have known the apu3 and apu4 could not use **two** wifi cards internal... i probably have bought the apu2 4gb version... **or does it work?** can I use the a second wifi card in the middle slot?

or can I only use a mini-pci usb extender...???

also https://www.bol.com/nl/nl/p/samsung-evo-plus-microsdxc-256-gb-versie-2020/9300000001986684/?s2a= does not work in sd 3.0 mode... in debian bullseye (and ipfire). Total kernel crash in dmesg... Small files will copy... large files will crash the kernel... it will copy a couple of gigs... en then crash... disabling sd 3.0 will fix it... speed reduction for backups made to the sd card is significant....  i have 3 apu4 and 3  samsung sd cards... all have the same behaviour...so that is no hardware fluke..

4gb is ok for firewall... but barely enough to run a vm homeserver webserver email database on top of a hypervisor... not to mention docker... please consider a model for home (lab) users...
upgrading to more memory... ;-)

For the recommended wifi cards... I have an ath driver for debian automatic patching scripts... yes... it recompile the ath drivers when a new kernel is updated...  why? the last remaining card (wlx600 in my instance) and linux NEW behaviour if the country is not set... which 2 of 3 of my card i bought was the case... this will limit the card to almost NO channel... it excludes almost every channel...

patching it will remove that limitation... will upload on request :-)

Added https://nl.aliexpress.com/item/1005003444392756.html for 2.5gb network... throughput seems to be right...

You can use the scripts...

Tristan79, 2022-08-02:
also no throtling with 100 cpu on 4 cores due to temperature.... never ever got above 80...  recommended swap size at least 10gb... (zram can help :-) 4gb is just... minimal :-)

scripts tested on ipfire & debian.. :-)

Monkeycat-nl, 2022-08-02:
@Tristan79 show the scripts...

Monkeycat-nl, 2022-08-02:
@Tristan79 thanks...

debian... wifi

https://github.com/Monkeycat-nl/APUWifi/
in combo with https://github.com/singe/wifi-frequency-hacker

Monkeycat-nl, 2022-08-02:
@macpijan send new hardware to test.... you can use scripts :-)

Tristan79, 2022-08-02:
@Monkeycat-nl all good... 

no worries! here is my hostapd for 4 wifi network... in combo with the driver recompile it should provide 4 networks... the extra wifi networks will disappear when the driver is not recompiled :( 

make sure things like bridge=bryellow (in my case vlan 600 defines as bridge yellow) are set... could not go beyond 4 wifi networks... adding an extra will not work...

```
hw_mode=a
driver=nl80211
ssid=H
channel=0
ignore_broadcast_ssid=0
bridge=brblue
wpa_pairwise=TKIP
rsn_pairwise=CCMP
auth_algs=1
#macaddr_acl=0

interface=wlp5s0
country_code=NL
ieee80211d=1
ieee80211h=1

max_num_sta=128
ieee80211ac=1
ieee80211n=1
wmm_enabled=1
#ht_capab=[OFDM][BPSK][QPSK][DBPSK][DQPSK][16-QAM][64-QAM][256-QAM][HT20][HT40+][HT40-]
ht_capab=[OFDM][BPSK][QPSK][DBPSK][DQPSK][16-QAM][64-QAM][256-QAM][HT20][HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40][HT40-][MAX-AMSDU-7935]
#vht_oper_centr_freq_seg0_idx=42
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][BF-ANTENNA-2][SOUNDING-DIMENSION-2][VHT-LINK-ADAPT3]
vht_oper_chwidth=1

#logger_syslog=-1
#logger_syslog_level=0
#logger_stdout=-1
#logger_stdout_level=4
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0
disassoc_low_ack=1
#ignore_broadcast_ssid=0
#noscan=1
ieee80211w=2
wpa=2
wpa_passphrase=h
wpa_key_mgmt=WPA-PSK
wmm_enabled=1
#basic_rates=60
#supported_rates=60 90 120 180 240 360 480 540
preamble=1
uapsd_advertisement_enabled=1

wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

### TX queue parameters
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data2_burst=0
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5

bss=wifi5v0
bssid=10:13:10:95:fe:0b
ssid=E
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=e
bridge=brpurple

bss=wifi5v1
bssid=10:13:10:95:fe:0c
ssid=C
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=c
bridge=brblack

bss=wifi5v2
bssid=10:13:10:95:fe:0d
ssid=T
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=t
bridge=bryellow

bss=wifi5v3
bssid=10:13:10:95:fe:0e
ssid=X
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=x
bridge=brwhite
```

Monkeycat-nl, 2022-08-03:
@macpijan at least be in front of the queue of new hardware... :-)
@Tristan79 sadly sd 3.0 bios option is broken...  anyway wifi script work great! big test will be next debian kernel upgrade :-)

Tristan79, 2022-08-03:
No... no mesh (there are no docs or examples to follow for mesh)... it was already a bitsh getting hostap to work to have more then one bssid...  to get it to work required bssid to be harcoded into the config file for every extra bssid... would not work without it...

Monkeycat-nl, 2022-08-03:
my bad :-) four extra wifi networks... on top of the existing one...

Tristan79, 2022-08-03:
jup... same for realtek usb wifi stick I use... which host all the 2.4 stuff... (vlx600 does the 5 stuff)  seems 5 is the limit... for hostapd... do not know why, should not matter, as does the bssid macaddress... should not matter when not hardcoded but it does...)

Monkeycat-nl, 2022-08-03:
@macpijan running debian latest... on apu4... can we add an extra wifi card in the middle slot?

if not a lot of people are going to assume the one wireless card will handle both 2.4g and 5g which... almost no card will do... it's either, not both... so you need at least TWO wifi cards...

and a lot of people will go for the extra lan port and the higher number (apu4 vs apu2)... not knowing that one lan port less on the apu2 model... will work with two wifi cards... and still have 4gb of memory...

which means an external usb wifi card... seems to be a waste of a pcimini slot...

what's up with that?

Tristan79, 2022-08-03:
@macpijan @Monkeycat-nl ...not to mention the missing header for the second internal usb2 header... good luck soldering that when already placed the motherbord in the case.. why is that missing??? 

did I mention the coil whine? apu4c4 not... apu4d4 2x anoying whine... 
@Monkeycat-nl u got coil whine?

Monkeycat-nl, 2022-08-04:
@Tristan79 idk... the unit is in the boiler room.

@macpijan I fixed the init... will now only initialize when the gpio exports does not exists... should remove all error messages... when running twice. enjoy the code.

closed 2022-08-07 by Tristan79

miczyg1, 2022-08-16:
Referenced the APUCtl repository in the documentation: https://github.com/pcengines/apu2-documentation/pull/311