pcengines/apu2-documentation #312: pfSense 2.6.0 fails to install on apu3c

PrzemyslawBanasiak13, 2022-08-02:
**PCEngines firmware version**
4.17.0.2

**APU variant**
apu3c

**OS and OS Version**
pfSense 2.6.0

**Affected component(s), peripheral(s) or functionality**
OS installation

**Brief summary**
pfSense installation freezes

**How reproducible**
Try to install pfSense

**How to reproduce**

Steps to reproduce the behavior:
1. Flash pfSense 2.6.0 on a USB stick
2. Selec the USB with OS from the boot menu
3. Try to install pfSense

**Expected behavior**
Installation suceeds

**Actual behavior**
Installation freezes

**Screenshots**

Output: 
```
    _ __  / _|___  ___ _ __  ___  ___        
   | '_ \| |_/ __|/ _ \ '_ \/ __|/ _ \       
   | |_) |  _\__ \  __/ | | \__ \  __/       
   | .__/|_| |___/\___|_| |_|___/\___|       
   |_|                                       
                                             
                                                                               
 ╔══════════ Welcome to pfSense  ══════════       __________________________   
 ║                                         ║     /                       ___\  
 ║  1. Boot Multi user [Enter]             ║    |                      /`      
 ║  2. Boot Single user                    ║    |                     /    :-| 
 ║  3. Escape to loader prompt             ║    |      _________  ___/    /_ | 
 ║  4. Reboot                              ║    |    /` ____   / /__    ___/ | 
 ║  5. Cons: Serial                        ║    |   /  /   /  /    /   /     | 
 ║                                         ║    |  /  /___/  /    /   /      | 
 ║  Options:                               ║    | /   ______/    /   /       | 
 ║  6. Kernel: default/kernel (1 of 1)     ║    |/   /          /   /        | 
 ║  7. Boot Options                        ║        /          /___/         | 
 ║                                         ║       /                         | 
 ║                                         ║      /_________________________/  
 ╚2;03H═════════════════════════════════════════                                

   Autoboot in 3 seconds. [Space] to pause 
   Autoboot in 2 seconds. [Space] to pause 
   Autoboot in 1 seconds. [Space] to pause 
   Autoboot in 0 seconds. [Space] to pause 
 
 oading kernel... 
/boot/kernel/kernel text=0x1a5f37c data=0x140 data=0x1b63940 syms=[0x8+0x1c1370/
 0x8+0x1c7bb8] 
 oading configured modules... 
 an't find '/boot/entropy' 
 boot/kernel/zfs.ko size 0x39adb0 at 0x3cee000 
 oading required module 'opensolaris' 
 boot/kernel/opensolaris.ko size 0x9860 at 0x4089000 
 an't find '/etc/hostid' 
\ 
```

miczyg1, 2022-08-16:
Links that may be useful: 
https://github.com/pcengines/apu2-documentation/blob/master/docs/pfsense_installerconfig.md
https://github.com/pcengines/apu2-documentation/blob/master/docs/pfSense-install-guide.md
https://people.freebsd.org/~trhodes/doc/handbook/serialconsole-setup.html

Pokisiekk, 2023-01-24:
On firmware version 4.17.0.3, the installation of pfsense version 2.6.0 was successful.