pcengines/apu2-documentation #320: iommu Linux kernel WARNING on every boot

srcshelton, 2023-02-22:
**PCEngines firmware version**

Currently `v4.19.0.1`, but previous firmware releases were also affected.

**APU variant**

`apu2e4`

**OS and OS Version**

`Linux apu2e 6.1.4-gentoo #3 SMP Sat Feb  4 14:42:43 GMT 2023 x86_64 AMD GX-412TC SOC AuthenticAMD GNU/Linux`

**Affected component(s), peripheral(s) or functionality**

~20 second boot delay and kernel warning issued.

**Brief summary**

I have the AMD IOMMU enabled in the BIOS settings.

On every boot with at least every 5.x and 6.x kernel, the boot sequence output reads:

```
[    1.961190] pci 0000:00:02.2: PCI bridge to [bus 01]
[    1.966873] pci 0000:00:02.2:   bridge window [io  0x1000-0x1fff]
[    1.973677] pci 0000:00:02.2:   bridge window [mem 0xd0000000-0xd00fffff]
[    1.981176] pci 0000:00:02.3: PCI bridge to [bus 02]
[    1.986846] pci 0000:00:02.3:   bridge window [io  0x2000-0x2fff]
[    1.993647] pci 0000:00:02.3:   bridge window [mem 0xd0100000-0xd01fffff]
[    2.001142] pci 0000:00:02.4: PCI bridge to [bus 03]
[    2.006813] pci 0000:00:02.4:   bridge window [io  0x3000-0x3fff]
[    2.013617] pci 0000:00:02.4:   bridge window [mem 0xd0200000-0xd02fffff]
[    2.021141] pci_bus 0000:00: resource 4 [io  0x0000-0x0cf7 window]
[    2.028043] pci_bus 0000:00: resource 5 [io  0x0d00-0xffff window]
[    2.034926] pci_bus 0000:00: resource 6 [mem 0x000a0000-0x000dffff]
[    2.041902] pci_bus 0000:00: resource 7 [mem 0xd0000000-0xffffffff]
[    2.048878] pci_bus 0000:01: resource 0 [io  0x1000-0x1fff]
[    2.055161] pci_bus 0000:01: resource 1 [mem 0xd0000000-0xd00fffff]
[    2.062138] pci_bus 0000:02: resource 0 [io  0x2000-0x2fff]
[    2.068420] pci_bus 0000:02: resource 1 [mem 0xd0100000-0xd01fffff]
[    2.075389] pci_bus 0000:03: resource 0 [io  0x3000-0x3fff]
[    2.081669] pci_bus 0000:03: resource 1 [mem 0xd0200000-0xd02fffff]
[    2.089644] pci 0000:00:13.0: PME# does not work under D3, disabling it
[    2.097060] PCI: CLS 64 bytes, default 64
[    2.101816] pci 0000:00:00.2: AMD-Vi: IOMMU performance counters supported
[    2.109567] pci 0000:00:02.0: Adding to iommu group 0
[    2.115390] pci 0000:00:02.2: Adding to iommu group 1
[    2.121200] pci 0000:00:02.3: Adding to iommu group 2
[    2.127017] pci 0000:00:02.4: Adding to iommu group 3
[    2.132841] pci 0000:00:08.0: Adding to iommu group 4
[    2.138634] pci 0000:00:10.0: Adding to iommu group 5
[    2.144447] pci 0000:00:11.0: Adding to iommu group 6
[    2.150253] pci 0000:00:13.0: Adding to iommu group 7
[    2.156077] pci 0000:00:14.0: Adding to iommu group 8
[    2.161864] pci 0000:00:14.3: Adding to iommu group 8
[    2.167654] pci 0000:00:14.7: Adding to iommu group 8
[    2.173508] pci 0000:00:18.0: Adding to iommu group 9
[    2.179316] pci 0000:00:18.1: Adding to iommu group 9
[    2.185109] pci 0000:00:18.2: Adding to iommu group 9
[    2.190897] pci 0000:00:18.3: Adding to iommu group 9
[    2.196686] pci 0000:00:18.4: Adding to iommu group 9
[    2.202480] pci 0000:00:18.5: Adding to iommu group 9
[    2.208280] pci 0000:01:00.0: Adding to iommu group 10
[    2.214186] pci 0000:02:00.0: Adding to iommu group 11
[    2.220076] pci 0000:03:00.0: Adding to iommu group 12
[    2.226180] pci 0000:00:00.2: can't derive routing for PCI INT A
[    2.232897] pci 0000:00:00.2: PCI INT A: no GSI
[    2.238157] pci 0000:00:00.2: AMD-Vi: Found IOMMU cap 0x40
[    2.244349] AMD-Vi: Extended features (0x800290ad2, 0x0): PPR GT IA GA PC GA_vAPIC
[    2.252632] AMD-Vi: Interrupt remapping enabled
[   23.598239] ------------[ cut here ]------------
[   23.603562] WARNING: CPU: 1 PID: 1 at drivers/iommu/amd/init.c:930 enable_iommus_vapic+0x30d/0x370
[   23.613235] Modules linked in:
[   23.616995] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 6.1.4-gentoo #3
[   23.624144] Hardware name: PC Engines apu2/apu2, BIOS v4.19.0.1 01/31/2023
[   23.631720] RIP: 0010:enable_iommus_vapic+0x30d/0x370
[   23.637482] Code: 29 12 8a 4d 89 ec 41 bd e0 00 00 00 48 39 c3 0f 85 2a fe ff ff 80 0d 41 ab ba 00 01 48 c7 c7 0d 35 d0 89 e8 2e 82 b2 ff eb 02 <0f> 0b 65 48 8b 04 25 28 00 00 00 48 3b 44 24 08 75 48 48 83 c4 10
[   23.656937] RSP: 0018:ffffb6d380023b38 EFLAGS: 00010247
[   23.662873] RAX: 000000008449dcdb RBX: ffff9f5f80059800 RCX: 0000000000000001
[   23.670706] RDX: 000000000000272b RSI: 0000000000000001 RDI: 00000000000026fe
[   23.678541] RBP: 0800000000000000 R08: 0000000000000001 R09: 000000098449b5b0
[   23.686376] R10: ffffe5128402fe80 R11: 0000000000000000 R12: 0000000000000000
[   23.694212] R13: 000fffff80000000 R14: ffffffff7fffffff R15: 0000000000000000
[   23.702053] FS:  0000000000000000(0000) GS:ffff9f5faac80000(0000) knlGS:0000000000000000
[   23.710848] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   23.717296] CR2: 0000000000000000 CR3: 0000000109009000 CR4: 00000000000406e0
[   23.725131] Call Trace:
[   23.728292]  <TASK>
[   23.731105]  state_next+0x18e/0x1e0
[   23.735309]  ? iommu_setup+0x295/0x295
[   23.739768]  iommu_go_to_state+0x1f/0x33
[   23.744405]  pci_iommu_init+0x6/0x24
[   23.748697]  __initstub__kmod_pci_dma__239_208_pci_iommu_initrootfs+0x9/0xc
[   23.756363]  do_one_initcall+0x11a/0x280
[   23.761008]  do_initcall_level+0x71/0x84
[   23.765647]  do_initcalls+0x44/0x6b
[   23.769848]  kernel_init_freeable+0x10c/0x167
[   23.774918]  ? rest_init+0xb0/0xb0
[   23.779034]  kernel_init+0x15/0x190
[   23.783237]  ret_from_fork+0x1f/0x30
[   23.787521]  </TASK>
[   23.790412] ---[ end trace 0000000000000000 ]---
```

**How reproducible**

Every boot.

**How to reproduce**

Steps to reproduce the behavior:
1. Enable IOMMU in BIOS settings
2. Reboot apu2e4
3. Observe kernel boot-log output

**Expected behavior**

Kernel WARNINGs shouldn't be triggered.

**Actual behavior**

There is a ~20s delay during the early-boot sequence and then the above WARNING is output.

(I believe that @metux is the kernel maintainer for APU boards?  This may be unrelated if this is a general IOMMU issue...)

Monkeycat-nl, 2023-03-10:
same on apu4c4, apu4d4

Monkeycat-nl, 2023-03-10:
another apu4d4 with ipfire installed but with older bios (v4.17.0.3)

Linux redberry.monkeycat.nl 6.1.11-ipfire #1 SMP PREEMPT_DYNAMIC Wed Feb 22 17:07:36 GMT 2023 x86_64 AMD GX-412TC SOC AuthenticAMD GNU/Linux

```
[    0.000000] Linux version 6.1.11-ipfire (root@x86-01.haj.ipfire.org) (gcc (GCC) 11.3.0, GNU ld (GNU Binutils) 2.39) #1 SMP PREEMPT_DYNAMIC Wed Feb 22 17:07:36 GMT 2023
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.1.11-ipfire root=UUID=e32e520d-58e0-4c63-8968-3c84a96b18d1 ro panic=10 rd.auto console=ttyS0,115200n8
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1360
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000cfe8dfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cfe8e000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000012effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000012f000000-0x000000012fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: PC Engines apu4/apu4, BIOS v4.17.0.3 08/22/2022
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 998.070 MHz processor
[    0.000364] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000381] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000405] last_pfn = 0x12f000 max_arch_pfn = 0x400000000
[    0.001153] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001631] last_pfn = 0xcfe8e max_arch_pfn = 0x400000000
[    0.021094] Using GB pages for direct mapping
[    0.022634] RAMDISK: [mem 0x3490f000-0x3647efff]
[    0.022690] ACPI: Early table checksum verification disabled
[    0.022714] ACPI: RSDP 0x00000000000F2990 000024 (v02 COREv4)
[    0.022734] ACPI: XSDT 0x00000000CFEAB0E0 00007C (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022787] ACPI: FACP 0x00000000CFEACE10 000114 (v06 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022815] ACPI: DSDT 0x00000000CFEAB280 001B87 (v02 COREv4 COREBOOT 00010001 INTL 20200925)
[    0.022834] ACPI: FACS 0x00000000CFEAB240 000040
[    0.022850] ACPI: SSDT 0x00000000CFEACF30 0001F3 (v02 COREv4 COREBOOT 0000002A CORE 20200925)
[    0.022868] ACPI: MCFG 0x00000000CFEAD130 00003C (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022899] ACPI: TPM2 0x00000000CFEAD170 00004C (v04 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022930] ACPI: APIC 0x00000000CFEAD1C0 00007E (v03 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022948] ACPI: HEST 0x00000000CFEAD240 0001D0 (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022965] ACPI: IVRS 0x00000000CFEAD410 0000E0 (v02 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.022982] ACPI: SSDT 0x00000000CFEAD4F0 0048A6 (v02 AMD    AGESA    00000002 MSFT 04000000)
[    0.023000] ACPI: SSDT 0x00000000CFEB1DA0 0007C8 (v01 AMD    AGESA    00000001 AMD  00000001)
[    0.023017] ACPI: DRTM 0x00000000CFEB2570 00007C (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.023035] ACPI: HPET 0x00000000CFEB25F0 000038 (v01 COREv4 COREBOOT 00000000 CORE 20200925)
[    0.023049] ACPI: Reserving FACP table memory at [mem 0xcfeace10-0xcfeacf23]
[    0.023057] ACPI: Reserving DSDT table memory at [mem 0xcfeab280-0xcfeace06]
[    0.023062] ACPI: Reserving FACS table memory at [mem 0xcfeab240-0xcfeab27f]
[    0.023068] ACPI: Reserving SSDT table memory at [mem 0xcfeacf30-0xcfead122]
[    0.023073] ACPI: Reserving MCFG table memory at [mem 0xcfead130-0xcfead16b]
[    0.023078] ACPI: Reserving TPM2 table memory at [mem 0xcfead170-0xcfead1bb]
[    0.023083] ACPI: Reserving APIC table memory at [mem 0xcfead1c0-0xcfead23d]
[    0.023088] ACPI: Reserving HEST table memory at [mem 0xcfead240-0xcfead40f]
[    0.023093] ACPI: Reserving IVRS table memory at [mem 0xcfead410-0xcfead4ef]
[    0.023098] ACPI: Reserving SSDT table memory at [mem 0xcfead4f0-0xcfeb1d95]
[    0.023103] ACPI: Reserving SSDT table memory at [mem 0xcfeb1da0-0xcfeb2567]
[    0.023108] ACPI: Reserving DRTM table memory at [mem 0xcfeb2570-0xcfeb25eb]
[    0.023113] ACPI: Reserving HPET table memory at [mem 0xcfeb25f0-0xcfeb2627]
[    0.023242] Zone ranges:
[    0.023246]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.023256]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.023264]   Normal   [mem 0x0000000100000000-0x000000012effffff]
[    0.023273] Movable zone start for each node
[... 335 lines cut ...]
[   23.938693] RIP: 0010:enable_iommus_vapic+0x321/0x370
[... cut ...]
[   24.010259] Call Trace:
[   24.011432]  <TASK>
[   24.012252]  iommu_go_to_state+0xde0/0x1444
[   24.015175]  ? iommu_setup+0x281/0x281
[   24.017649]  amd_iommu_init+0xf/0x2d
[   24.019982]  pci_iommu_init+0xe/0x38
[   24.022287]  do_one_initcall+0x57/0x220
[   24.024858]  kernel_init_freeable+0x1f4/0x23c
[   24.027937]  ? rest_init+0xc0/0xc0
[   24.030075]  kernel_init+0x16/0x120
[   24.032296]  ret_from_fork+0x22/0x30
[   24.034620]  </TASK>
[   24.035529] ---[ end trace 0000000000000000 ]---
```

Monkeycat-nl, 2023-03-10:
on my debian 5 system I switched from kvm to xen
(installed/apt get xen)... 

no more errors regarding iommu in dmesg

i have no idea about the difference... or how it fixed it... but currently the only work around.
Since we don't know what this crash represents... (it seems to have no impact)

corbolais, 2024-09-22:
still happening on `PC Engines apu6/apu6, BIOS v4.19.0.1 01/31/2023` (APU6B4) with std Debian, std kernel `Not tainted 6.10.9-amd64 #1  Debian 6.10.9-1`:

```
[Sun Sep 22 12:45:27 2024] ------------[ cut here ]------------
[Sun Sep 22 12:45:27 2024] WARNING: CPU: 0 PID: 1 at drivers/iommu/amd/init.c:933 amd_iommu_enable_interrupts+0x557/0x870
[Sun Sep 22 12:45:27 2024] Modules linked in:
[Sun Sep 22 12:45:27 2024] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.10.9-amd64 #1  Debian 6.10.9-1
[Sun Sep 22 12:45:27 2024] Hardware name: PC Engines apu6/apu6, BIOS v4.19.0.1 01/31/2023
[Sun Sep 22 12:45:27 2024] RIP: 0010:amd_iommu_enable_interrupts+0x557/0x870
[Sun Sep 22 12:45:27 2024] Code: 0c 48 03 05 8b 0d cb 00 48 89 85 d0 00 00 00 0f 85 a3 fd ff ff 48 8b 85 b8 00 00 00 48 85 c0 0f 85 5f 01 00 00 e9 e4 01 00 00 <0f> 0b 48 8b 15 f0 fe f7 01 48 89 d0 48 f7 d0 a8 12 0f 85 4d fc ff
[Sun Sep 22 12:45:27 2024] RSP: 0018:ffffa3d4c0023d68 EFLAGS: 00010246
[Sun Sep 22 12:45:27 2024] RAX: 0000000bae85eebd RBX: 00000000001e8480 RCX: 0000000000000000
[Sun Sep 22 12:45:27 2024] RDX: 00000000000026d6 RSI: 0000000000000000 RDI: 0000000bae85c7e7
[Sun Sep 22 12:45:27 2024] RBP: ffff8a084005a000 R08: 00000000000026ae R09: 0000000000000000
[Sun Sep 22 12:45:27 2024] R10: 000000000003b640 R11: 0000000000000028 R12: 0000000080000000
[Sun Sep 22 12:45:27 2024] R13: 000ffffffffffff8 R14: 0800000000000000 R15: 2000000000000000
[Sun Sep 22 12:45:27 2024] FS:  0000000000000000(0000) GS:ffff8a086ac00000(0000) knlGS:0000000000000000
[Sun Sep 22 12:45:27 2024] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Sun Sep 22 12:45:27 2024] CR2: ffff8a086efff000 CR3: 000000008e420000 CR4: 00000000000406f0
[Sun Sep 22 12:45:27 2024] Call Trace:
[Sun Sep 22 12:45:27 2024]  <TASK>
[Sun Sep 22 12:45:27 2024]  ? __warn+0x80/0x120
[Sun Sep 22 12:45:27 2024]  ? amd_iommu_enable_interrupts+0x557/0x870
[Sun Sep 22 12:45:27 2024]  ? report_bug+0x164/0x190
[Sun Sep 22 12:45:27 2024]  ? handle_bug+0x3c/0x80
[Sun Sep 22 12:45:27 2024]  ? exc_invalid_op+0x17/0x70
[Sun Sep 22 12:45:27 2024]  ? asm_exc_invalid_op+0x1a/0x20
[Sun Sep 22 12:45:27 2024]  ? amd_iommu_enable_interrupts+0x557/0x870
[Sun Sep 22 12:45:27 2024]  ? amd_iommu_enable_interrupts+0x3db/0x870
[Sun Sep 22 12:45:27 2024]  state_next+0x1091/0x2450
[Sun Sep 22 12:45:27 2024]  amd_iommu_init+0x21/0x80
[Sun Sep 22 12:45:27 2024]  ? __pfx_pci_iommu_init+0x10/0x10
[Sun Sep 22 12:45:27 2024]  pci_iommu_init+0x12/0x40
[Sun Sep 22 12:45:27 2024]  do_one_initcall+0x5b/0x320
[Sun Sep 22 12:45:27 2024]  kernel_init_freeable+0x31b/0x460
[Sun Sep 22 12:45:27 2024]  ? __pfx_kernel_init+0x10/0x10
[Sun Sep 22 12:45:27 2024]  kernel_init+0x1a/0x1c0
[Sun Sep 22 12:45:27 2024]  ret_from_fork+0x34/0x50
[Sun Sep 22 12:45:27 2024]  ? __pfx_kernel_init+0x10/0x10
[Sun Sep 22 12:45:27 2024]  ret_from_fork_asm+0x1a/0x30
[Sun Sep 22 12:45:27 2024]  </TASK>
[Sun Sep 22 12:45:27 2024] ---[ end trace 0000000000000000 ]---
```

This also happened with previous firmware versions.

pietrushnic, 2024-09-23:
Because of [this](https://github.com/pcengines/apu2-documentation/issues/319), development continues only under Dasharo. I would not expect any feedback from the PC Engines of 3mdeb teams. Of course, the whole organization remains open to the community, and we don't plan to archive it, at least for now.

Firmware development happens under [Dasharo](https://github.com/dasharo) and is reported regularly during [Dasharo User Group](https://vpub.dasharo.com/o/1) virtual meetup. Next firmware release milestone is [here](https://github.com/Dasharo/dasharo-issues/milestone/39).