pcengines/apu2-documentation #323: apu_fw_updater_opnsense.sh fails. file has moved.

dedors, 2023-11-16:
**PCEngines firmware version**
`v4.17.0.1`

**APU variant**
`apu2`

**Brief summary**
apu firmware update script (apu_fw_updater_opnsense.sh) no longer working. the files has moved.

**How reproducible**
run script.
cat /tmp/apu2_v4.19.0.1.SHA256

**Expected behavior**
script downloads, verify and flashes firmware

**Actual behavior**
it stops at verifying.

cat /tmp/apu2_v4.19.0.1.SHA256
```
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>302 Found</title>
</head><body>
<h1>Found</h1>
<p>The document has moved <a href="https://dl.3mdeb.com/open-source-firmware/pcengines/apu2/apu2_v4.19.0.1.SHA256">here</a>.</p>
</body></html>
```

**Solutions you've tried**
added "dl." to SRC host in the script worked: `SRC="https://dl.3mdeb.com/open-source-firmware/pcengines/${TYPE}/"`

pietrushnic, 2023-11-16:
Thanks for reporting. This looks like a trivial change, we would be glad to accept PR.

doktornotor, 2023-12-09:
See https://github.com/pcengines/apu2-documentation/pull/326