pcengines/apu2-documentation #328: My thanks and congratulations

seamusdemora, 2023-12-18:
**PCEngines firmware version**

I don't actually know!

**APU2 variant**

I don't know that either!

**Affected component(s), peripheral(s) or functionality (if applicable)**

All components I suppose...

**Brief summary**

I'd like for you guys to stop for a moment, and give yourself a big pat on the back! Seriously - I'll explain under the next heading.

**Additional context**

I've owned my APU2 for several years now. About 3 years ago I installed Ubuntu server on it... this wasn't my first choice of OS, but it was one that worked - i.e. actually installed & ran without nagging & whining. 

Since installing this OS on the APU2, I have had no downtime due to firmware issues, or SD card issues - or anything at all really. Actually zero down time, except for the occasional reboot required to upgrade the kernel. This makes me very happy 'cause I love it when "shit just works". LOL. You've made an ultra-reliable small computer, and you are to be congratulated. Please pass my thanks along to the "head guy" - whoever he may be. 

You could do me one other favor: Please buy the Raspberry Pi organization, and re-tool it so that they can make machines as reliable as yours! 

Have a great day!
~Seamus

pietrushnic, 2023-12-18:
Dear Seamus,
First and foremost, thank you for taking the time to share your experience with PC Engines apu2. Of course, all platform reliability results from the beautiful work of Pascal Dornier and other PC Engines employees. We (3mdeb) added firmware to the equation and improved the situation slightly.

Your experience with the apu2's stability and reliability is exactly what we strive for at 3mdeb. We believe in transparent, open, trustworthy devices as much as possible.

On a lighter note, while acquiring the Raspberry Pi organization is beyond our current scope, and PC Engines decided to discontinue business, we appreciate the sentiment and your confidence.
I also want to share some exciting news that might interest you. During our recent [Dasharo User Group #4](https://github.com/Dasharo/presentations/blob/main/dasharo_roadmap.md#dasharo-network-appliance-roadmap), we announced the upcoming launch of [Dasharo Entry Subscription](https://docs.dasharo.com/ways-you-can-help-us/#become-a-dasharo-entry-subscription-subscriber) support tailored explicitly for Network Appliance devices. This new offering enhances network-oriented hardware security, reliability, and performance for PC Engines hardware.

We aim to extend PC Engines hardware's life and care for its firmware's sustainability in upcoming years.

Thank you again for your support and for being part of the PC Engines community, and feel free to join [Dasharo Community](https://docs.dasharo.com/#community) to stay up to date.

Wishing you continued zero downtime and a great day!

Best regards,
Piotr Król
Founder, 3mdeb

seamusdemora, 2023-12-19:
Piotr,

Thank you for your kind reply - and for the news re PC Engines! I was not aware they had gone out of business  :( 

I never changed the firmware on my APU2, and so I guess it is the same firmware that was installed when I bought it? Oh well - I don't anticipate a *need* for upgraded firmware, but it's good to know where to find it if the need should arise.

Best wishes,
~S

pietrushnic, 2023-12-20:
> I never changed the firmware on my APU2, and so I guess it is the same firmware that was installed when I bought it? Oh well - I don't anticipate a need for upgraded firmware, but it's good to know where to find it if the need should arise.

If everything works as expected and you don't have the need for new features or fixes, then it is better to stay with something that works reliably for you.

Once more time, thank you for the feedback. I guess we can close this issue?

seamusdemora, 2023-12-20:
Yes - thanks again, and best wishes

closed 2023-12-20 by seamusdemora