pcengines/apu2-documentation #330: pce-fw-builder doesnt support building latest firmware versions

mikehoyio, 2024-03-09:
**PCEngines firmware version**
v4.19.0.1

**APU2 variant**
apu2b4

**Question text**
Issues aren't open on the https://github.com/pcengines/pce-fw-builder repo so askign the question here. 

Following the guide [HERE](https://github.com/pcengines/apu2-documentation/blob/master/docs/microcode_patching.md) I am unable to build version v4.19.0.1 of the firmware using the pce-fw-builder scripts. I can see there is a line checking versions in build.sh, is there a reason the newer firmwares aren't supported? Is there a more up to date way of building the firmware and adding microcode patching? My aim is to build a version of the latest firmware for apu2 that provides microcode patching to provide protection from spectre/metldown attacks.

Cheers
Mike

miczyg1, 2024-03-11:
@mikehoyio I just have tired the procedure quickly and did not run into problem with build the release binary. However, doing a rebuild with microcode does not work indeed. It is caused by the coreboot memtest86plus repository change, i.e. the master branch has been renamed to main for inclusivity reasons (see also here: https://github.com/github/renaming), without preserving the master branch. The missing master branch irreversibly broke building the code from before the branch name change.

To workaround that, please do the following right before step 5 in the guide:

1. `./build.sh dev-build $PWD/release/coreboot apu2 distclean`
2. `cp $PWD/release/coreboot/configs/config.pcengines_apu2 $PWD/release/coreboot/.config`

And then proceed with step 5 as in the guide.

mikehoyio, 2024-03-11:
Thanks for the help, I'm pretty sure this was user error.

I've managed to patch and apply the new firmware now.