pcengines/apu2-documentation #331: `pce-fw-builder` `build_apus.sh` fails with "no configuration" error

srcshelton, 2024-10-29:
Having followed the instructions at https://github.com/pcengines/pce-fw-builder branch `master` (and https://github.com/Dasharo/dasharo-pq branch `24.05.00.01`), the build-process as documented at https://github.com/pcengines/pce-fw-builder#building-v24050001-or-newer fails with:

```
docker build . -t pcengines/pce-fw-builder:v24.05.00.01
STEP 1/6: FROM coreboot/coreboot-sdk:2024-03-30_cccada28f7
STEP 2/6: MAINTAINER Piotr Król <piotr.krol@3mdeb.com>
--> Using cache 73687455ea93fe5ee16a6daf34de03d78661716ee9d7b8629f748a0ec1645646
--> 73687455ea93
STEP 3/6: USER root
--> Using cache 1f713951cd38c298b39bb02001661a1df1553230b8a63773d5574ace8dd4ca80
--> 1f713951cd38
STEP 4/6: COPY ./scripts/entrypoint.sh /entrypoint.sh
--> Using cache 4bb5e333c8af28dd3ee155dd60c856f87982de6faa886210d88d2a3ba667200e
--> 4bb5e333c8af
STEP 5/6: RUN chmod +x /entrypoint.sh
--> Using cache 71be40f2ac6d96fd4ce04af495a25423009cd7b3819e66b7a637bd31e09fb27c
--> 71be40f2ac6d
STEP 6/6: ENTRYPOINT ["/entrypoint.sh"]
--> Using cache 3f1abe54439092768e9455eb596e234e76d049708eacaa0b886eae6d9f2ecbb7
COMMIT pcengines/pce-fw-builder:v24.05.00.01
--> 3f1abe544390
Successfully tagged localhost/pcengines/pce-fw-builder:v24.05.00.01
3f1abe54439092768e9455eb596e234e76d049708eacaa0b886eae6d9f2ecbb7
+ SDK_VER=v24.05.00.01
+ ./build_apus.sh ../coreboot apu2
Starting build for apu2...
/usr/src/dasharo/coreboot /usr/src/dasharo/pce-fw-builder
/usr/src/dasharo/pce-fw-builder
Dev-build coreboot mainline
usermod: no changes
Local storage:
  Cache size (GiB): 0.0 / 5.0 ( 0.00%)
Build coreboot for apu2
ERROR: no configuration exist for apu2
Error: Trying to build, but NOCOMPILE is set.
  Please file a bug with the following information:
- MAKECMDGOALS: 
- HAVE_DOTCONFIG: 
- HAVE_KCONFIG_MAKEFILE_REAL: build/util/kconfig/Makefile.real
make: *** [Makefile:174: real-all] Error 1
Build failed for apu2 with status code 2.
```

… as a result of build_apus.sh ultimately executing `docker run --rm -it -v /usr/src/dasharo/coreboot:/home/coreboot/coreboot -e USER_ID=1000 -e GROUP_ID=100 -v /usr/src/dasharo/pce-fw-builder/ccache:/home/coreboot/.ccache -v /usr/src/dasharo/pce-fw-builder/scripts:/home/coreboot/scripts pcengines/pce-fw-builder:v24.05.00.01 /home/coreboot/scripts/pce-fw-builder.sh 0 true apu2`

On the `2.0.1` branch, the result is instead:

```
+ ./build.sh dev-build /usr/src/dasharo/coreboot apu2 distclean
/usr/src/dasharo/coreboot /usr/src/dasharo/pce-fw-builder
/usr/src/dasharo/pce-fw-builder
 is UNSUPPORTED
```

… which comes from the line `echo "$tag3 is UNSUPPORTED"`, where `tag3` is never defined - although this version doesn't seem to support versions after 4.19 in any case (although this check is broken as there's a trailing `.` missing, so actually versions matching `4\.{7..19}.*` are accepted) 😟

Additionally, the documentation at https://github.com/pcengines/pce-fw-builder/tree/2.0.1#mainline & https://github.com/pcengines/pce-fw-builder/tree/master?tab=readme-ov-file#mainline is out of date, as `Dockerfile.ml` no longer exists on either branch.

srcshelton, 2024-10-29:
This is because `pce-fw-builder`'s `scripts/pce-fw-builder.sh` references `configs/config.pcengines_$1`, but this should be `config.pcengines_seabios_$1` for these versions/branches.

pietrushnic, 2024-10-29:
Yes, this is a bug. Easy to fix: https://github.com/pcengines/pce-fw-builder/commit/da9951ed9d0fbb94eabc5655c97ee02267ee0ead

I plan to keep this backward compatible only if I get a good reason. Also, `pce-fw-builder` needs versioning that match releases. I have to introduce that in v24.08.00.01. Since Dasharo (coreboot+UEFI) does not use pce-fw-builder, I may also eliminate it: https://github.com/Dasharo/dasharo-issues/issues/1117.