pcengines/apu2-documentation #335: https://github.com/pcengines/apu2-documentation/blob/master/scripts/installerconfig

hardwarebxl, 2025-02-07:
**PCEngines firmware version**

**APU2 variant**

**Question text**

As I am trying to automate the installation of pfsense (unattended installation), I was following your explanation/tutorial.
The installerconfig you show as an example and the comments you wrote that sometimes at extraction phase the installer halts, I experience the same issue but more often than not.
Since the config file and explanation is almost 7 years old, perhaps you guys made newer versions and found a reason why the extraction as halting - maybe you got a new setup for unattended installation overall?
I would be very happy if you could share this with me (or the public) - I am stuck at trying to get this done myself for over months now.

Thanks for replying!

pietrushnic, 2025-02-07:
@hardwarebxl thanks for reporting, but support for PC Engines era scripts and binaries is no longer provided due to [EOL announcement](https://docs.dasharo.com/variants/pc_engines/post-eol-fw-announcement/)

Development was moved to the paid version [UEFI support](https://shop.3mdeb.com/shop/dasharo-pro-package/1-year-dasharo-entry-subscription-for-network-appliance/) and [SeaBIOS support](https://shop.3mdeb.com/shop/dasharo-pro-package/1-year-dasharo-entry-subscription-for-network-appliance-corebootseabios/) (I'm the founder of the 3mdeb company behind Dasharo), where we support and maintain firmware releases. The only way I can see this model could work with unattended is pfSense/OPNsense would accept [fwupd support from FreeBSD](https://blog.3mdeb.com/2021/2021-02-16-fwupd-compilation-under-freebsd/) that we contributed some time ago. Then, the fwupd/LVFS ecosystem will be configured to download paid binaries. This method has not yet been implemented, but if some customers want it, I see no reason why we would not implement it.

Another choice is to use unvalidated binaries, published as the effect of build jobs in [this repository](https://github.com/Dasharo/dasharo-pq), or build binaries yourself.

gretel, 2025-02-08:
> Another choice is to use unvalidated binaries, published as the effect of build jobs in [this repository](https://github.com/Dasharo/dasharo-pq), or build binaries yourself.

side note, artifacts have all expired.

pietrushnic, 2025-02-08:
> side note, artifacts have all expired.

Thanks for letting me know. I'm very sorry about that. Still, instruction for building should be straightforward. I'm trying to get back on track with the release.

gretel, 2025-02-09:
> Thanks for letting me know. I'm very sorry about that. Still, instruction for building should be straightforward. I'm trying to get back on track with the release.

thanks @pietrushnic